Koenkk/zigbee2mqtt #25273: Philips hue turns on automatically right after turning it off

antlarr, 2024-12-20:
### What happened?

Sometimes some light (always Philips hue E27 bulbs) gets into a state where, if I turn it off (using home assistant or the 'Expose' tab in zigbee2mqtt web ui), it turns on automatically right afterwards (some times it's immediately after turning it off, sometimes it takes ~5 seconds)

I stopped home assistant to be sure there was no automation doing anything in the background. I opened the 'Exposes' tab of the light device ('Luz-Despacho' aka 0x00178801027dd1f3) in the zigbee2mqtt web ui and turn the State off :
```
[2024-12-20 09:50:05] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Luz-Despacho/set' with data '{"state":"OFF"}'
[2024-12-20 09:50:05] debug:    z2m: Publishing 'set' 'state' to 'Luz-Despacho'
[2024-12-20 09:50:05] debug:    zh:controller:endpoint: ZCL command 0x00178801027dd1f3/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2024-12-20 09:50:05] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x00178801027dd1f3:48783/11 (0,0,2)
[2024-12-20 09:50:05] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Luz-Despacho', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":366,"linkquality":127,"power_on_behavior":"on","state":"OFF","update":{"installed_version":16787456,"latest_version":16787456,"state":"idle"},"update_available":false}'
[2024-12-20 09:50:05] debug:    zh:zstack: Data confirm error (0x00178801105c11aa:29694,205,0)
```
The light turned on again almost immediately,  but the logs didn't show anything (related to that device) until ~6 seconds later, when they showed:

```
[2024-12-20 09:50:11] debug:    zh:controller: Received ZDO response: clusterId=END_DEVICE_ANNOUNCE, status=SUCCESS, payload={"nwkAddress":48783,"eui64":"0x00178801027dd1f3","capabilities":{"alternatePANCoordinator":0,"deviceType":1,"powerSource":1,"rxOnWhenIdle":1,"reserved1":0,"reserved2":0,"securityCapability":0,"allocateAddress":1}}
[2024-12-20 09:50:11] debug:    zh:controller: Device announce from '0x00178801027dd1f3:48783'
[2024-12-20 09:50:11] debug:    z2m: Device 'Luz-Despacho' announced itself
[2024-12-20 09:50:11] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Luz-Despacho"},"type":"device_announced"}'
[2024-12-20 09:50:11] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Luz-Despacho","ieee_address":"0x00178801027dd1f3"},"type":"device_announce"}'
[...]
[2024-12-20 09:50:13] debug:    z2m: Retrieving state of 'Luz-Despacho' after reconnect
[2024-12-20 09:50:13] debug:    zh:controller:endpoint: ZCL command 0x00178801027dd1f3/11 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2024-12-20 09:50:13] debug:    zh:zstack: sendZclFrameToEndpointInternal 0x00178801027dd1f3:48783/11 (0,0,1)
[2024-12-20 09:50:13] debug:    z2m: Received Zigbee message from 'Luz-Despacho', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
[2024-12-20 09:50:13] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Luz-Despacho', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":366,"linkquality":127,"power_on_behavior":"on","state":"ON","update":{"installed_version":16787456,"latest_version":16787456,"state":"idle"},"update_available":false}'
```

I'll attach the logs from the start of z2m and from around that time, just in case there's anything I missed that is of value for debugging. Also, please note that I have 63 zigbee devices and almost half of them give `Data request failed with error: 'No network route'` errors continuously because I have them powered off.

### What did you expect to happen?

I expect the light to remain turned off.

### How to reproduce it (minimal and precise)

It's not easy to reproduce since it happens maybe every few weeks (with different lights). Once a light gets into that state, it keeps doing that until it's powered off for a while and powered on again. Sometimes lights that are always turned on/off at the same time don't show the same behavior and one of them keeps turning on automatically while the other works fine.

Note that the light needs to be powered off really for some time. I just tested powering it off for 15 minutes and when I switched it on again, it kept in the same state. (I tested before 1, 3, and 5 minutes). Other times I just kept the lights unpowered for several hours to make them work correctly again.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

zStack3x0 20230507

### Adapter

Texas Instruments LAUNCHXL-CC1352P-2

### Setup

Z2m runs as a systemd service on a Raspberry Pi 5. No containers involved. The adapter is connected to the Raspberry Pi through a powered usb hub.

### Debug log

I removed lines that are clearly unrelated from the beginning of the log (not the very beginning since that probably contains useful information about the system) and  until a few seconds before I reproduced the issue . I also removed lines and after a few seconds after the issue finished to the end. If you want the full file, please tell me and I'll attach it.
[log-redacted.log](https://github.com/user-attachments/files/18208629/log-redacted.log)

antlarr, 2024-12-20:
Just for the record, I tested keeping the light bulb powered off for an hour and that "fixed" it (but it will happen again in a few days, for sure)

TechNerd010, 2024-12-22:
I have the same problem with power sockets As soon as I give the command to switch off the socket via Alexa, it is immediately switched on again. Likewise, when I want to switch the socket on, it is Automatically switched off again Update from the socket made but nothing
No home assistant automation interferes in any way.

closed 2025-03-01 by github-actions[bot]