Koenkk/zigbee2mqtt #25274: [New device support]: AVATTO 4 Channel Switch (vendor: _TZE204_f5efvtbv, model: TS0601)

dark-m0de, 2024-12-20:
### Link

https://de.aliexpress.com/item/1005008035036531.html

### Database entry

{"id":57,"type":"Router","ieeeAddr":"0xa4c1380b83586f40","nwkAddr":57737,"manufId":4417,"manufName":"_TZE204_f5efvtbv","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\r\u0000\u0000\u0000\u0013\u000e\u0000\u0000\u0000e\u000f\u0000\u0000\u0000\u0012\u0012\u0000\u0000\u0000\u0016","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1734691337348}

### Zigbee2MQTT version

1.42.0

### Comments

I tried to follow the new device guide, but my external definitions seems to not have any additional const values (aka no features discovered). The device I want to add is a 4 channel switch from a company called AVATTO.

There seems to a similar device from a company call MHCOZY (see https://www.zigbee2mqtt.io/devices/TYWB_4ch-RF.html), but I can't find any definition for that one.

Can someone guide me on what else I could try?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_f5efvtbv',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

No features discovered

dark-m0de, 2024-12-29:
anyone?

eduperez, 2025-01-27:
Discussion here: https://github.com/Koenkk/zigbee2mqtt/discussions/26003

closed 2025-04-06 by github-actions[bot]