Koenkk/zigbee2mqtt #25278: Aqara SRTS-A01 internal_heating_setpoint is different from occupied_heating_setpoint

maxicasa, 2024-12-21:
### What happened?

Sometimes when auto schedule (with https://github.com/nielsfaber/scheduler-card ) sets the setpoint to a value (e.g. 20°), the setpoint does not change. I noticed that the problem occurs in the valve with the `internal_heating_setpoint` different from the `occupied_heating_setpoint`. Below is a state with the problem:

```
{
    "away_preset_temperature": "15.0",
    "battery": 87,
    "calibrated": true,
    "child_lock": true,
    "device_temperature": 20,
    "internal_heating_setpoint": 18.5,
    "last_seen": "2024-12-21T09:23:00+01:00",
    "linkquality": 47,
    "local_temperature": 19,
    "occupied_heating_setpoint": 20,
    "power_outage_count": 1,
    "preset": "manual",
    "schedule": false,
    "schedule_settings": "mon,tue,wed,thu,fri,sat,sun|6:00,20.0|19:57,19.0|22:39,18.5|6:00,18.5",
    "sensor": "external",
    "setup": false,
    "system_mode": "heat",
    "valve_alarm": false,
    "valve_detection": false,
    "voltage": 2900,
    "window_detection": true,
    "window_open": false,
    "calibrate": null,
    "external_temperature_input": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}
```

The problem is also present when the setpoint is set manually, from zigbee2mqtt ui: 
[image]

I have 7 Aqara TRV and seems to occurs in 3 valves in particular after all night from last valve change (I don't know if there is a correlation).

### What did you expect to happen?

I expect that the setpoint to set the same value for `occupied_heating_setpoint` and the `internal_heating_setpoint`.

### How to reproduce it (minimal and precise)

Unfortunately, this happens randomly and I have not found a way to reproduce it. Probably the hours between setpoint changes need to be at least 8 hours.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20240710

### Adapter

SMLIGHT SLZB-06

### Setup

Add-on Home Assistant

### Debug log

[log.log](https://github.com/user-attachments/files/18217907/log.log)

closed 2025-02-27 by github-actions[bot]

maxicasa, 2025-03-01:
Unfortunately no. I've noticed after some random time the temperature will set correctly, but it's unpredictable. I am gradually replacing aqara TRV with sonoff because they are more stable.

agnyp, 2025-03-04:
Thanks for your answer and the tip!