Koenkk/zigbee2mqtt #25281: [New device support]: ROB_200-040-0 not recognized as Sunricher SR-ZG9092A but as ROB-200-040-0

jhaand, 2024-12-21:
### Link

https://www.robbshop.nl/robb-smarrt-zigbee-thermostaat-elektrische-vloerverwarming

### Database entry

{"id":42,"type":"Router","ieeeAddr":"0x1fff00040000001d","nwkAddr":4780,"manufId":4644,"manufName":"ROBB smarrt","powerSource":"Mains (single phase)","modelId":"ROB_200-040-0","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,9,10,513,516,1026,1794,2820],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"stackVersion":0}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1598}},"haElectricalMeasurement":{"attributes":{"4096":16,"acPowerDivisor":1,"acPowerMultiplier":1,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"acVoltageDivisor":10,"acVoltageMultiplier":1,"activePower":0,"rmsCurrent":0,"rmsVoltage":0,"acCurrentOverload":16000}},"seMetering":{"attributes":{"divisor":10,"multiplier":1,"currentSummDelivered":2}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea3b80a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea3b80a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea3b80a","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":2820,"attrId":1291,"minRepIntval":10,"maxRepIntval":65000,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":10,"maxRepIntval":65000,"repChange":50},{"cluster":2820,"attrId":1285,"minRepIntval":10,"maxRepIntval":65000,"repChange":50},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":7,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":23,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"6.9.1.0_r4","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1734782102912}

### Zigbee2MQTT version

    1.40.2 commit: e06848d

### Comments

I ordered the electric floor heater from Robbshop. The old versions were marketed as ROB_200-040-0 and advertise themselves as SR-ZG9092A from Sunricher. As mentioned in other issues. 

These units identify themselves as ROB_200-040-0 and this device is not recognized. Furthermore only a few things are exposed. While I would expect the same parameters as the SR-ZG9092A. 
[image]

I will write a PR to include the definition of the new thermostat shortly.

### External definition

```shell
const {deviceEndpoints, temperature, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['ROB_200-040-0'],
    model: 'ROB_200-040-0',
    vendor: 'ROBB smarrt',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), temperature(), electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

These units identify themselves as ROB_200-040-0 and are not recognized. Furthermore only a few parameters are exposed. While I would expect the same parameters as the SR-ZG9092A. 
[image]

Koenkk, 2024-12-22:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-22 by Koenkk

jhaand, 2024-12-27:
Thank you for fixing this so easily. 

I currently run Z2M via Podman at the moment. It will take a while before I can test this.