Koenkk/zigbee2mqtt #25283: EZSP to EMBER: TRADFRI E1745 unresponsive

weimdall, 2024-12-21:
### What happened?

After the migration from EZSP to EMBER, my E1745 motion sensors became unresponsive.
I have about 20 zigbee devices that works properly after the migration (including TRAFRI E1525 motion sensors !).

I tried to troubleshoot, but I am a bit clueless at that point.
The interview process seems to work, the occupancy is set to `null` and then the sensor become silent until I reset it again.

If I compare to the working TRAFRI E1525 sensor, the occupancy value is never `null`, 
even right after the interview process, it is a boolean...

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.4

### Adapter

EmberZNet - Skyconnect

### Setup

Add-on on Home Assistant

### Debug log

```
warning 2024-12-21 10:56:31 z2m: Device 'Motion Kitchen' left the network
info 2024-12-21 10:56:31 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Motion Kitchen","ieee_address":"0x842e14fffe1c4f14"},"type":"device_leave"}'
info 2024-12-21 10:56:38 zh:ember:ezsp: Received network/route error ROUTE_ERROR_NON_TREE_LINK_FAILURE for "54643".
info 2024-12-21 10:56:38 z2m: Zigbee: allowing new devices to join.
info 2024-12-21 10:56:38 zh:ember: [STACK STATUS] Network opened.
info 2024-12-21 10:56:39 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"0tb4n-2"}'
info 2024-12-21 10:56:42 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Motion Kitchen', payload '{"battery":60,"illuminance_above_threshold":null,"last_seen":"2024-12-21T15:56:42.208Z","linkquality":232,"occupancy":null,"requested_brightness_level":null,"requested_brightness_percent":null,"update":{"installed_version":537011747,"latest_version":537011747,"state":"idle"},"update_available":null}'
info 2024-12-21 10:56:42 zh:controller: Interview for '0x842e14fffe1c4f14' started
info 2024-12-21 10:56:42 z2m: Device 'Motion Kitchen' joined
info 2024-12-21 10:56:42 z2m: Starting interview of 'Motion Kitchen'
info 2024-12-21 10:56:42 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Motion Kitchen","ieee_address":"0x842e14fffe1c4f14"},"type":"device_joined"}'
info 2024-12-21 10:56:42 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Motion Kitchen","ieee_address":"0x842e14fffe1c4f14","status":"started"},"type":"device_interview"}'
info 2024-12-21 10:56:44 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Motion Kitchen', payload '{"battery":60,"illuminance_above_threshold":null,"last_seen":"2024-12-21T15:56:44.361Z","linkquality":232,"occupancy":null,"requested_brightness_level":null,"requested_brightness_percent":null,"update":{"installed_version":537011747,"latest_version":537011747,"state":"idle"},"update_available":null}'
info 2024-12-21 10:56:44 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Motion Kitchen', payload '{"battery":60,"illuminance_above_threshold":null,"last_seen":"2024-12-21T15:56:44.597Z","linkquality":232,"occupancy":null,"requested_brightness_level":null,"requested_brightness_percent":null,"update":{"installed_version":537011747,"latest_version":537011747,"state":"idle"},"update_available":null}'
info 2024-12-21 10:56:44 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Motion Kitchen","ieee_address":"0x842e14fffe1c4f14"},"type":"device_announce"}'
info 2024-12-21 10:56:48 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Motion Kitchen', payload '{"battery":60,"illuminance_above_threshold":null,"last_seen":"2024-12-21T15:56:48.872Z","linkquality":176,"occupancy":null,"requested_brightness_level":null,"requested_brightness_percent":null,"update":{"installed_version":537011747,"latest_version":537011747,"state":"idle"},"update_available":null}'
info 2024-12-21 10:56:48 zh:controller: Succesfully interviewed '0x842e14fffe1c4f14'
info 2024-12-21 10:56:48 z2m: Successfully interviewed 'Motion Kitchen', device has successfully been paired
info 2024-12-21 10:56:48 z2m: Device 'Motion Kitchen' is supported, identified as: IKEA TRADFRI motion sensor (E1525/E1745)
info 2024-12-21 10:56:48 z2m: Configuring 'Motion Kitchen'
info 2024-12-21 10:56:48 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"TRADFRI motion sensor","exposes":[{"access":1,"description":"Indicates whether the device detected occupancy","label":"Occupancy","name":"occupancy","property":"occupancy","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Indicates whether the device detected bright light (works only in night mode)","label":"Illuminance above threshold","name":"illuminance_above_threshold","property":"illuminance_above_threshold","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","label":"Requested brightness level","name":"requested_brightness_level","property":"requested_brightness_level","type":"numeric","value_max":254,"value_min":76},{"access":1,"category":"diagnostic","label":"Requested brightness percent","name":"requested_brightness_percent","property":"requested_brightness_percent","type":"numeric","value_max":100,"value_min":30},{"access":2,"category":"config","description":"Initiate device identification. This device is asleep by default.You may need to wake it up first before sending the identify command.","label":"Identify","name":"identify","property":"identify","type":"enum","values":["identify"]},{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E1525/E1745","options":[{"access":2,"description":"Sets the duration of the identification procedure in seconds (i.e., how long the device would flash).The value ranges from 1 to 30 seconds (default: 3).","label":"Identify timeout","name":"identify_timeout","property":"identify_timeout","type":"numeric","value_max":30,"value_min":1},{"access":2,"description":"Time in seconds after which occupancy is cleared after detecting it (default 90 seconds).","label":"Occupancy timeout","name":"occupancy_timeout","property":"occupancy_timeout","type":"numeric","value_min":0},{"access":2,"description":"Set to false to also send messages when illuminance is above threshold in night mode (default true).","label":"Illuminance below threshold check","name":"illuminance_below_threshold_check","property":"illuminance_below_threshold_check","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"IKEA"},"friendly_name":"Motion Kitchen","ieee_address":"0x842e14fffe1c4f14","status":"successful","supported":true},"type":"device_interview"}'
info 2024-12-21 10:56:50 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Motion Kitchen', payload '{"battery":60,"illuminance_above_threshold":null,"last_seen":"2024-12-21T15:56:50.827Z","linkquality":172,"occupancy":null,"requested_brightness_level":null,"requested_brightness_percent":null,"update":{"installed_version":537011747,"latest_version":537011747,"state":"idle"},"update_available":null}'
info 2024-12-21 10:56:52 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Motion Kitchen', payload '{"battery":100,"illuminance_above_threshold":null,"last_seen":"2024-12-21T15:56:52.006Z","linkquality":176,"occupancy":null,"requested_brightness_level":null,"requested_brightness_percent":null,"update":{"installed_version":537011747,"latest_version":537011747,"state":"idle"},"update_available":null}'
info 2024-12-21 10:56:52 z2m: Successfully configured 'Motion Kitchen'
info 2024-12-21 10:56:53 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Motion Kitchen', payload '{"battery":100,"illuminance_above_threshold":null,"last_seen":"2024-12-21T15:56:53.453Z","linkquality":172,"occupancy":null,"requested_brightness_level":null,"requested_brightness_percent":null,"update":{"installed_version":537011747,"latest_version":537011747,"state":"idle"},"update_available":null}'
info 2024-12-21 10:57:36 z2m: Zigbee: disabling joining new devices.
info 2024-12-21 10:57:37 zh:ember: [STACK STATUS] Network closed.
info 2024-12-21 10:57:37 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"0tb4n-3"}'
info 2024-12-21 10:57:39 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Motion Kitchen', payload '{"battery":100,"illuminance_above_threshold":null,"last_seen":"2024-12-21T15:57:39.273Z","linkquality":176,"occupancy":null,"requested_brightness_level":null,"requested_brightness_percent":null,"update":{"installed_version":537011747,"latest_version":537011747,"state":"idle"},"update_available":null}'
```

sohnvonoff, 2024-12-29:
I have the same problem. I migrated the adapter from Conbee2 to ZBT-1. I have the same configuration as #weimdal. Pairing the E1745 works. The E1745 also appears to report regularly to the concentrator. Interviews started manually are successful. However, no movements are reported.
I have ten of them in use. They all have the same problem.
Please Help...
Many Thanks

dev-a, 2025-01-04:
Visibly I have the same problem
since update to zigbee2mqtt 2.0. the "occupancy" stay to "true"

kwatt67, 2025-01-04:
I got similar and have been advised to report it here. 

Using Home Assistant Green with a ZBT-1 I tried the adapter set to stack and then ember with no luck error (or one of them at least) below but on advice I tried using ezsp and all worked perfectly well after changing it to ezsp. 

Using zstack:

`[12:41:12] INFO: Preparing to start...
[12:41:12] INFO: Socat not enabled
[12:41:13] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 12:41:21] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 12:41:21] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 12:41:21] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 12:41:21] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 12:41:21] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 12:41:21] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 12:41:21] error: 	z2m: Exiting...
[2025-01-04 12:41:21] error: 	z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:523:15)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:33)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[2025-01-04 12:41:21] error: 	z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364` 

Using ember:

`tarting Zigbee2MQTT without watchdog.
[2025-01-04 13:15:02] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 13:15:02] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 13:15:02] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 13:15:02] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB1","manufacturer":"Nabu Casa","serialNumber":"6aca14c4533aef11a3352b1455516304","pnpId":"usb-Nabu_Casa_Home_Assistant_Connect_ZBT-1_6aca14c4533aef11a3352b1455516304-if00-port0","vendorId":"10c4","productId":"ea60"} => ember: 2
[2025-01-04 13:15:02] info: 	zh:ember: Using default stack config.
[2025-01-04 13:15:03] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-04 13:15:03] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-04 13:15:03] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-04 13:15:03] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-01-04 13:15:03] info: 	zh:ember:uart:ash: Serial port opened
[2025-01-04 13:15:03] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-04 13:15:04] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-01-04 13:15:04] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-01-04 13:15:04] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-01-04 13:15:04] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 13:15:04] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 13:15:04] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 13:15:04] error: 	z2m: Exiting...
[2025-01-04 13:15:04] error: 	z2m: Error: Adapter EZSP protocol version (9) is not supported by Host [13-16].
    at EmberAdapter.emberVersion (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1403:19)
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:674:9)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)`

spry-salt, 2025-01-04:
The problem happens for me with Conbee II adapter set to Deconz.

bakgaard, 2025-01-06:
I updated from `ezsp` to `ember`, and updated from 1.42.0 to 2.0.0. Now the E1745-sensors don't work.

- I tried to undo `ember` to `ezsp`, but that didn't work.
- I downgraded from 2.0.0 to 1.42.0, but that didn't work.
- I tried rolling back the `configuration.yaml`, but the sensors never send any updates

bakgaard, 2025-01-07:
> I updated from `ezsp` to `ember`, and updated from 1.42.0 to 2.0.0. Now the E1745-sensors don't work.
>
> * I tried to undo `ember` to `ezsp`, but that didn't work.
> * I downgraded from 2.0.0 to 1.42.0, [...]

Update: I use teh ZBT-1 device, and had it upgraded to version 7.4.4.0 from 7.3.x.y. Downgrading this, using Zigbee2mqtt 2.0.0 and `ezsp` solved the issue for me. 
Downgrading was done following [this guide](https://community.home-assistant.io/t/skyconnect-v7-3-2-0-ezsp-to-ember-driver-depricated/730866/4)

sohnvonoff, 2025-01-08:
the E1745 are really tricky. I had to take the battery out and put it back in every E1745, then pair it, then start the firmware update. One of them even needed this procedure twice for the firmware update to run through.

It is also important to set this setting 

[image]

and only then do the pairing

weimdall, 2025-01-11:
The latest E1745 firmware update ( v. 24.4.5 )  fixed the issue on my end !  🎉

closed 2025-03-21 by github-actions[bot]