Koenkk/zigbee2mqtt #25293: Moes ZS-SR-EUD-2 Backlight mode & Power-on behavior no longer work

miloumeeloo, 2024-12-22:
### What happened?

Backlight mode and Power-on behavior no longer work for Moes ZS-SR-EUD-2 Star ring smart dimmer switch 2 gangs (_TZE204_zenj4lxv)
Still works for ZS-SR-EUD-1 (_TZE204_hlx9tnzb)

### Zigbee2MQTT version

1.42

### Adapter firmware version

7.4.2 [GA]

### Adapter

ZBDONGLE-E (EmberZNet)

### Setup

Add-on HAOS

### Debug log

error: 	z2m: No converter available for 'backlight_mode' ("normal")
error: 	z2m: No converter available for 'backlight_mode' ("off")
error: 	z2m: No converter available for 'backlight_mode' ("inverted")
error: 	z2m: No converter available for 'power_on_behavior' ("on")
error: 	z2m: No converter available for 'power_on_behavior' ("off")

Nevynnyi, 2024-12-23:
Same. And too much spam in z2m every seconds

miloumeeloo, 2025-01-14:
Following the update of Z2M to 2.0.0-2 and adapter firmware to 7.4.4, both functions are available again.

closed 2025-01-14 by miloumeeloo

Nevynnyi, 2025-01-14:
Yes, z2m 2.0 and ember firmware 7.4.5 fix all problem.