Koenkk/zigbee2mqtt #25294: Adapter Crashing on opening window with ikea sensors

BloodWorkXGaming, 2024-12-22:
### What happened?

Recently (maybe since upgrading the dongle to 8.0.2) the adapter always crashes on some actions in the zigbee network, like for example opening a window where a ikea PARASOLL window sensor is attached.
Other actions on the zigbee network, like reporting the temp of sensors works just fine.

I've attached the log below. Is ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT the ACKs to a zigbee device or to the adapter itself?

The adapter is connected via a usb extension cable.

Any help is appreciated! Thanks!

### Zigbee2MQTT version

    1.42.0 commit: 861cba6

### Adapter firmware version

EmberZNet  8.0.2 [GA]

### Adapter

Sonoff Dongle E (EFR32MG21 )

### Setup

Docker container, x86

### Debug log

info 2024-12-22 17:29:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bathroom_window_contact', payload '{"battery":60,"contact":true,"last_seen":"2024-12-22T16:29:48.835Z","linkquality":120,"update":{"installed_version":16777241,"latest_version":16777241,"state":"idle"},"update_available":null}'
info 2024-12-22 17:29:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bathroom_temp', payload '{"battery":80,"humidity":58.9,"last_seen":"2024-12-22T16:29:53.536Z","linkquality":196,"temperature":20}'
error 2024-12-22 17:29:53zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-12-22 17:29:53zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-12-22 17:29:53zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-12-22 17:29:53zh:ember: Adapter fatal error: HOST_FATAL_ERROR
info 2024-12-22 17:29:53zh:ember:uart:ash: ASH COUNTERS since last clear:
info 2024-12-22 17:29:53zh:ember:uart:ash: Total frames: RX=754, TX=965
info 2024-12-22 17:29:53zh:ember:uart:ash: Cancelled : RX=0, TX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: DATA frames : RX=749, TX=208
info 2024-12-22 17:29:53zh:ember:uart:ash: DATA bytes : RX=20574, TX=5112
info 2024-12-22 17:29:53zh:ember:uart:ash: Retry frames: RX=3, TX=3
info 2024-12-22 17:29:53zh:ember:uart:ash: ACK frames : RX=0, TX=753
info 2024-12-22 17:29:53zh:ember:uart:ash: NAK frames : RX=0, TX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: nRdy frames : RX=0, TX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: CRC errors : RX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: Comm errors : RX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: Length < minimum: RX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: Length > maximum: RX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: Bad controls : RX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: Bad lengths : RX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: Bad ACK numbers : RX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: Out of buffers : RX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: Retry dupes : RX=3
info 2024-12-22 17:29:53zh:ember:uart:ash: Out of sequence : RX=0
info 2024-12-22 17:29:53zh:ember:uart:ash: ACK timeouts : RX=3
error 2024-12-22 17:30:01zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=206 Len=10].
error 2024-12-22 17:30:01zh:ember:ezsp: ERROR Transaction failure; status=ASH_NCP_FATAL_ERROR. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=206 Len=10].
info 2024-12-22 17:30:23zh:ember:uart:ash: Port closed.
info 2024-12-22 17:30:23zh:ember:uart:ash: ======== ASH stopped ========
info 2024-12-22 17:30:23zh:ember:ezsp: ======== EZSP stopped ========
info 2024-12-22 17:30:23zh:ember: ======== Ember Adapter Stopped ========
error 2024-12-22 17:30:23z2m: Adapter disconnected, stopping

BloodWorkXGaming, 2024-12-22:
Could be related to #23855

BloodWorkXGaming, 2024-12-22:
Or rather related to #23861

Koenkk, 2024-12-22:
Could you try if downgrading your dongle to 7.x fixes the issue?

BloodWorkXGaming, 2024-12-24:
Thanks for the input! It seems to work more reliably now!

Things I changed:
- Downgrade to 7.4.5 (down from 8.0.2; previous 7.x Version I was on was 7.4.4)
- add `rtscts: false` to config (shouldn't change anything as I think false is default)
- increase baudrate to 230400 from 11520 (also flashing the stick with [ncp-uart-hw-v7.4.5.0-zbdonglee-230400.gbl](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.5.0-zbdonglee-230400.gbl))

The most notable change is problaby the increased baudrate.
Since increasing it, I was able to run the OTA Upgrade for a bosch thermostat II, which I was never able to do before (neither on 8.0.2 or 7.x with default baud).

After the update the thermostat error a lot but was fixed by powercycling the thermostat by removing the battery:
```
[2024-12-24 15:12:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bedroom_thermostat', payload '{"battery":100,"boost_heating":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"normal","displayed_temperature":"measured","last_seen":"2024-12-24T14:12:18.138Z","linkquality":192,"local_temperature":19.3,"local_temperature_calibration":-1.2,"occupied_heating_setpoint":5,"operating_mode":"pause","pi_heating_demand":0,"remote_temperature":0,"running_state":"idle","setpoint_change_source":"externally","system_mode":"heat","update":{"installed_version":923014420,"latest_version":923014420,"state":"idle"},"update_available":null,"valve_adapt_process":false,"valve_adapt_status":"success","window_detection":"OFF"}'
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=1](frmRx=1) Added to rxQueue
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=6)
[2024-12-24 15:12:18] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=29 Len=22]
[2024-12-24 15:12:18] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=ZIGBEE_DELIVERY_FAILED], [type=DIRECT], [indexOrDestination=14390], [apsFrame={"profileId":260,"clusterId":513,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":12}], [messageTag=101]
[2024-12-24 15:12:18] debug: 	zh:ember: ~x~> DELIVERY_FAILED [indexOrDestination=14390 apsFrame={"profileId":260,"clusterId":513,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":12} messageTag=101]
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=2](frmRx=2) Added to rxQueue
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=6)
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=3](frmRx=3) Added to rxQueue
[2024-12-24 15:12:18] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=6)
[2024-12-24 15:12:18] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=29 Len=8]
[2024-12-24 15:12:18] debug: 	zh:ember:ezsp: ezspIncomingNetworkStatusHandler(): callback called with: [errorCode=ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY], [target=14390]
[2024-12-24 15:12:18] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "14390".
```

So as it seems these steps have increased the stability of z2m, but I will keep an eye on it and report back if it degrades again!

Thanks for the help and thanks for developing such a cool project!

BloodWorkXGaming, 2024-12-28:
Quick update on my situation:
No more crashes or errors since the changes described above.
So I guess this issue can be closed as fixed.

Maybe a tip to try increasing the baud on similar issues could be added to the wiki to help others?

Thanks!

closed 2024-12-28 by BloodWorkXGaming