Koenkk/zigbee2mqtt #25303: Sonoff ZBminiR2 Inching not working as expected

hendrik001973, 2024-12-23:
### What happened?

[image]
When you enter the time for inching, i.e. 1200 sec, Zigbee2MQTT delivers an Error message:

> error 2024-12-23 14:38:41z2m: Publish 'set' 'inching_control_set' to 'Waschraum Keller' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0x70c59cfffe499e0a/1 customClusterEwelink.protocolData({"data":[1,23,7,128,128,0,2400,9,0,0,2424]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"manufacturerCode":4742,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 2400)'
[image]

According to its manual, the ZBminiR2 accepts values from 0.5 to 3599.5 seconds for inching time. However according to this error message, only values between 0 and 255 are accepted. Additionally, the entered values are doubled (here from 1200 to 2400), so that the highest value you can enter is 127.5 seconds.

Also the inching capabilities are not transferred to Homeassistant as entities.
[image]

### What did you expect to happen?

I could setup an inching time for longer than 127 seconds and that I can activate the Inching via Automations.

### How to reproduce it (minimal and precise)

As described above

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P

### Setup

Addon on HomeAssistant on ThinClient with Intel i5

bobo-cher, 2024-12-26:
I think this is a restriction of zbminir2, which allows values only up to 127.5 seconds. The validation and option description in Z2M seems to be wrong (allowing larger values).

I reported this issue to Sonoff two months ago asking them to increase the limit. The response was that my request was forwarded to product team. There was a firmware update after that but nothing has changed in this regard.

I suggest you to write to Sonoff with a similar request. The more requests they get the higher are the chances for an improvement.

FinnVonB, 2025-02-04:
Just reported same issue to Sonoff Tech - annoying

bobo-cher, 2025-02-10:
@hendrik001973 Thank you! I've tested the fix and it works as expected.

hendrik001973, 2025-02-10:
Thank you very much, but what I have to do, to correct this issue?

FinnVonB, 2025-02-10:
?? sounds promising - but what script ?  will there be a new release soon ?

bobo-cher, 2025-02-10:
This will be included with the next Z2M update (once the pull request is merged into main code base).

KaaNee, 2025-03-02:
@bobo-cher Can you link the Issue# and/or Pull-Request please ? 
So people falling over this thread they will find the change.
Thanks.

bobo-cher, 2025-03-02:
The fix #8790 has been already released with 2.1.2.

closed 2025-03-05 by Nerivec