Koenkk/zigbee2mqtt #25306: [New device support]: Philips Hue Tento White 29,1 cm - white (and black)

RemyyB, 2024-12-23:
### Link

https://www.philips-hue.com/nl-nl/p/hue-white-tento-rond-w-led-plafondpaneel-29-1-cm-wit/8720169330436

### Database entry

{"id":42,"type":"Router","ieeeAddr":"0x001788010e169090","nwkAddr":50425,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"929003822701","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003822701","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10","zclVersion":2,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1734966310156}

### Zigbee2MQTT version

1.42.0 commit: 861cba6

### Comments

I think this one is the same as other colors and sizes, but not 100% certain:
29,1 cm white (this one): 8720169330436
29,1 cm black: 8720169330498

### External definition

```shell
const {philipsLight} = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['929003822701'],
    model: '929003822701',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philipsLight()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Everything seems fine

RemyyB, 2024-12-23:
See also the related PR to the docs:
https://github.com/Koenkk/zigbee2mqtt.io/pull/3350

Koenkk, 2024-12-24:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-24 by Koenkk