Koenkk/zigbee2mqtt #25314: Devices almost immediately stop working when upgrading to 1.42.0-2

MathijsG, 2024-12-24:
### What happened?

When I am on 1.42.0-1 everything works perfectly fine. As soon as I upgrade to 1.42.0-2 the devices work correctly for just a few minutes, but after that the devices stop working. However Zigbee2mqtt is running fine still, so nothing weird in the logs.

I already changed to config path and copied the config to both places as well, but still the same weird behaviour.

Restoring back to 1.42.0-1 solves everything for now. But I rather not keep delaying updating to a newer version.

### What did you expect to happen?

It continues to operate normally.

### How to reproduce it (minimal and precise)

Updating to 1.42.0-2 and then it happens almost immediately.

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

HA Add-on within a virtual Home Assistant VM within Proxmox

aka13-404, 2024-12-24:
I experienced a similar problem today. Restarting the z2m addon in ha did not help, however restarting the full HA setup for whatever reason made the devices start responding again (were throwing (Data request failed with error: 'MAC channel access failure' (225)) before. Currently I experience significant slowdowns (pressing a switch used to turn on a relay in less than a second, now takes about 5), but maybe that will subside after some time.

aka13-404, 2024-12-25:
I decided to unpower everything for 30 minutes for the alleged panic mode, or whatever it is, when the coordinator is down. Seems that everything is back to normal for me, responding fast, and so far no new mac channel acces failures. 

Perhaps that will help you as well.

MathijsG, 2025-01-05:
Had it ocassionally with my older version as well. As it was locking up. And then all of a sudden would do all the actions at once.

Removed the USB-adapter. Flashed its firmware to the newer as well, and currently seems fine. On 2.X as well. Will monitor this, but for now looks fine.

MathijsG, 2025-03-07:
Never occured again, after flashing the new firmware, removing and reinserting the usb device.

closed 2025-03-07 by MathijsG