Koenkk/zigbee2mqtt #25323: Timed off cannot be set to 2 hours

markchalloner, 2024-12-25:
### What happened?

When using onWithTimedOff, the on_time parameter seems to be multiplied by 10 so cannot be set to a value of greater than 6553. 

This seems to be independent of device and was tested with both an [IKEA Tretakt E2204 UK](https://www.zigbee2mqtt.io/devices/E2204.html#ikea-e2204) and [SONOFF ZBMINI](https://www.zigbee2mqtt.io/devices/ZBMINI.html#sonoff-zbmini).

zigbee2mqtt/Test Plug/set

```
{"state" : "ON", "on_time": 10800}
```

```
z2m: Publish 'set' 'state' to 'Test Plug' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0x6cfd22fffe3d0909/1 genOnOff.onWithTimedOff({"ctrlbits":0,"ontime":108000,"offwaittime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 108000)'
```

### What did you expect to happen?

onWithTimedOff can be set with a value between 6554 and 65535

### How to reproduce it (minimal and precise)

Send a MQTT message with an on_time of > 6553 to the topic:

zigbee2mqtt/<FRIENDLY NAME>/set

E.g.:

zigbee2mqtt/Test Plug/set

```
{"state" : "ON", "on_time": 7200}
```

Observe that z2m shows a RangeError and value in the error message is 10x that sent. E.g.:

```
z2m: Publish 'set' 'state' to 'Test Plug' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0xe0798dfffed3b7fc/1 genOnOff.onWithTimedOff({"ctrlbits":0,"ontime":72000,"offwaittime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 72000)'
```

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20230507

### Adapter

Smartlight SLZB-06

### Setup

Add-on on Home Assistant OS in ProxMox VM on Intel NUC

markchalloner, 2024-12-25:
Actually this may not be a bug but my misunderstanding of the ZCL see https://github.com/Koenkk/zigbee-herdsman-converters/pull/2326#issuecomment-792265514:

> OnTime attribute specifies the length of time (in 1/10ths second)

and

> 0x0000 or 0xffff, the device SHALL remain in its current state.

Strongly implies a highest decimal value of 65534 1/10ths second i.e. 6553.4 seconds 😢.

Feel free to close this issue if that's the case.

Koenkk, 2024-12-26:
That's indeed the max by ZCL spec, I will close this.

closed 2024-12-26 by Koenkk