Koenkk/zigbee2mqtt #25328: [New device support]: I would like to add the TUYA TS0601_TZE204_dhotiauw energy meter

JBLSteeve, 2024-12-26:
### Link

https://www.aliexpress.com/item/1005007499685162.html

### Database entry

{"id":70,"type":"Router","ieeeAddr":"0xa4c13898af39ac02","nwkAddr":3947,"manufId":4417,"manufName":"_TZE204_dhotiauw","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�6\u0000/i�6\u0000/i�6\u0000/i�6\u0000/i�6\u0000/i�6\u0000/i�6\u0000/i�6\u0000/i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_dhotiauw","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1735224161134} {"id":71,"type":"Router","ieeeAddr":"0xa4c138baaf283e6d","nwkAddr":34239,"manufId":4417,"manufName":"_TZE204_dhotiauw","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u00067\u0000/i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_dhotiauw","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1735224161574}

### Zigbee2MQTT version

1.42.0

### Comments

I made a file to include it
It's work. You can just check and include it in the next version

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const utils = require('zigbee-herdsman-converters/lib/utils');
const logger = require('zigbee-herdsman-converters/lib/logger');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_dhotiauw',
        },
    ],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Bidirectional energy meter with 80A current clamp',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.ac_frequency(),
        e.voltage(),
        tuya.exposes.powerWithPhase('a'),
        tuya.exposes.powerWithPhase('b'),
        tuya.exposes.currentWithPhase('a'),
        tuya.exposes.currentWithPhase('b'),
        tuya.exposes.powerFactorWithPhase('a'),
        tuya.exposes.powerFactorWithPhase('b'),
        tuya.exposes.energyWithPhase('a'),
        tuya.exposes.energyWithPhase('b'),
        e
            .numeric('update_frequency', ea.STATE_SET)
            .withUnit('s')
            .withDescription('Update frequency')
            .withValueMin(3)
            .withValueMax(60)
            .withPreset('default', 10, 'Default value'),
        e.numeric('timestamp_a', ea.STATE).withDescription('Timestamp at power measure (phase a)'),
        e.numeric('timestamp_b', ea.STATE).withDescription('Timestamp at power measure (phase b)'),
        e.numeric('calibration_ac_frequency', ea.STATE_SET).withDescription('Calibration AC frequency').withValueMin(0.5).withValueMax(1.5).withValueStep(0.01).withPreset('default',1.0,'Default value'),
        e.numeric('calibration_voltage', ea.STATE_SET).withDescription('Calibration voltage').withValueMin(0.5).withValueMax(1.5).withValueStep(0.01).withPreset('default',1.0,'Default value'),
        e.numeric('calibration_power_a', ea.STATE_SET).withDescription('Calibration power A').withValueMin(0.5).withValueMax(1.5).withValueStep(0.01).withPreset('default',1.0,'Default value'),
        e.numeric('calibration_current_a', ea.STATE_SET).withDescription('Calibration current A').withValueMin(0.5).withValueMax(1.5).withValueStep(0.01).withPreset('default',1.0,'Default value'),
        e.numeric('calibration_power_b', ea.STATE_SET).withDescription('Calibration power B').withValueMin(0.5).withValueMax(1.5).withValueStep(0.01).withPreset('default',1.0,'Default value'),
        e.numeric('calibration_current_b', ea.STATE_SET).withDescription('Calibration current B').withValueMin(0.5).withValueMax(1.5).withValueStep(0.01).withPreset('default',1.0,'Default value'),
    ],
    meta: {
        multiEndpointSkip: ['power_factor', 'power_factor_phase_a', 'power_factor_phase_b', 'energy'],
        tuyaDatapoints: [
            [105, 'ac_frequency', tuya.valueConverter.divideBy100],
            [103, 'power_a', tuya.valueConverter.divideBy10],
            [104, 'power_b', tuya.valueConverter.divideBy10],
            [19, 'power', tuya.valueConverter.divideBy10],
            [20, 'voltage', tuya.valueConverter.divideBy10],
            [101, 'current_a', tuya.valueConverter.divideBy1000],
            [102, 'current_b', tuya.valueConverter.divideBy1000],
            [120, 'power_factor_a', tuya.valueConverter.raw],
            [121, 'power_factor_b', tuya.valueConverter.raw],
            [115, 'energy_a', tuya.valueConverter.divideBy100],
            [117, 'energy_b', tuya.valueConverter.divideBy100],
            [122, 'update_frequency', tuya.valueConverter.raw],
            [123, 'calibration_voltage', tuya.valueConverter.divideBy1000],
            [125, 'calibration_current_a', tuya.valueConverter.divideBy1000],
            [126, 'calibration_power_a', tuya.valueConverter.divideBy1000],
            [130, 'calibration_current_b', tuya.valueConverter.divideBy1000],
            [131, 'calibration_power_b', tuya.valueConverter.divideBy1000],
            [124, 'calibration_ac_frequency', tuya.valueConverter.divideBy1000],
        ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I test the power, current and voltage, it's work

Koenkk, 2024-12-28:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

JBLSteeve, 2025-01-08:
I click on your link but i don't know what i have to do after
Can you tell me how to modify the file ?

Koenkk, 2025-01-09:
Added it!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `2CT.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

JBLSteeve, 2025-01-14:
Thanks, 
I add the picture : public/images/devices/TS0601_TZE204_dhotiauw.png

sleepy-salmon, 2025-01-18:
Thank you both for your work on this device! 
For me it is still showing as unsupported, should this already be incorporated or will it be in an upcoming release?

JBLSteeve, 2025-01-25:
Sorry , i'm not an expert.
You can put the definition that i put in this topic inside Z2M and it's work (that how it's work on my side for the moment)

andreacasarin, 2025-01-26:
Hi! I'm testing it and I confirm it works great, thanks!

The only issue I see is that `Energy a` and `Energy b` will not show total energy consumption/production, they just stay at 0 (as a workaround I created an integral sensor via HomeAssistant).

Could this be fixed?

Chnillapoil, 2025-01-28:
> Sorry , i'm not an expert. You can put the definition that i put in this topic inside Z2M and it's work (that how it's work on my side for the moment)

Hi, could you explain where to put the definition ? It seems in version 2 it is different....

SapozhnikUA, 2025-02-02:
> > Sorry , i'm not an expert. You can put the definition that i put in this topic inside Z2M and it's work (that how it's work on my side for the moment)
> 
> Hi, could you explain where to put the definition ? It seems in version 2 it is different....

I created external converter for ZWPM16-2 but I think it has mistakes -  I'm not programmer. I will  try to add code here and share link