Koenkk/zigbee2mqtt #25334: No battery shown TS0203

Jerome69200, 2024-12-26:
### What happened?

After pairong TS0203 door sensor, battery is not shown, even after waiting more than 48 hours.
On the [**device webpage**](https://www.zigbee2mqtt.io/devices/TS0203.html#tuya-ts0203), the battery percentage is available.
When I used the ZHA pluggin, the device shown the battery percentage.

### What did you expect to happen?

I want to see the battery percentage.

### How to reproduce it (minimal and precise)

Just pair the device

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

ZBDongle-E using the ember driver

### Setup

Add-On on Home Assistant OS, using Proxmox

megustaeliphone, 2024-12-29:
Same here, only 'Battery low (false)' is shown.

mad3max3, 2025-02-16:
Same here as well, using a Conbee III with TS0203.
Have the exact problem with my Aqara door sensors also.

megustaeliphone, 2025-02-16:
For some time now, the battery % appears to me.

Jerome69200, 2025-02-26:
I have update Z2M to the lastest version but the battery doesn't appears anymore.
I tried to delete one device and re-paired it but is still not show battery...

littlbee, 2025-04-08:
I just added a Marmitek Sense Mi (door sensor) which is using the same chipset. The battery percentage is reported as unavailable. The only thing from battery perspective that seems to work is reporting on low battery.

cloudbr34k84, 2025-06-01:
i have the moes zigbeen door/window sensor which reports it is as _TZ3000_8yhypbo7 

only exposes these
{
    "battery_low": false,
    "contact": false,
    "last_seen": "2025-06-02T07:35:39+10:00",
    "linkquality": 43,
    "tamper": false
}

[image]