Koenkk/zigbee2mqtt #25336: Frequently: All mains powered devices / routers offline

thogens, 2024-12-27:
### What happened?

Not sure when this started, but it might have to do with a recent update.
After some hours, all my mains powered routers go offline and the zigbee net is dead.
I'm getting following warnings for all routers:
`Warning 2024-12-27 09:47:15z2m: Failed to ping 'Bulb_1' (attempt 1/1, ZCL command 0x842e14fffe4d995c/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":57351,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":92,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,93,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))`

Restarting zigbee2mqtt does not help.
Restarting HA does not help.
Restarting VM does not help.
Restarting server does help and fixes the issue for a while.
Replugging the zigbee coordinator and re-enabling the interface in my VM does help and fixes the issue for a while.

### What did you expect to happen?

The zigbee routers will stay in the mesh and continue to work as expected.

### How to reproduce it (minimal and precise)

Start Zigbee2Mqtt after a fresh start and wait until every device is online.
-- wait --
Suddenly all repeaters go offline. - Mesh dead.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle Plus

### Setup

HomeAssistant Add-on

### Debug log

`Warning 2024-12-27 09:47:15z2m: Failed to ping 'Bulb_1' (attempt 1/1, ZCL command 0x842e14fffe4d995c/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":57351,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":92,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,93,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))`

thogens, 2024-12-27:
I found out that the Zigbee network crashes when I'm manipulating the color temperature and brightness of a big group of bulbs (~20) all at once via an automation. 
Only re-plugging coordinator or server reboot helps.
Any idea?

Koenkk, 2024-12-29:
Can you try setting [`adapter_concurrent`](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#advanced-configuration) to `6` and see if that fixes the issue?

thogens, 2024-12-29:
Well hmmmm... This morning I was finally downgrading the firmware from 20240710 to 20221226 and for now it seems to run much, much better, quicker, more reliable. I'd like to give it some days to see if this fixed it.
But what does the "adapter_concurrent" do and does the firmware version matter?

Koenkk, 2024-12-29:
It reduces the amount of concurrent messages on the network. But if `20221226` works, let monitor that for some days.

thogens, 2025-01-03:
Wanted to give some feedback:
20221226 indeed is working great! There was any single crash since I downgraded.
2 routers got disconnected (for whatever reason) and I waited 2 days to see what will happen. Except the corresponding connected battery devices of those routers going offline, nothing happened. - The net was still stable and responsive.
A simple power cycle brought the routers back.
Additionally some battery powered (Aqara) devices went offline and apparently needed a new battery. (That could be a coincidence).
Until now, the net is super stable with the firmware downgrade and really fast. I'm not sure if I really should try out 2024 again, as is was so bad. On the other hand I'm curious if the concurrent setting would do the trick. WDYT?

53RT, 2025-01-03:
@thogens do you see any updates in the mqtt topics of your router devices?

I noticed something similar but I haven't updated my conbee II adapter. 

In case our problems are related I describe what I have noticed. I think after the update to 1.42.0 the problems started for me. All the router devices appear as availability=offline and cannot be switched using homeassistant. With the z2m UI I can operate the devices but the MQTT topics don't update when I try to switch my devices. So I think somehow the connection with the mqtt broker doesn't work as expected so that the router devices become offline as the ping is not forwarded to the mqtt broker. If I publish a set command manually to one of the devices it switches them. 

After a restart of my setup and repairing some devices they become online again and I can turn them on and off but after the 10 minutes they are marked offline again and don't respond.

thogens, 2025-01-04:
Well... with downgrading the firmware of my Sonoff stick, all works so much better and devices are staying online like they should.
I notice that lqi is pretty low on a couple of devices, although there are good repeaters nearby and they are connected to those. Not sure how lqi is calculated, but hey - it's working for now.

dariosun, 2025-01-21:
Had this twice now, too!
Also recently updated to 20240710, also have the Sonoff Dongle P.
Also, only server restart helps, while VM restart doesn‘t.
I‘m on Z2M 2.0.
Will monitor. If it happens again, I will downgrade again, too. Thank you so much for posting!

closed 2025-03-30 by github-actions[bot]