Koenkk/zigbee2mqtt #25351: ZigBee network dead after upgrading to latest-dev (#120a5b4)

ikarisan, 2024-12-28:
### What happened?

Hi!

After upgrading "latest-dev" from 08. december 2024 to the todays version (28. december 2024) my complete zigbee network is broken. With loglevel debug I can see incomming ZigBee messages but nothing happens. No message is send to MQTT and the WebUI is also broken. :(  After downgrading to 1.42.0 the network and MQTT is working, again. 

### Zigbee2MQTT version

latest-dev

### Adapter firmware version

20230507

### Adapter

Sonoff with zStack3x0

### Setup

plain Zigbee2MQTT as docker image on Lenovo ThinkCentre

Koenkk, 2024-12-29:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

ikarisan, 2024-12-29:
> Could you provide the debug log of this?

Yes, I can. Coincidentally, I already had the log level set to “debug”. :) 
But that's almost 4MB. I have to sort it out first.

ikarisan, 2024-12-29:
The logfile after upgrading:

[log-ce4bdfa3.log](https://github.com/user-attachments/files/18270080/log-ce4bdfa3.log)

Koenkk, 2024-12-30:
In this log z2m is still starting, could you provide the debug log from starting z2m until you observe this:

> With loglevel debug I can see incomming ZigBee messages but nothing happens. No message is send to MQTT

ikarisan, 2024-12-30:
> In this log z2m is still starting, could you provide the debug log from starting z2m until you observe this:

I can post the rest of the 4MB log files. But they all look the same. :(

ikarisan, 2024-12-30:
[log_2024-12-28-1.42.0_RELEASE.log](https://github.com/user-attachments/files/18277758/log_2024-12-28-1.42.0_RELEASE.log)
[log_2024-12-28.16-14-1.42.0-dev.log](https://github.com/user-attachments/files/18277759/log_2024-12-28.16-14-1.42.0-dev.log)
[log_2024-12-28.15-49-1.42.0-dev.log](https://github.com/user-attachments/files/18277760/log_2024-12-28.15-49-1.42.0-dev.log)
[log_2024-12-28.15-41-1.42.0-dev.log](https://github.com/user-attachments/files/18277761/log_2024-12-28.15-41-1.42.0-dev.log)
[log_2024-12-28.14-19-1.42.0-dev.log](https://github.com/user-attachments/files/18277762/log_2024-12-28.14-19-1.42.0-dev.log)

ikarisan, 2025-01-01:
> In this log z2m is still starting, could you provide the debug log from starting z2m until you observe this:

What I wanted to say is that there are no more entries. I have pressed Zigbee buttons, but nothing has happened. I tried to trigger ZigBee actions via Amazon Alexa/NodeRed, nothing happened. The system remains in the state that can be seen in the log files.

Koenkk, 2025-01-03:
Could you provide your `configuration.yaml` and `devices.yaml` (in case you have one)?

ikarisan, 2025-01-04:
Here it is. 

[configuration.yaml.txt](https://github.com/user-attachments/files/18307269/configuration.yaml.txt)

I see that there is now an configuration_backup_v2.yaml file with the following diffs: 
[config_patch.txt](https://github.com/user-attachments/files/18307298/config_patch.txt)

I'm just wondering why the new v3 configuration file works with the “old” stable version.

ikarisan, 2025-01-04:
Tried to update to the 2.0.0 release with the same result. No WebUI and no functionality.

Koenkk, 2025-01-05:
The config should be on v4, can you remove all `icon` lines from your `configuration.yaml`?

ikarisan, 2025-01-05:
I am on v4 now nut still getting this:

`[2025-01-05 16:05:20] error:    z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'close')
    at /app/lib/extension/frontend.ts:132:52
    at new Promise (<anonymous>)
    at Frontend.stop (/app/lib/extension/frontend.ts:132:15)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.stop (/app/lib/controller.ts:233:9)
    at stop (/app/index.js:172:5)
    at process.handleQuit (/app/index.js:180:13))
`

ikarisan, 2025-01-06:
Interesting. During my tests, I reordered the entries in the yaml file. It still doesn't work (no frontend, no ZigBee function - I pressed "TuYa_Vierfach_Schalter" a multiple times), but there seems no longer be an error entry in the log file. Started the container and stopped it 6 minutes after. 

Log: [log_2025-01-06_2.0.0.log.txt](https://github.com/user-attachments/files/18322800/log_2025-01-06_2.0.0.log.txt)
New config: [configuration_2025-01-06_resorted.yaml.txt](https://github.com/user-attachments/files/18322901/configuration_2025-01-06_resorted.yaml.txt)

closed 2025-03-15 by github-actions[bot]