Koenkk/zigbee2mqtt #25357: Tuya temperature sensor TS0201 _TZ3000_akqdg6g7 doesn't report battery level

CrazyVito11, 2024-12-28:
### What happened?

I recently got a couple of temperature sensors from AliExpress and they seem to work fine, they report the temperature properly. But it doesn't seem like it's able to report the battery level and the image/description is also incorrect for the device.

[image]
[image]

Physically the device looks exactly like a [IH-K009](https://www.zigbee2mqtt.io/devices/IH-K009.html).

### What did you expect to happen?

That the sensor was able to report it's battery level and show the correct image/description in Z2M.

### How to reproduce it (minimal and precise)

No special steps required besides pairing of the device.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20210708

### Adapter

zStack3x0 (SONOFF ZB Dongle-P)

### Setup

Via Docker on a x86-64 Debian host machine

### Debug log

[z2m.log](https://github.com/user-attachments/files/18267056/z2m.log)

GigiPompieru, 2025-01-14:
I just got 2 [IH-K009](https://www.zigbee2mqtt.io/devices/IH-K009.html#tuya-ih-k009 ) and have the same problem with battery not being reported/displayed.

The 'funny' thing is that I have 3 older [WSD500A](https://www.zigbee2mqtt.io/devices/WSD500A.html#tuya-wsd500a) which are also TS0201 and they work fine, or at least they show the battery value, even if incorrectly.

I'm running Z2M 2.0.0-2 via SLZB-06 on LAN.

LeonCame, 2025-01-22:
Same problem here, in z2m
            the battery - % - flashes red,
and a long time later becomes
            - ? -

it is identified by z2m as:

Zigbee Model: TS0201
Zigbee Manufacturer: _TZ3000_akqdg6g7
Description: Temperature & humidity sensor

The one I have: https://www.aliexpress.com/item/1005007755890938.html
is not round but rectangular like this: https://www.zigbee2mqtt.io/devices/TH02Z.html

Problem: Battery level is never reported.

bluemorphe, 2025-01-25:
same probleme

bluemorphe, 2025-01-25:
#koenkk help

Miky028, 2025-02-07:
bump, pls

LeonCame, 2025-02-28:
I have a Tuya IH-K009 and I solved the problem as follows:

in the Bind tab I found that PowerCfg was checked for the coordinator, so I simply removed this link, and shortly after (but it might take longer, be patient) the battery percentage finally became available.

Reminder: to remove a Bind from a "passive" device, you have to briefly tickle the reset button of the device, in order to wake it up while you delete (uncheck the bind).

GigiPompieru, 2025-03-01:
> I have a Tuya IH-K009 and I solved the problem as follows:
>
> in the Bind tab I found that PowerCfg was checked for the coordinator, so I simply removed this link, and shortly after (but it might take [...]

I just check both my devices and none have any bindings configured.

Miky028, 2025-03-04:
> I just check both my devices and none have any bindings configured.

Me neither.
One of my devices started showing battery status. I suspect it shows battery status when it is less than 100%.

puttemanp, 2025-03-07:
> I have a Tuya IH-K009 and I solved the problem as follows:
> 
> in the Bind tab I found that PowerCfg was checked for the coordinator, so I simply removed this link, and shortly after (but it might take longer, be patient) the battery percentage finally became available.

I had no bindings, but I created one to one of my light bulbs. 
When I deleted it again my sensor started reporting battery status.

Thx for the workaround

rolandas-valantinas, 2025-03-24:
Does yours's actually have screen? Because mine is not recognised correctly https://github.com/Koenkk/zigbee2mqtt/issues/25018#issuecomment-2749629416 but it started showing battery after multiple resets and swapping batteries in and out.

BePalm2022, 2025-03-27:
still null for battery: any hints how rto get the battery % into HA?
also bind is not working...

 {
    "battery": null,
    "device": {
        "applicationVersion": 70,
        "dateCode": "",
        "friendlyName": "Temperature07_cellar",
        "hardwareVersion": 1,
        "ieeeAddr": "xxxxx",
        "manufacturerID": 4417,
        "manufacturerName": "_TZ3000_dowj6gyi",
        "model": "IH-K009",
        "networkAddress": 27001,
        "powerSource": "Battery",
        "stackVersion": 0,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "humidity": 39,
    "last_seen": "2025-03-27T12:53:38+01:00",
    "linkquality": 47,
    "temperature": 19.6,
    "voltage": null
}

BePalm2022, 2025-03-27:
I fi connect them directly with the with the coordinator (Sonoff_Zigbee_3.0_USB_Dongle_Plus) connected on Raspi i get the Battery %, but if they are connected over a router in my net, like a ZB-dongle-E, or Lights,  I get Null for battery....
I have also other Temp sensors, they work fine with all values, only this type  IH-K009 has issue

GigiPompieru, 2025-05-29:
yea, this is still not fixed.

CrazyVito11, 2025-09-13:
Just checked with Z2M version `2.6.1` and the bug is still present.

[image]