Koenkk/zigbee2mqtt #25358: SZ-T04 ( _TZE200_qrztc3ev) does not report any data after joining the network

heisenberg2980, 2024-12-28:
### What happened?

Temperature and humidity sensor SZ-T04 (_TZE200_qrztc3ev) joins the zigbee network but then it does not report any value, so it doesn´t report the temperature or humidity

### What did you expect to happen?

Sensor should report temperature and humidity after joining the network

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

 20230507

### Adapter

Sonoff zigbee 3.0

### Setup

Home Assistant Add-on

heisenberg2980, 2025-01-12:
Thanks @JKGHPO but I just checked and realised that the device started to reported data a couple of days ago. I guess this is thanks to the latest z2m update (I say "guess" because I´m not sure when I updated it, but I assume it was around the time when the issue was fixed?).

Anyway I´ll close this issue as it seems to be working now.

Phoenixxx3, 2025-04-11:
@heisenberg2980 Is this the device with display and clock? Do you use a custom quirk? I have such a device and it does not report any values.