Koenkk/zigbee2mqtt #25361: Lutron Aurora dimmer knob (Z3-1BRL) has missing battery functionality

fractalcounty, 2024-12-29:
### What happened?

Despite [being supported in zigbee2mqtt](https://www.zigbee2mqtt.io/devices/Z3-1BRL.html) and its hardware itself supporting the functionality, the Lutron Aurora Z3-1BRL dimmer knob does not expose or report remaining battery % in zigbee2mqtt.

### What did you expect to happen?

The Z3-1BRL should expose and report battery usage % in the zigbee2mqtt frontend automatically upon device adoption similar to other battery-powered devices like the [Philips Hue dimmer switch](https://www.zigbee2mqtt.io/devices/929002398602.html#philips-929002398602). Additionally, a battery sensor should be automatically created in HomeAssistant. 

[image]

A similar issue was outlined 3 years ago in #9736, but its scope was limited to creating a custom converter for the Z3-1BRL to manually expose battery usage. Despite being marked as completed, no out-of-the-box functionality relating to battery usage exists for the Z3-1BRL to this day (despite the device fully supporting it). Users have continued to observe this problem [as recently as this week](https://www.reddit.com/r/homeassistant/comments/1hlhl5c/battery_reporting_for_lutron_aurora_button_z31brl/), so it appears to just be a case of missing functionality that's slipped under the radar over the years. 

I suspect this may have something to do the fact that the Z3-1BRL seems to only want to communicate with zigbee2mqtt while in an "awake" state (i.e after being physically pressed or rotated), as outlined in a similar (now stale) issue I opened for the Z3-1BRL in 2021 (see #20668). If this is a case, perhaps battery reporting could occur whenever the device is in its active state? 

### How to reproduce it (minimal and precise)

1. Factory reset the Z3-1BRL dimmer switch
2. Permit joining in zigbee2mqtt and wait for device adoption
3. With the Z3-1BRL in its awake state, bind the `PowerCfg` cluster's `batteryPercentageRemaining` attribute to the coordinator endpoint in the device's "Reporting" tab

### Zigbee2MQTT version

1.42.0-dev commit: [c3b4628](https://github.com/Koenkk/zigbee2mqtt/commit/c3b4628)

### Adapter firmware version

20240710

### Adapter

SMLight SLZB-06 (zStack3x0)

### Setup

Add-on in Home Assistant OS VM via Proxmox VE on a Intel N100 mini PC 

### Debug log

N/A

ajma, 2024-12-29:
I finally got rid of my hue hub and moved it all to zigbee2mqtt and this is the last issue I'm dealing with. I have three of these buttons and I'm not able to figure out how to get battery readings

x10, 2025-01-03:
I also recently started using these dimmers and it is the only device that doesn't report its battery level in my set up.

Koenkk, 2025-01-03:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/5ffe64800400da15c87ae0bc135d5132): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- reconfigure the device via the frontend (yellow refresh button), make sure to wake up the device right before doing so (by pressing a button on it)
- check if issue is fixed

fractalcounty, 2025-01-04:
`ext_converter.js` seems to be working and battery is being reported! Thank you.

Also, no idea if this is relevant, but saving the script at `/config/zigbee2mqtt/ext_converter.js` and adding it as an external converter in the adjacent `configuration.yaml` file had no effect after restarting Home Assistant & zigbee2mqtt. Even adding scripts with arbitrary names as external converters in `configuration.yaml` produced zero errors or relevant debug messages in the logs. I was only able to add it by sending a MQTT message to `zigbee2mqtt/bridge/request/converter/save` with the payload containing the script contents, which actually ended up saving it to `/config/zigbee2mqtt/external_converters/converter.js` instead. I most definitely just missed something in the docs or misconfigured something, though.

closed 2025-01-05 by Koenkk