Koenkk/zigbee2mqtt #25362: Unable to start zigbee2mqtt : error when starting zigbee-herdsman failed to connect to the adapter

epicurean70, 2024-12-29:
### What happened?

Found out recently when i logged into HA that there is a z2m error. Last time I checked all was well
z2m adapter is an SLZ06 (connected via POE). I have made no changes to HA, z2m configuration

Starting Zigbee2MQTT without watchdog.
[2024-12-29 18:05:28] info: 	z2m: Logging to console, file (filename: log.log)
[2024-12-29 18:05:28] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #861cba6)
[2024-12-29 18:05:28] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-29 18:05:28] info: 	zh:zstack:znp: Opening TCP socket with 192.168.0.49:6638
[2024-12-29 18:05:28] info: 	zh:zstack:znp: Socket connected
[2024-12-29 18:05:28] info: 	zh:zstack:znp: Socket ready
[2024-12-29 18:05:28] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-12-29 18:05:29] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-12-29 18:05:47] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-29 18:05:47] error: 	z2m: Failed to start zigbee
[2024-12-29 18:05:47] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-29 18:05:47] error: 	z2m: Exiting...
[2024-12-29 18:05:48] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)
Using '/app/data' as data directory

 

### How to reproduce it (minimal and precise)

The errors keep recycling. Restarting the z2m docker container makes no difference. 

### Zigbee2MQTT version

1.42

### Adapter firmware version

 20221226

### Adapter

SLZ06

### Setup

z2m is a docker container, as is the HA container  and the mqtt container. Ubuntu server

### Debug log

[_zigbee2mqtt_logs.txt](https://github.com/user-attachments/files/18269099/_zigbee2mqtt_logs.txt)

chris-1243, 2024-12-29:
Which version of SLZB06 ? 
  - SLZB06
  - SLZB06M

Could you provide your `configuration.yaml` ?

epicurean70, 2024-12-30:
It would be the SLZB06

epicurean70, 2024-12-30:
configuration.yaml

homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.0.16
serial:
  port: tcp://192.168.0.49:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
advanced:
  transmit_power: 20
  log_level: info
  ikea_ota_use_test_url: true
experimental:
  output: json
  new_api: true
frontend:
  port: 8085
  host: 0.0.0.0
devices:
  '0x00124b0021b77f0a':
    friendly_name: zzh_router
    retain: false
  '0x00124b00199de581':
    friendly_name: cc2530_wall_router_diningroom
    retain: false
  '0x00124b00199decc0':
    friendly_name: cc2530_usb_router1_playroom
    retain: false
  '0x90fd9ffffe3112d7':
    friendly_name: TRADFRI_E14_bulb_2
  '0x90fd9ffffe31fbf8':
    friendly_name: TRADFRI_E14_bulb_1
  '0xec1bbdfffea316ee':
    friendly_name: TRADFRI_E1810_Remote
  '0x00158d00042d8384':
    friendly_name: Aqara_ContactSensor_DiningRoom_Side
    debounce: 1
  '0x00158d0004773327':
    friendly_name: Aqara_ContactSensor_DiningRoom_Back
    debounce: 1
  '0x00158d00047731cb':
    friendly_name: Aqara_ContactSensor_KitchenWindow_Right
    debounce: 1
  '0x00158d00047c6609':
    friendly_name: Aqara_ContactSensor_KitchenWindow_Left
    debounce: 1
  '0x00158d00047c6652':
    friendly_name: Aqara_ContactSensor_Backgate
    debounce: 1
  '0x00158d0004773402':
    friendly_name: Aqara_ContactSensor_Storeroom_Rear
    debounce: 1
  '0x00158d00047d0c0c':
    friendly_name: Aqara_ContactSensor_LivingRoom_Side
    debounce: 1
  '0x00158d00047c6715':
    friendly_name: Aqara_ContactSensor_LivingRoom_Front
    debounce: 1
  '0x00158d00049da824':
    friendly_name: Aqara_ContactSensor_Playroom_Left
    debounce: 1
  '0x00158d00053ef481':
    friendly_name: Aqara_ContactSensor_Playroom_Top
    debounce: 1
  '0x00158d00049fb3ea':
    friendly_name: Aqara_ContactSensor_Playroom_Bottom
    debounce: 1
  '0x00158d000464a625':
    friendly_name: Aqara_MovementSensor_LivingRoom
    retain: false
  '0x00158d000465fbae':
    friendly_name: Aqara_MovementSensor_DiningRoom
    retain: false
  '0xa4c13830ced2af65':
    friendly_name: Tuya_Lightswitch2G_MasterBR_Right
    availability: true
  '0xa4c138444201a401':
    friendly_name: Tuya_Lightswitch2G_MasterBR_Left
    availability: true
  '0xa4c138a47caaadac':
    friendly_name: Tuya_Lightswitch_Storeroom
    availability: true
  '0xa4c1387760bb2e8e':
    friendly_name: Tuya_Lightswitch2G_L2Landing
    availability: true
  '0xa4c1387ffbe10164':
    friendly_name: Tuya_Lightswitch_Playroom
    availability: true
  '0xa4c138cb5ffa5b41':
    friendly_name: Tuya_Lightswitch_Level1Toilet
    availability: true
  '0xa4c138b74e9bda55':
    friendly_name: Tuya_Lightswitch_AtticToilet
    availability: true
  '0xa4c138c236c341f0':
    friendly_name: Tuya_Boilerswitch_MB
    availability: true
  '0xa4c138656d96a8e0':
    friendly_name: Tuya_Boilerswitch_Landing
    availability: true
  '0xa4c13867d59f446b':
    friendly_name: Tuya_Lightswitch3G_MasterBR_Toilet
  '0xa4c138984da941ff':
    friendly_name: Useelink_powerstrip_LivingRoom
  '0xa4c138b76e04e1b6':
    friendly_name: Useelink_powerstrip_Attic
  '0xa4c138196e906741':
    friendly_name: Tuya_Lightswitch_StudyToilet
  '0x00158d000400e89e':
    friendly_name: Aqara_Leaksensor_Kitchen
  '0xa4c138b8477739e7':
    friendly_name: Tuya_Lightswitch_KidBRToilet
  '0x00158d0005cf24bf':
    friendly_name: Aqara_MovementSensor_2ndfloorlanding
    retain: false
  '0x00158d00045da42d':
    friendly_name: Aqara_MovementSensor_Atticlanding
    retain: false
  '0x54ef44100053c6a7':
    friendly_name: Aqara_PresenceSensor_LR
  '0xf082c0fffefc452d':
    friendly_name: Tuya_IR_remote_1
  '0xb4e3f9fffe6c3bd3':
    friendly_name: Tredansen_leftblind_study
    availability: true
  '0xb4e3f9fffec13c78':
    friendly_name: Tradfri_repeater_blinds_study
  '0x2c1165fffea5e8c2':
    friendly_name: Tredansen_rightblind_study
    availability: true
  '0x58a69c82342d585a':
    friendly_name: Xiaomi_temphumidity_sensor
  '0x8cf681fffe2a5845':
    friendly_name: Tradfri_remote_blinds_study
  '0x0cae5ffffe3b49dd':
    friendly_name: Tradfri_LEDcontroller_Kitchen
  '0xdc8e95fffe2562e6':
    friendly_name: HZC_dimmer_2ndfloorlanding
  '0x00124b001f2b2630':
    friendly_name: Backyard_gban_router
    retain: false
blocklist:
  - '0x01124b001c2e7569'
  - '0x01124b001c2e5840'
  - '0x000d6f000e858559'
  - '0xa4c138e4dc87248d'
  - '0xa4c1385186728587'
ota:
  zigbee_ota_override_index_location: my_index.json
external_converters:
  - ZCGKD2.js

chris-1243, 2024-12-30:
Everything looks fine... It might be related to #21157

epicurean70, 2024-12-31:
zigbee2mqtt came back working after I unplugged and replugged the SLZB06 again with the POE cable

thank you

closed 2025-03-10 by github-actions[bot]