Koenkk/zigbee2mqtt #25363: Philips hue dimmer 929002398602 (RWL022) incorrect triggerred action

ardhipoetra, 2024-12-29:
### What happened?

I have a Philips hue dimmer 929002398602 (RWL022) fw version 2.44.0_hBB3C188. I have been able to set it up by bind it with Philips Enrave through the web interface. So far it was good for power on/off and dim level.

However, I couldn't get the button marked as 'hue' to work. In the triggered action, it shows "off_press" instead of "recall_*". This might be related to https://github.com/Koenkk/zigbee-herdsman-converters/pull/3815 too. I cannot get any unique "group id" for my remote dimmer. I have set up scene 1 and 2 for my bulb (Enrave).

Instead, I have `data '{"button":4,"time":0,"type":0,"unknown1":3145728,"unknown2":0}` in my debug log. I tried updating the remote firmware, but it also didn't work. I couldn't update for whatever reason.

I wonder if this is a bug? Or do I need some specific quirk to make it work? I have attached the relevant debug output below.

Also, a bit of OOT: Somehow I often get a lot of "timeout" interacting with the dimmer, despite I always press the button whenever I want to do something (in order to wake it up).

### What did you expect to happen?

Expected that the bind feature properly bind scene switch from dimmer v2 to the bulb scene.

### Zigbee2MQTT version

1.42.0 

### Adapter firmware version

7.4.4 [GA]

### Adapter

ember

### Setup

plain z2m on rpi3b using ZBDongle-E

### Debug log

```
....
[2024-12-29 09:31:10] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":64512,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":26}], [packetInfo:{"senderShortId":47806,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":255,"lastHopRssi":-36,"lastHopTimestamp":0}], [messageContents=1d0b1001000400003000210000]
[2024-12-29 09:31:10] debug: 	zh:controller: Received payload: clusterID=64512, address=47806, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":1,"commandIdentifier":0},"payload":{"button":4,"unknown1":3145728,"type":0,"unknown2":0,"time":0},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2024-12-29 09:31:10] debug: 	z2m: Received Zigbee message from '0x001788010b78cd35', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
[2024-12-29 09:31:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x001788010b78cd35', payload '{"action":"off_press","battery":100,"linkquality":255}'
[2024-12-29 09:31:10] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-12-29 09:31:10] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=5 frmTx=5)
[2024-12-29 09:31:10] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=7](frmRx=7) Added to rxQueue
[2024-12-29 09:31:10] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=5)
[2024-12-29 09:31:10] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=76 Len=38]
[2024-12-29 09:31:10] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":64512,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":27}], [packetInfo:{"senderShortId":47806,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":255,"lastHopRssi":-36,"lastHopTimestamp":0}], [messageContents=1d0b1002000400003002210100]
[2024-12-29 09:31:10] debug: 	zh:controller: Received payload: clusterID=64512, address=47806, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4107,"transactionSequenceNumber":2,"commandIdentifier":0},"payload":{"button":4,"unknown1":3145728,"type":2,"unknown2":0,"time":1},"command":{"ID":0,"parameters":[{"name":"button","type":32},{"name":"unknown1","type":34},{"name":"type","type":32},{"name":"unknown2","type":32},{"name":"time","type":32},{"name":"unknown2","type":32}],"name":"hueNotification"}}
[2024-12-29 09:31:10] debug: 	z2m: Received Zigbee message from '0x001788010b78cd35', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":1,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
[2024-12-29 09:31:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x001788010b78cd35', payload '{"action":"off_press_release","battery":100,"linkquality":255}'

....
```

daniel-gll, 2025-02-16:
Same issue with 2.1.1. The "hue" button doesn't work after binding and I get "off_press" instead of "recall_*"

daniel-gll, 2025-02-16:
Related: 

- https://github.com/Koenkk/zigbee-herdsman-converters/pull/3815
- https://github.com/Koenkk/zigbee2mqtt/issues/22231
- https://github.com/Koenkk/zigbee2mqtt/issues/14379

wduda, 2025-03-06:
I con confirm this still exists, I also have the same Philipps Dimmer 929002398602 (RWL022). Firmware version is [2.77.39](https://www.philips-hue.com/en-us/support/release-notes/accessories)

`debug 2025-03-06 21:29:52z2m: Received Zigbee message from 'Bathroom Switch', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":4,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2025-03-06 21:29:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bathroom Switch', payload '{"action":"off_press","battery":100,"linkquality":255,"update":{"installed_version":33574183,"latest_version":33574183,"state":"idle"}}'
info 2025-03-06 21:29:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bathroom Switch/action', payload 'off_press'
debug 2025-03-06 21:29:52zh:ezsp:uart: <-- [22]
debug 2025-03-06 21:29:52zh:ezsp:uart: <-- [7d33b1a90d2aefd22f6cc02babdd8549639d26]
debug 2025-03-06 21:29:52zh:ezsp:uart: <-- [50bc163c7e]
debug 2025-03-06 21:29:52zh:ezsp:uart: <-- DATA (2,2,0): 2213b1a90d2aefd22f6cc02babdd8549639d2650bc163c7e
debug 2025-03-06 21:29:52zh:ezsp:uart: --> ACK (3)
debug 2025-03-06 21:29:52zh:ezsp:uart: --> [83401b7e]
debug 2025-03-06 21:29:52zh:ezsp:uart: <-- ACK (2): 2213b1a90d2aefd22f6cc02babdd8549639d2650bc163c7e
debug 2025-03-06 21:29:52zh:ezsp:ezsp: <== Frame: 5190015900fa6076f88a0e01881700ffd301fb51
debug 2025-03-06 21:29:52zh:ezsp:ezsp: <== 0x59: {"_cls_":"incomingRouteRecordHandler","_id_":89,"_isRequest_":false,"source":24826,"longId":{"type":"Buffer","data":`

wduda, 2025-03-06:
I now realize that the incorrect action is only shown in log if Bind is not set up correctly. After binding the Scenes cluster to the coordinator in the bind settings of the dimmer I was able to get the correct actions to show up in the log. Everything but Scenes cluster was unbound from coordinator, then Scenes cluster was bound.

```
info 2025-03-06 22:00:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bathroom Switch', payload '{"action":"recall_0","battery":100,"linkquality":255,"update":{"installed_version":33574183,"latest_version":33574183,"state":"idle"}}'
info 2025-03-06 22:00:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bathroom Switch/action', payload 'recall_0'
```

wduda, 2025-03-06:
After setting up a group with the id obtained that way I created the group, set scenes 0 and 1 for that group, bound the dimmer to this group incl. scenes cluster and now the Hue button is switching between scenes correctly by using recall_0 and recall_1.

To [ardhipoetra](https://github.com/ardhipoetra), consider binding Scenes cluster between Dimmer and Coordinator and reading the debug log again.

[image]

daniel-gll, 2025-03-10:
> I now realize that the incorrect action is only shown in log if Bind is not set up correctly. After binding the Scenes cluster to the coordinator in the bind settings of the dimmer I was able to get the correct actions to show up in the log. Everything but Scenes cluster was unbound from coordinator, then Scenes cluster was bound.
> 
> ```
> info 2025-03-06 22:00:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bathroom Switch', payload '{"action":"recall_0","battery":100,"linkquality":255,"update":{"installed_version":33574183,"latest_version":33574183,"state":"idle"}}'
> info 2025-03-06 22:00:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bathroom Switch/action', payload 'recall_0'
> ```
> 
> This also allowed me to find out the groupId for the group with scenes which Philips wants me to use.

After playing with the Scenes cluster binding I was able to get the correct action recall_x. However, I still do no see the groupId, not sure what else I can try:

`info 2025-03-10 20:16:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Mando Hue', payload '{"action":"recall_0","battery":100,"last_seen":"2025-03-10T20:16:35+01:00","linkquality":255,"update":{"installed_version":33574183,"latest_version":33574183,"state":"idle"}}'
info 2025-03-10 20:16:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Mando Hue/action', payload 'recall_0'`

wduda, 2025-03-11:
> info

You will not see the group ID in an "info" level message. You need to activate debug level logging in the settings of Z2M, then click save, then restart z2m with the red button top right, then you will get messages in the log that are debug level:

```debug 2025-03-11 11:00:52z2m: Received Zigbee message from 'Dimmer - Bathroom', type 'commandRecall', cluster 'genScenes', data '{"groupid":14719,"sceneid":0}' from endpoint 1 with groupID 14719```

Only the debug level message will have the groupID you seek.

daniel-gll, 2025-03-11:
Thanks a lot, I was indeed missing the debug level logs... 
Thanks for the PR too, I was going to do that as well :)

wduda, 2025-03-12:
It would be great if @ardhipoetra could report back if the steps worked for them as well, but I feel like this issue can be closed since with the steps now in the documentation the issue can be resolved by users themselves.

closed 2025-05-20 by github-actions[bot]