Koenkk/zigbee2mqtt #25365: [Wrong device]: Zbeacon TH01 detects as SONOFF SNZB-02

mrmaximas, 2024-12-29:
### Link

https://aliexpress.ru/item/1005005323872621.html

### Model

TH01

### Description

Indoor Temperature And Humidity Sensor

### Vendor

Zbeacon

### Picture (link)

https://ae04.alicdn.com/kf/S4a695352b8c24aa18dfb71046f392d16L.jpg

### Database entry

{"id":99,"type":"EndDevice","ieeeAddr":"0xa4c138191f783281","nwkAddr":42592,"manufId":4742,"manufName":"Zbeacon","powerSource":"Battery","modelId":"TH01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1,32,1026,1029],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TH01","manufacturerName":"Zbeacon","powerSource":3,"zclVersion":3,"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":30,"maxRepIntval":300,"repChange":20},{"cluster":1029,"attrId":0,"minRepIntval":30,"maxRepIntval":300,"repChange":100},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0}],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":861431685},"lastSeen":1735404635883,"checkinInterval":3600}

### Notes

Device works ok, but wrong manufacturer and picture.
[image]

Koenkk, 2024-12-30:
Fixed!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `TH01.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-12-30 by Koenkk

mrmaximas, 2024-12-31:
> Fixed!
>
> Could you also submit a picture for the docs? This can be done by:
>
> 1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
> 2. Go to the `public/images/devices` [...]

done

mrmaximas, 2025-01-03:
@Koenkk no changes after 2.0.0 release... Re-pair does not help.
[image]

mrmaximas, 2025-01-05:
for now device type is ok but exposes page from another device, btw state is ok
{
    "battery": 100,
    "humidity": 6.9,
    "last_seen": "2025-01-05T17:48:50+03:00",
    "linkquality": 105,
    "temperature": 45.3,
    "voltage": 3000
}
[image]
[image]

mrmaximas, 2025-01-07:
@Koenkk https://www.zigbee2mqtt.io/devices/TH01.html is not a contact sensor, it's TH sensor.

Koenkk, 2025-01-07:
Fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

oxan, 2025-01-12:
@mrmaximas how did you pair this device?

mrmaximas, 2025-01-12:
@oxan yes!
[image]
[image]