Koenkk/zigbee2mqtt #25367: TS0504B requires turning on twice to work

dziedzic99, 2024-12-30:
### What happened?

I have one of TS0504B which vary in form factor.
Mine was bought [here](https://allegro.pl/oferta/inteligentny-kontroler-led-rgbw-5-24v-15a-zigbee-3-0-do-hue-tuya-smart-12261657810) and looks exactly like the picture.
The problem occurs randomly but can be manually triggered by turning the state to off twice (i.e. turning off when it's off already).
Then a single turn on command makes the LEDs brighten up for _transition time_ and then immediately turns off.
To turn it on properly it requires to send the on command twice. This is impossible to do from HA level as apparently it won't send on command when it's on already.

### How to reproduce it (minimal and precise)

Faulty state can be manually triggered by turning the state to off twice (i.e. turning off when it's off already).
Then try turning on by a single on command.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20220928

### Adapter

SONOFF ZB Dongle-P Zigbee 3.0 USB Dongle Plus

### Setup

HA addon on Rpi4, Mosquitto broker

dziedzic99, 2024-12-30:
Apparently @CodingGhost had the same problem with a very similar device back in 2023
See [here](https://github.com/Koenkk/zigbee2mqtt/issues/5962#issuecomment-1610095552_)

dziedzic99, 2024-12-30:
I have already tested a workaround and sending _on_ command twice in quick succession doesn't work. At least 0.5s must pass between the commands. Obviously I can have a HA automation to retry sending _on_ command after said 0.5s but it results in a visible blink.

**Proposed solution:**
Hard ban on sending the off command twice, i.e. override the _off_ command to check whether it's already off and don't send the _off_ command then.

It would be nice to also add a second _on_ command after a delay so that for whatever reason the device enters the broken state it has a predictable way to reset but that obviously opens a can of worms of delaying sending a command in the converters.

CodingGhost, 2024-12-30:
I also have never found a satisfying solution to this, maybe creating an extra converter with the fixes in place like ts0503_fix or something so people can choose which one to use would be a solution.
If someone points me in a direction where to start I could try it myself too if it isnt too conplicated.

WhoTheHeck, 2025-01-20:
I’m experiencing similar issues with a TS0503B on the latest version of z2m. Specifically, adjusting the light while it’s still in transition makes its behavior quite unpredictable. Sometimes it turns back off after being turned on mid-transition, and other times it stays off entirely. Adjusting the color or brightness settings usually restores functionality.

I’ve found the most reliable solution is to add a 1-second delay after the light.toggle command in my HA automation when turning the light on or off with a remote.

I recently switched from deCONZ, where the light worked flawlessly.

CodingGhost, 2025-01-20:
Really interesting to know it wors with deconz, that means there is definitely a way to fix this in z2m.

w-marco, 2025-03-23:
I am also seeing this issue with two of my TS0503B devices. Didn't have it with ZHA either, so it's somewhat Z2M specific.

WhoTheHeck, 2025-07-27:
Issue persists. To make matters worse, I now barely use the TS0503B anymore, because it often gets stuck in dark-state although state gets toggled on/off

closed 2025-10-04 by github-actions[bot]