Koenkk/zigbee2mqtt #25368: Error starting HomeAssistant Addon 1.42.0-2  zigbee2mqtt

thomas682, 2024-12-30:
### What happened?

Try to start with home assist zigbee2mqtt V1.42.0-2 and get an error.

Configuration of HA: 
[image]

After try to start i get a new configuration.yaml file:

homeassistant: true
advanced:
  network_key:
    - 23
    - 35
    - 180
    - 61
    - 84
    - 225
    - 178
    - 141
    - 115
    - 47
    - 163
    - 232
    - 188
    - 48
    - 165
    - 81
  pan_id: 10145
  ext_pan_id:
    - 5
    - 133
    - 240
    - 64
    - 27
    - 33
    - 103
    - 145
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: atee6efeekiethu0uthar7coh1jeizahnai0paiT1ahgh2keeZ2Chai5Oat2zaet
frontend:
  port: 8099

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

20241001

### Adapter

HamGeek POE (HG POE Plus)

### Setup

HA as a VM installation...

### Debug log

Starting Zigbee2MQTT without watchdog.
[2024-12-30 08:49:10] info: 	z2m: Logging to console, file (filename: log.log)
[2024-12-30 08:49:10] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown)
[2024-12-30 08:49:10] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-30 08:49:10] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-30 08:49:10] error: 	z2m: Failed to start zigbee
[2024-12-30 08:49:10] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-30 08:49:10] error: 	z2m: Exiting...
[2024-12-30 08:49:10] error: 	z2m: Error: No path provided and failed to auto detect path
    at Function.create (/app/node_modules/zigbee-herdsman/src/adapter/adapter.ts:107:23)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:133:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)

vladimirdarak, 2024-12-30:
Maybe this link will be helpful:
https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/664#issuecomment-2511465475

thomas682, 2024-12-30:
@vladimirdarak I try that, reading all the links comment, but I get always the same error in the log!
data_path: /addon_config/zigbee2mqtt
and copy all files from config/zigbee2mqtt -> /addon_configs/45df7312_zigbee2mqtt/

Folder /addon_configs/45df7312_zigbee2mqtt/:
[image]
[image]

log file shows always:
Starting Zigbee2MQTT without watchdog.
[2024-12-30 14:55:40] info: 	z2m: Logging to console, file (filename: log.log)
[2024-12-30 14:55:40] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown)
[2024-12-30 14:55:40] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-30 14:55:40] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-30 14:55:40] error: 	z2m: Failed to start zigbee
[2024-12-30 14:55:40] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-30 14:55:40] error: 	z2m: Exiting...
[2024-12-30 14:55:40] error: 	z2m: Error: No path provided and failed to auto detect path
    at Function.create (/app/node_modules/zigbee-herdsman/src/adapter/adapter.ts:107:23)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:133:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)

I deinstall the zigbe2mqtt addon, install new - same result
PS. thats my first installation with zigbee2mqtt, i havent install before (i use zha) and have not done a update...

I want to use zigbe2mqtt because i want to use the 20 dbm power amplifier from zigbee2mqtt (i dont find that parameter in zha)

vladimirdarak, 2024-12-30:
> @vladimirdarak, thanks for your support. I try what you wrote and reading the comments from the links, but I get always the same error in the log!
> 
> I change the data_path to: data_path: /addon_config/zigbee2mqtt and copy all files from config/zigbee2mqtt -> /addon_configs/45df7312_zigbee2mqtt/
> 
> Folder /addon_configs/45df7312_zigbee2mqtt/ <img alt="Screenshot at Dec 30 15-04-43" width="493" src="https://private-user-images.githubusercontent.com/64754166/399306481-78aa7fa2-0a46-4331-bc67-888498f5785a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1Njg0MjAsIm5iZiI6MTczNTU2ODEyMCwicGF0aCI6Ii82NDc1NDE2Ni8zOTkzMDY0ODEtNzhhYTdmYTItMGE0Ni00MzMxLWJjNjctODg4NDk4ZjU3ODVhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjMwVDE0MTUyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNjYTc0MmE1YzM4ZGM0ZWQ3N2UzOTc0MTU4ZGUzZjk3ZjkyZDAyMjEwZGQ3ZWUzY2UyMDM1OTEwMDg5NzU5NGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.zuia6sC42j9m4N7khPmaMmAIQ5Z8jUbhcOHvO1ggf_8">
> 
> Folder /addon_configs/45df7312_zigbee2mqtt/zigbee2mqtt/: <img alt="Screenshot at Dec 30 15-05-02" width="534" src="https://private-user-images.githubusercontent.com/64754166/399306478-99883ffc-8aa1-4227-839c-df069f82420b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1Njg0MjAsIm5iZiI6MTczNTU2ODEyMCwicGF0aCI6Ii82NDc1NDE2Ni8zOTkzMDY0NzgtOTk4ODNmZmMtOGFhMS00MjI3LTgzOWMtZGYwNjlmODI0MjBiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjMwVDE0MTUyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYzZWJmMjQyODQ2NmY4Y2JmNTA4NzUxMzUyMzk5MDAzYjhjYTA5OWE3MDU4YmEzYTllZjZiNjk4NDA0ZWQxMWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0AVVEeGSa-YFAKLcOYn8UoocRuPnIsTE5khRvbFik6A">
> 
> After "start" i get the log error : Starting Zigbee2MQTT without watchdog. [2024-12-30 14:55:40] info: z2m: Logging to console, file (filename: log.log) [2024-12-30 14:55:40] info: z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown) [2024-12-30 14:55:40] info: z2m: Starting zigbee-herdsman (2.1.9) [2024-12-30 14:55:40] error: z2m: Error while starting zigbee-herdsman [2024-12-30 14:55:40] error: z2m: Failed to start zigbee [2024-12-30 14:55:40] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions [2024-12-30 14:55:40] error: z2m: Exiting... [2024-12-30 14:55:40] error: z2m: Error: No path provided and failed to auto detect path at Function.create (/app/node_modules/zigbee-herdsman/src/adapter/adapter.ts:107:23) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:133:24) at Zigbee.start (/app/lib/zigbee.ts:69:27) at Controller.start (/app/lib/controller.ts:161:27) at start (/app/index.js:154:5)
> 
> I deinstall the zigbe2mqtt addon, install new - same result PS. thats my first installation with zigbee2mqtt, i havent install before (i use zha) and have not done a update...
> 
> I want to use zigbe2mqtt because i want to use the 20 dbm power amplifier from zigbee2mqtt (i dont find that parameter in zha)

Looks like zigbee2mqtt don't have specified correct path to coordinator

vladimirdarak, 2024-12-30:
Try to add adapter in to serial section:
[image]

vladimirdarak, 2024-12-30:
https://community.home-assistant.io/t/getting-zigbee2mqtt-to-run-with-hamgeek-poe-zigbee-3-0-coordinator/759802/2

vladimirdarak, 2024-12-30:
Is port correct?

thomas682, 2024-12-30:
Connection to 192.168.2.33 port 6638 [udp/*] succeeded!

If i change to adapter: ezsp i get:

[15:38:39] INFO: Preparing to start...
[15:38:39] INFO: Socat not enabled
[15:38:40] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-12-30 15:38:41] info: 	z2m: Logging to console, file (filename: log.log)
[2024-12-30 15:38:41] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown)
[2024-12-30 15:38:41] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-30 15:38:41] warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2024-12-30 15:38:51] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
Error: Failure to connect
    at SerialDriver.resetForReconnect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:341:19)
    at SerialDriver.emit (node:events:519:28)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:341:22
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at SerialDriver.reset (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:326:16)
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:149:17)

my Config:
[image]

vladimirdarak, 2024-12-30:
Looks like telnet works but on udp port. Zigbee2mqtt requires TCP. So you can:
I'm unable to send emails directly, but I can provide the instructions in English for you to copy and send. Here's the solution in English:

---

Problem: UDP Protocol Detected Instead of TCP

The message Connection to 192.168.2.33 port 6638 [udp/*] succeeded! indicates that your HamGeek POE Zigbee coordinator is responding on port 6638 using the UDP protocol instead of the expected TCP protocol. Since Zigbee2MQTT requires TCP, this mismatch causes the error.

---

Solution for UDP Connection

If your coordinator uses UDP, here are the steps to create a TCP bridge using the socat tool:

---

Step 1: Verify Coordinator Configuration

Check the documentation or settings of your HamGeek POE coordinator to see if you can switch the communication protocol from UDP to TCP.

If the coordinator does not support TCP, proceed with the following workaround.

---

Step 2: Use socat to Bridge UDP to TCP

The socat tool can create a TCP-to-UDP bridge, allowing Zigbee2MQTT to communicate over TCP while the coordinator continues using UDP.

1. Install socat:

sudo apt-get install socat

2. Start the socat Bridge: Run the following command to listen on TCP port 6638 and forward data to the coordinator's UDP port:

socat TCP-LISTEN:6638,fork UDP:192.168.2.33:6638

3. Update Zigbee2MQTT Configuration: Modify the configuration.yaml file to use the local TCP bridge:

serial:
  port: tcp://127.0.0.1:6638
  adapter: ezsp

---

Step 3: Test the Connection

Restart Zigbee2MQTT and verify if the connection is successful.

Check the logs to confirm that Zigbee2MQTT is running correctly:

journalctl -u zigbee2mqtt -f

---

Step 4: Make socat Persistent

To ensure that socat starts automatically after a reboot:

1. Create a Systemd Service:

sudo nano /etc/systemd/system/socat.service

2. Add the Following Configuration:

[Unit]
Description=Socat UDP to TCP bridge
After=network.target

[Service]
ExecStart=/usr/bin/socat TCP-LISTEN:6638,fork UDP:192.168.2.33:6638
Restart=always

[Install]
WantedBy=multi-user.target

3. Enable and Start the Service:

sudo systemctl enable socat
sudo systemctl start socat

thomas682, 2024-12-30:
@vladimirdarak , many thanks for your support!
Properly i have solved the problem!!

Type of Connector:
Hamgeek HG POE Plus (Ethernet & USB Connector)
Alibaba or Amazon: Hangeek HMG-01 plus/plus poe zigbee gateway adapter zigbee 3,0 netzwerk koordinator unterstützung ethernet lan/usb/wifi/bluetooth

a) Firmware must be an older one (I take "20230507")
[image]

b) configuration in HA Addon should be:
[image]

After that the connection can be extablished and i get devices...

Thanks for all.
Thomas

closed 2024-12-30 by Koenkk