Koenkk/zigbee2mqtt #25370: NodOn si -4-FP-21 Stop report energy and sometime the power

steve41200, 2024-12-30:
### What happened?

I havé 9 of this devices.
Some times 2 or 3 (not always the same) stop to report the energy.
When i reconfigure them in z2m, few time after it works again. In the energy Homeassitant Time i see i big point of energy (dépend of the Time with no report) and a négativ non track energy to the count of day before.

this device sometime break report of instant power too.
But the reconfiguration in z2m don r résolve it. I reboot the device by turn off the electricity

the control with firewire seems always work

### What did you expect to happen?

It works without trick like reconfigure eaxh day the device

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20240710

### Adapter

Texas board CC1352P

### Setup

Add one home assistant on a proxmox (intelNuc)

ericsche, 2025-02-05:
Hello I do have the same issue, any log we should provide

GilRoiB, 2025-02-12:
Bonjour, probleme identique

qdgtruie, 2025-04-14:
still an issue

closed 2025-06-23 by github-actions[bot]