Koenkk/zigbee2mqtt #25371: TS0202 _TZ3000_tqj7fiv7 - Tesla motion sensor

vladimirdarak, 2024-12-30:
### What happened?

Can't change occupancy timeout for this device. Occupancy is correctly detected but timout is maybe 2-3s and then device report not occupied.

### What did you expect to happen?

When occypancy is detected longer occupancy delay will be better e.g. 30, 60 or 90s. I tried customize definition of converter but with no luck.

### How to reproduce it (minimal and precise)

Every time when device detect occupancy status occupancy=false is fired in 2-3s

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

SONOFF Dongle E - EmberZNet

### Setup

Add-on on Homeassistant OS on Intel NUC (X86-64) in VM

GuessWho101010, 2025-01-05:
I have just bought multiple these PIRs. All behave as was desceibed above.

Timeout after 2-3 is totally anoying. It will overload my Zigbee network by useless state changes.

I am afraid that this will be a reason why I will return these PIR to seller if there is no solution comming.

GuessWho101010, 2025-03-07:
Unfortunately, this issue is still relevant

GuessWho101010, 2025-04-25:
> Any news?

No. The new Z2M version did not bring any solution.

Still no way to setup the occupancy timeout.