Koenkk/zigbee2mqtt #25372: [New device support]: Moes Thermostat BHT-002-GCLZB _TZE204_xalsoe3m

damo8353, 2024-12-30:
### Link

https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html

### Database entry

?

### Zigbee2MQTT version

1.42.0-2

### Comments

I tried to add the device but it was not recognised. It seems to use a different id: **_TZE204_xalsoe3m**

### External definition

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'BHT-002-GCLZB',
    vendor: 'Moes',
    description: 'Smart knob thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
        e.binary('rgb_backlight', ea.STATE_SET, 'OFF', 'ON').withLabel('RGB backlight').withDescription('Default: Off'),
        e.binary('eco_mode', ea.STATE_SET, 'OFF', 'ON').withLabel('ECO mode NOT SET').withDescription('Default: Off'),
        e.numeric('eco_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(30)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Max temperature in ECO mode. Default: 20'),
        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET)
            .withDescription('Default: -3'),
        e.numeric('deadzone_temperature', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(5)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Hysteresis. Default: 1'),
        e.numeric('min_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 5'),
        e.numeric('max_temperature', ea.STATE_SET)
            .withValueMin(15)
            .withValueMax(45)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 35'),
        e.numeric('min_temperature_limit', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('Low temperature protection')
            .withDescription('Default: 0'),
        e.numeric('lower_limit_of_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('The lower limit of temperature')
            .withDescription('Default: 5'),
        e.enum('sensor', ea.STATE_SET, ['IN', 'OU', 'AL']).withLabel('Sensor')
            .withDescription('Choose which sensor to use. Default: AL'),
        e.numeric('external_temperature_input', ea.STATE)
            .withLabel('Floor temperature')
            .withUnit('°C')
            .withDescription('Temperature from floor sensor'),
        e.numeric('brightness', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(100)
            .withValueStep(1)
            .withLabel('Backlight brightness')
            .withDescription('Backlight brightness in %'),
        e.text('schedule_monday', ea.STATE_SET)
            .withLabel('Schedule for monday')
            .withDescription('Default: 06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0'),
        e.text('schedule_tuesday', ea.STATE_SET)
            .withLabel('Schedule for tuesday')
            .withDescription('Default: 06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0'),
        e.text('schedule_wednesday', ea.STATE_SET)
            .withLabel('Schedule for wednesday')
            .withDescription('Default: 06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0'),
        e.text('schedule_thursday', ea.STATE_SET)
            .withLabel('Schedule for thursday')
            .withDescription('Default: 06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0'),
        e.text('schedule_friday', ea.STATE_SET)
            .withLabel('Schedule for friday')
            .withDescription('Default: 06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0'),
        e.text('schedule_saturday', ea.STATE_SET)
            .withLabel('Schedule for saturday')
            .withDescription('Default: 06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0'),
        e.text('schedule_sunday', ea.STATE_SET)
            .withLabel('Schedule for sunday')
            .withDescription('Default: 06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0'),
        e.enum('factory_reset', ea.STATE_SET, ['factory_reset'])
            .withLabel('Factory reset')
            .withDescription('Reset all settings to factory ones'),
            
        e.enum('screen_time_set', ea.STATE_SET, ['off', '10s', '30s', '60s'])
            .withLabel('Screen Timeout')
            .withDescription('Set screen timeout duration. Default: off'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'system_mode', tuya.valueConverterBasic.lookup({off: false, heat: true})],
            [2, 'preset', tuya.valueConverterBasic.lookup({manual: tuya.enum(0), 'temporary manual': tuya.enum(1), schedule: tuya.enum(2), eco: tuya.enum(3)})],
            [16, 'local_temperature', tuya.valueConverter.divideBy10],
            [18, 'lower_limit_of_temperature', tuya.valueConverter.divideBy10],
            [28, 'factory_reset', tuya.valueConverterBasic.lookup({factory_reset: true})],
            [32, 'sensor', tuya.valueConverterBasic.lookup({IN: tuya.enum(0), OU: tuya.enum(2), AL: tuya.enum(1)})],
            [34, 'max_temperature', tuya.valueConverter.divideBy10],
            [39, 'child_lock', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
            [40, 'eco_mode', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
            [47, 'valve_state', tuya.valueConverter.trueFalseInvert],
            [48, 'brightness', tuya.valueConverter.raw],
            [50, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [101, 'local_temperature_calibration', tuya.valueConverter.divideBy10],
            [102, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [103, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
            [110, 'deadzone_temperature', tuya.valueConverter.raw],
            [111, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [112, 'min_temperature_limit', tuya.valueConverter.divideBy10],
            [113, 'eco_temperature', tuya.valueConverter.divideBy10],
            [114, 'screen_time_set', tuya.valueConverterBasic.lookup({
                off: tuya.enum(0), 
                '10s': tuya.enum(1), 
                '30s': tuya.enum(2), 
                '60s': tuya.enum(3)
            })],
            [115, 'rgb_backlight', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

i copied this definition from a different MOES device, i cannot find the correct one for Moes Thermostat BHT-002-GCLZB. Many of the data point are not correct. I assume adding the Zigbee ID _TZE204_xalsoe3m to the existing device converter will fix this issue. Sorry my knowledge in this area is poor.

CrazyBavarian, 2025-01-03:
@damo8353 could you tell me how exactly did you adjsut the external converter to use the BHT-002-GCLZB _TZE204_xalsoe3m. I have the some one and i would like to fix the same issue

damo8353, 2025-01-03:
This works for me with this device. It refers to zigbeeModel: ['TS0601']. I guess it should specify the fingerprint id but I haven't tested it yet.

CrazyBavarian, 2025-01-03:
thanks for the fast respone where i need to add the fingerpint?

damo8353, 2025-01-03:
Yes that should work, you can compare it to other Tuya (New device) issues. Like the one below:

```

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE204_a2jcoyuk'}
    ],
    model: 'TS0601_cover_dev',
    vendor: 'Tuya',
    description: 'Curtain motor',

```

damo8353, 2025-01-03:
So, if you want try this:

```

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE204_xalsoe3m'}
    ],
    zigbeeModel: ['TS0601'],
    model: 'BHT-002-GCLZB',
    vendor: 'Moes',
    description: 'Smart thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
        e.binary('rgb_backlight', ea.STATE_SET, 'OFF', 'ON').withLabel('RGB backlight').withDescription('Default: Off'),
        e.binary('eco_mode', ea.STATE_SET, 'OFF', 'ON').withLabel('ECO mode NOT SET').withDescription('Default: Off'),
        e.numeric('eco_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(30)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Max temperature in ECO mode. Default: 20'),
        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET)
            .withDescription('Default: -3'),
        e.numeric('deadzone_temperature', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(5)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Hysteresis. Default: 1'),
        e.numeric('min_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 5'),
        e.numeric('max_temperature', ea.STATE_SET)
            .withValueMin(15)
            .withValueMax(45)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 35'),
        e.numeric('min_temperature_limit', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('Low temperature protection')
            .withDescription('Default: 0'),
        e.numeric('lower_limit_of_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('The lower limit of temperature')
            .withDescription('Default: 5'),
        e.enum('sensor', ea.STATE_SET, ['IN', 'OU', 'AL']).withLabel('Sensor')
[... 57 lines cut ...]
            [104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
            [110, 'deadzone_temperature', tuya.valueConverter.raw],
            [111, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [112, 'min_temperature_limit', tuya.valueConverter.divideBy10],
            [113, 'eco_temperature', tuya.valueConverter.divideBy10],
            [114, 'screen_time_set', tuya.valueConverterBasic.lookup({
                off: tuya.enum(0), 
                '10s': tuya.enum(1), 
                '30s': tuya.enum(2), 
                '60s': tuya.enum(3)
            })],
            [115, 'rgb_backlight', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;

```

CrazyBavarian, 2025-01-03:
@damo8353 thank it works :)

CrazyBavarian, 2025-01-04:
@damo8353  one more question. where did you find this code ? My Thermostat is connected but some values are wrong need to multiple by 10 i would like to use the the code from  this decice just to try [https://www.zigbee2mqtt.io/devices/BHT-002_BHT-006.html](url)

damo8353, 2025-01-04:
@CrazyBavarian I copied the code from:
https://github.com/Koenkk/zigbee2mqtt/issues/25038

Regarding the data points, I have looked for the correct ones but I cannot find them. I was hoping someone can help with this. 

For the x10 error just change the code from tuya.valueConverter.divideBy10 to tuya.valueConverter.raw as below: 

'''

[50, 'current_heating_setpoint', tuya.valueConverter.raw],

'''

I also noticed there is an error when the set point is set to one decimel place. If you change to a whole number it works: 
Full code below: 

'''

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'BHT-002-GCLZB',
    vendor: 'Moes',
    description: 'Smart thermostat -5',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
        e.binary('eco_mode', ea.STATE_SET, 'OFF', 'ON').withLabel('ECO mode').withDescription('Default: Off'),
        e.numeric('eco_temperature', ea.STATE_SET)
            .withValueMin(10)
            .withValueMax(25)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Max temperature in ECO mode. Default: 20'),
        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])
            .withSetpoint('current_heating_setpoint', 10, 25, 1, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET)
            .withDescription('Default: -3'),
        e.numeric('deadzone_temperature', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(5)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Hysteresis. Default: 1'),
        e.numeric('min_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 5'),
        e.numeric('max_temperature', ea.STATE_SET)
            .withValueMin(15)
            .withValueMax(25)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 25'),
        e.numeric('min_temperature_limit', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withUnit('°C')
[... 55 lines cut ...]
            [39, 'child_lock', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
            [40, 'eco_mode', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
            [47, 'valve_state', tuya.valueConverter.raw],
            [50, 'current_heating_setpoint', tuya.valueConverter.raw],
            [101, 'local_temperature_calibration', tuya.valueConverter.divideBy10],
            [102, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [103, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
            [110, 'deadzone_temperature', tuya.valueConverter.raw],
            [111, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [112, 'min_temperature_limit', tuya.valueConverter.divideBy10],

        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;

'''

CrazyBavarian, 2025-01-05:
@damo8353 thank you but there some values wrong i try to get the informartion for the other device maybe i can see there the changes.
like 
- The lower limit of temperature 
- Local temperature calibration my feeling it is not working the local temperature never chnaged

noose01, 2025-01-05:
I have these for electric floor heating with same manufacturer name:
https://moeshouse.com/products/zigbee-002-thermostat-for-heating-boiler?variant=44628000571707
SKU: ZHT-NV-GB-WH-MS

Zigbee Model: TS0601
Zigbee Manufacturer: _TZE204_xalsoe3m
Support status: Not supported

It seems to have somewhat identical functionalities with OP's device.
What can I do to get it working and maybe contribute to get it supported?

CrazyBavarian, 2025-01-05:
@damo8353  i just found the code [https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/moes.ts](url) jfi

damo8353, 2025-01-06:
Hi @CrazyBavarian I read this file before, it references https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/legacy.ts#L563 which contains datapoints. However the datapoints are different: for example childLock is listed as '7' but on the device (_TZE204_xalsoe3m) childLock=39 (which works). 
I can't seem to find a list of datapoints that work with this device.

damo8353, 2025-01-06:
@noose01  you need to the code below as an 'external definition', refer to: the link below if you are on zigbee2mqtt v2.
https://www.zigbee2mqtt.io/advanced/more/external_converters.html

Pre- version 2 you need to  save the code below as [filename].js file in /config/zigbee2mqtt folder. 
Then add the following to the zigbee2mqtt config: 

```
external_converters:
  - [filename].js
```

I commented out the bits that don't work:
```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'BHT-002-GCLZB',
    vendor: 'Moes',
    description: 'Smart thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
        e.binary('eco_mode', ea.STATE_SET, 'ON', 'OFF').withLabel('ECO mode').withDescription('Default: Off'),
//        e.numeric('eco_temperature', ea.STATE_SET)
//            .withValueMin(10)
//            .withValueMax(25)
//            .withValueStep(1)
//            .withUnit('°C')
//            .withDescription('Max temperature in ECO mode. Default: 20'),
        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])
            .withSetpoint('current_heating_setpoint', 10, 25, 1, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET)
            .withDescription('Default: -3'),
//        e.numeric('deadzone_temperature', ea.STATE_SET)
//            .withValueMin(1)
//            .withValueMax(5)
//            .withValueStep(1)
//            .withUnit('°C')
//            .withDescription('Hysteresis. Default: 1'),
//        e.numeric('min_temperature', ea.STATE_SET)
//            .withValueMin(5)
//            .withValueMax(15)
//            .withValueStep(1)
//            .withUnit('°C')
//            .withDescription('Default: 5'),
//        e.numeric('max_temperature', ea.STATE_SET)
//            .withValueMin(15)
//            .withValueMax(25)
//            .withValueStep(1)
//            .withUnit('°C')
//            .withDescription('Default: 25'),
//        e.numeric('min_temperature_limit', ea.STATE_SET)
//            .withValueMin(0)
//            .withValueMax(10)
//            .withValueStep(1)
//            .withUnit('°C')
//            .withLabel('Low temperature protection')
//            .withDescription('Default: 0'),
//        e.numeric('lower_limit_of_temperature', ea.STATE_SET)
//            .withValueMin(5)
//            .withValueMax(15)
//            .withValueStep(1)
//            .withUnit('°C')
//            .withLabel('The lower limit of temperature')
//            .withDescription('Default: 5'),
        e.enum('sensor', ea.STATE_SET, ['IN', 'OU', 'AL']).withLabel('Sensor')
            .withDescription('Choose which sensor to use. Default: AL'),
        e.numeric('external_temperature_input', ea.STATE)
            .withLabel('Floor temperature')
            .withUnit('°C')
            .withDescription('Temperature from floor sensor'),
        e.enum('factory_reset', ea.STATE_SET, ['factory_reset'])
            .withLabel('Factory reset')
            .withDescription('Reset all settings to factory ones'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'system_mode', tuya.valueConverterBasic.lookup({off: false, heat: true})],
            [2, 'preset', tuya.valueConverterBasic.lookup({manual: tuya.enum(0), 'temporary manual': tuya.enum(1), schedule: tuya.enum(2), eco: tuya.enum(3)})],
            [8, 'open_window', tuya.valueConverter.onOff],
            [16, 'local_temperature', tuya.valueConverter.divideBy10],
//            [18, 'lower_limit_of_temperature', tuya.valueConverter.divideBy10],
//            [26, 'min_temperature', tuya.valueConverter.divideBy10],
            [28, 'factory_reset', tuya.valueConverterBasic.lookup({factory_reset: true})],
            [32, 'sensor', tuya.valueConverterBasic.lookup({IN: tuya.enum(0), OU: tuya.enum(2), AL: tuya.enum(1)})],
// .divideBy10
//            [34, 'max_temperature', tuya.valueConverter.raw],
            [39, 'child_lock', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
            [40, 'eco_mode', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
// INVERT .trueFalseInvert
            [47, 'valve_state', tuya.valueConverter.raw],
            [50, 'current_heating_setpoint', tuya.valueConverter.raw],
            [101, 'local_temperature_calibration', tuya.valueConverter.divideBy10],
            [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
//            [110, 'deadzone_temperature', tuya.valueConverter.raw],
//            [111, 'max_temperature_limit', tuya.valueConverter.divideBy10],
//            [112, 'min_temperature_limit', tuya.valueConverter.divideBy10],
//            [113, 'eco_temperature', tuya.valueConverter.divideBy10],

        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;

```

damo8353, 2025-01-06:
> @damo8353 thank you but there some values wrong i try to get the informartion for the other device maybe i can see there the changes. like
> 
> * The lower limit of temperature
> * Local temperature calibration my feeling it is not working the local temperature never chnaged

Same for me. The calibration would be useful as it not possible to set on the device. 
The lower / upper limits as well as the ECO set point can be fixed on the device.

noose01, 2025-01-09:
@damo8353 I run zigbee2mqtt v2.
I created file TS0601.js in data/external_converters/ folder with your code above. I did also put same file under zigbee2mqtt/data/external_converters/.
Do I need to activate it somehow or is that enough?

I removed devices from interface and re-paired them.
[image]

adamkorski, 2025-01-16:
zigbee2mqtt/external_converters/

noose01, 2025-01-16:
Thanks! Now it works for me.
Floor temperature from external sensor is Null regardless what sensor option I use.
It might be so that temperature from external sensor is seen in Local temperature as my bathroom temperature isn't definitely 27 celcius but floor might be.

damo8353, 2025-01-29:
@SSG1111  that's great, thank you.

SSG1111, 2025-02-01:
there is a mistake in the code above. 
running_state must be changed from e.binary() to e.numeric()

ftz19, 2025-02-01:
This are the Query Properties for ZHT-002, xalsoe3m:
{
  "result": {
    "properties": [
      {
        "code": "switch",
        "custom_name": "",
        "dp_id": 1,
        "time": 1738439377701,
        "type": "bool",
        "value": false
      },
      {
        "code": "mode",
        "custom_name": "",
        "dp_id": 2,
        "time": 1738433484383,
        "type": "enum",
        "value": "auto"
      },
      {
        "code": "temp_current",
        "custom_name": "",
        "dp_id": 16,
        "time": 1738439378111,
        "type": "value",
        "value": 235
      },
      {
        "code": "lower_temp",
        "custom_name": "",
        "dp_id": 18,
        "time": 1738428635053,
        "type": "value",
        "value": 10
      },
      {
        "code": "temp_correction",
        "custom_name": "",
        "dp_id": 19,
        "time": 1738429573768,
        "type": "value",
        "value": -5
      },
      {
        "code": "work_days",
        "custom_name": "",
        "dp_id": 23,
        "time": 1738428635464,
        "type": "enum",
        "value": "5_2"
      },
      {
        "code": "sensor_choose",
        "custom_name": "",
        "dp_id": 32,
        "time": 1738429261055,
        "type": "enum",
        "value": "in"
      },
      {
        "code": "upper_temp",
        "custom_name": "",
        "dp_id": 34,
        "time": 1738428638067,
        "type": "value",
        "value": 35
      },
      {
        "code": "child_lock",
        "custom_name": "",
        "dp_id": 39,
        "time": 1738428638436,
        "type": "bool",
        "value": false
      },
      {
        "code": "eco",
        "custom_name": "",
        "dp_id": 40,
        "time": 1738433582077,
        "type": "bool",
        "value": false
      },
      {
        "code": "valve_state",
        "custom_name": "",
        "dp_id": 47,
        "time": 1738439377602,
        "type": "enum",
        "value": "close"
      },
      {
        "code": "temp_set",
        "custom_name": "",
        "dp_id": 50,
        "time": 1738439379240,
        "type": "value",
        "value": 26
      },
      {
        "code": "week_program_4",
        "custom_name": "",
        "dp_id": 68,
        "time": 1738439379365,
        "type": "raw",
        "value": "BgIAGgseABoNHgAaER4AGgYAABoMAAAaDh4AGhEeABoGAAAaDB4AGg4eABoSHgAa"
      },
      {
        "code": "overheat_protection",
        "custom_name": "",
        "dp_id": 101,
        "time": 1738428639147,
        "type": "value",
        "value": 45
      },
      {
        "code": "blind_spot",
        "custom_name": "",
        "dp_id": 102,
        "time": 1738428639252,
        "type": "value",
        "value": 1
      }
    ]
  },
  "success": true,
  "t": 1738445011855,
  "tid": "c96b2699e0e211efa76ab217c6c149bc"
}

SSG1111, 2025-02-02:
thanks @ftz19 ! I didn't figure out "work_days"... is it somehow related to the sheduled mode?

noose01, 2025-02-02:
I think there were update for Z2MGTT and now "current heating setpoint" is 240.
If I try to lower it manually, it just pops back to 240.

SSG1111, 2025-02-03:
yes, the same here...
setpoint and calibration values are not properly scaled. this needs to be reported

damo8353, 2025-02-03:
To scale properly use the 'divideby10' function: 

            [16, 'local_temperature', tuya.valueConverter.**divideBy10]**,

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_xalsoe3m',
    description: 'MOES thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
        e.binary('running_state', ea.STATE).withLabel('Running state'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])            
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 45, 1, ea.STATE_SET)
            .withLocalTemperatureCalibration(-10, 10, 1, ea.STATE_SET),
            e.binary('eco_mode', ea.STATE_SET, 'ON', 'OFF').withLabel('ECO mode').withDescription('Default: Off'),
            e.enum('sensor', ea.STATE_SET, ['IN', 'OU', 'AL']).withLabel('Sensor')
            .withDescription('Choose which sensor to use. Default: AL'),            
        e.numeric('deadzone_temperature', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(5)
            .withValueStep(1)
            .withLabel('Dead zone temperature')
            .withUnit('°C'),
        e.numeric('temp_set_min', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(45)
            .withValueStep(1)
            .withLabel('Setpoint min')
            .withUnit('°C'),
        e.numeric('temp_set_max', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(45)
            .withValueStep(1)
            .withLabel('Setpoint max')
            .withUnit('°C'),
        e.numeric('max_temperature_limit', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(60)
            .withValueStep(1)
            .withLabel('Temperature limit')
            .withUnit('°C'),
//        e.numeric('work_days', ea.STATE).withLabel('work_days'),
        ],
    meta: {
        tuyaDatapoints: [
            [1, 'system_mode', tuya.valueConverterBasic.lookup({off: false, heat: true})],
            [2, 'preset', tuya.valueConverterBasic.lookup({schedule: tuya.enum(0), manual: tuya.enum(1)})],
            [16, 'local_temperature', tuya.valueConverter.divideBy10],
            [18, 'temp_set_min', tuya.valueConverter.raw],
            [19, 'local_temperature_calibration', tuya.valueConverter.raw],
//            [23, 'work_days', tuya.valueConverter.raw],
            [32, 'sensor', tuya.valueConverterBasic.lookup({IN: tuya.enum(0), OU: tuya.enum(1), AL: tuya.enum(2)})],
            [34, 'temp_set_max', tuya.valueConverter.raw],
            [39, 'child_lock', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
            [40, 'eco_mode', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
            [47, 'running_state', tuya.valueConverter.raw],
            [50, 'current_heating_setpoint', tuya.valueConverter.raw],            
            [68, 'week_program_4', tuya.valueConverter.raw],
            [101, 'max_temperature_limit', tuya.valueConverter.raw],
            [102, 'deadzone_temperature', tuya.valueConverter.raw],
        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};
module.exports = definition;

```

darchap, 2025-03-25:
> [@damo8353](https://github.com/damo8353) thank you! Since device was added to "supported devices" in Z2M, external converter is not working anymore...

Hello, Ive been strugling with this thermostat for 3 days now. I had before same model but wifi one, now Ive discovered that some people have the same issue but I dont know how to proceed since now its "supported"

SSG1111, 2025-03-26:
Hi! I have manage to make external converter running again (there were some issues with the code because of braking changes in recent HA udpate). Device is up running with the external converter.

darchap, 2025-03-26:
> Hi! I have manage to make external converter running again (there were some issues with the code because of braking changes in recent HA udpate).
> 
> save this to moes_mod.js file, copy it to \zigbee2mqtt\external_converters\ and restart z2m. there are still some issues with calibration scailing (had no time to fix it), but that is just visual, device works fine.
> 
> ```
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     fingerprint: tuya.fingerprint('TS0601', ['_TZE204_xalsoe3m']),
>     model: 'TS0601',
>     vendor: 'MOES',
>     description: 'MOES thermostat',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     exposes: [
>         e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
>         e.numeric('running_state', ea.STATE).withLabel('Running state'),
>         e.climate()
>             .withSystemMode(['off', 'heat'], ea.STATE_SET)
>             .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])            
>             .withSetpoint('current_heating_setpoint', 5, 45, 1, ea.STATE_SET)
>             .withLocalTemperature(ea.STATE)
>             .withLocalTemperatureCalibration(-10, 10, 1, ea.STATE_SET)
>             .withDescription('Default: -3'),
>         e.enum('sensor', ea.STATE_SET, ['IN', 'OU', 'AL']).withLabel('Sensor')
>             .withDescription('Choose which sensor to use. Default: AL'),            
>         e.numeric('deadzone_temperature', ea.STATE_SET)
>             .withValueMin(0)
>             .withValueMax(5)
>             .withValueStep(1)
>             .withLabel('Dead zone temperature')
>             .withUnit('°C'),
>         e.numeric('temp_set_min', ea.STATE_SET)
>             .withValueMin(5)
>             .withValueMax(45)
>             .withValueStep(1)
>             .withLabel('Setpoint min')
>             .withUnit('°C'),
>         e.numeric('temp_set_max', ea.STATE_SET)
>             .withValueMin(5)
>             .withValueMax(45)
>             .withValueStep(1)
>             .withLabel('Setpoint max')
>             .withUnit('°C'),
>         e.numeric('max_temperature_limit', ea.STATE_SET)
>             .withValueMin(5)
>             .withValueMax(60)
>             .withValueStep(1)
>             .withLabel('Temperature limit')
>             .withUnit('°C'),
>         ],
>     meta: {
>         tuyaDatapoints: [
>             [1, 'system_mode', tuya.valueConverterBasic.lookup({off: false, heat: true})],
>             [2, 'preset', tuya.valueConverterBasic.lookup({schedule: tuya.enum(0), manual: tuya.enum(1)})],
>             // [4, 'reg_4', tuya.valueConverter.raw],
>             [16, 'local_temperature', tuya.valueConverter.divideBy10],
>             [18, 'temp_set_min', tuya.valueConverter.raw],
>             [19, 'local_temperature_calibration', tuya.valueConverter.raw],
>             [23, 'reg_23', tuya.valueConverter.raw],
>             [32, 'sensor', tuya.valueConverterBasic.lookup({IN: tuya.enum(0), OU: tuya.enum(1), AL: tuya.enum(2)})],
>             [34, 'temp_set_max', tuya.valueConverter.raw],
>             [39, 'child_lock', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
>             [47, 'running_state', tuya.valueConverter.raw],
>             [50, 'current_heating_setpoint', tuya.valueConverter.raw],            
>             // [68, 'reg_68', tuya.valueConverter.raw],
>             [101, 'max_temperature_limit', tuya.valueConverter.raw],
>             [102, 'deadzone_temperature', tuya.valueConverter.raw],
>         ],
>     },
>     onEvent: tuya.onEventSetLocalTime,
> };
> module.exports = definition;
> ```

This works like a charm, at least its functional. I will try to test and help you with the other problem. Thanks!

noose01, 2025-03-29:
Thanks! I had to use Moes hub for a while but now I can get rid of it (again).

SSG1111, 2025-04-03:
Is it recognized like on the picture below?
[image]

pin26, 2025-05-23:
> Is it recognized like on the picture below? [image]

Hi. Yes. I have the same problem with it. Without any external converters.

[image]
[image]