Koenkk/zigbee2mqtt #25382: Dresden Elektronik  FLS-PP IP device joint as not supported

chilla288, 2024-12-30:
### What happened?

Pairing Dresden Elektronik FLS-PP IP device joints the network, but Manufacturer is "not supported" and model is listed as "FLS-PP3 White".
this state happend suddenly, as first time join did show up with the correct model "FLS-PP3" and the device was working correct. 
due to unknown issue all devices went offline (over night) and after forcing interview the mentioned devices show up as not supported.
I have 3 of these devices, all have the same issue.
[image]

### What did you expect to happen?

joining Dresden Elektronik FLS-PP IP should show up as supported device (FLS-PP3)

### How to reproduce it (minimal and precise)

join the device 

### Zigbee2MQTT version

1.40.1 commit: 403d3c0

### Adapter firmware version

20230507

### Adapter

cod.m  CZC v1.0 zigbee -> ethernet coordinator

### Setup

Docker container on RPI4 -> HA connected to docker instance

Koenkk, 2024-12-30:
Could you provide the data/database.db entry of this device?

chilla288, 2024-12-31:
hello @Koenkk, thanks for taking care. Assume this is the data you are looking for?! (I'm still a beginner in all this zigbee topic)
[coffeeBar-dbEntry.txt](https://github.com/user-attachments/files/18280397/coffeeBar-dbEntry.txt)

Koenkk, 2025-01-02:
Added!

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2025-01-02 by Koenkk