Koenkk/zigbee2mqtt #25392: Sonoff ZBMINIR2 Binding issue

kilian-f-za, 2024-12-31:
### What happened?

I have added the Sonos ZBMINIR2 and updated the firmware (1.0.4). Then I created a binding (onoff) and the message "Route successfully created" appeared. However, the device is not switched. I have tried it with several different devices, but it does not work with any of them.
I usw the SMLIGHT SLZB-06M as Coordinator.
[image]

### What did you expect to happen?

The connected device should switch on and off when the input S2 on sonoff is changed. 
With an Aeotec pico it works with the same procedure.

### How to reproduce it (minimal and precise)

Firmware Update then reset the sonoff and reconnect with zigbee2mqtt

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.4.5 [GA]

### Adapter

Ember

### Setup

on proxmox lxc Container installed with proxmox helper script

bobo-cher, 2025-01-01:
I'm afraid bindings do not work on zbminir2 at all. This is hardly a z2m issue but rather a problem (or specific) of the device. I have myself zbminir2 and never managed bindings to work but I haven't tried anything other than z2m. 

Also see [ZBMINIR2 direct zigbee binding](https://www.reddit.com/r/sonoff/comments/1hjuz4h/zbminir2_direct_zigbee_binding/).

An ultimate test would be to setup zbminir2 and other device with Sonoff gateway and their app; then power off the gateway and test if the binding between two devices works.

kilian-f-za, 2025-01-01:
Thanks for the answer, then I will switch to another device. However, the Sonoff is very small. Direct bindings are very important to me, if the gateway fails at least the light should still come on. It would be great if direct binding worked on many more devices.

bobo-cher, 2025-01-01:
Maybe you could create a support ticket at Sonoff. They should know users need the feature.

ius, 2025-01-01:
I've investigated this issue in detail. It's a limitation of the Sonoff firmware indeed (as mentioned by @bobo-cher).

> I'm afraid bindings do not work on zbminir2 at all.
Doesn't matter in practice for the issue at hand, but it _does_ support bindings. It doesn't however address them when sending on/off commands. Sad thing is that it's a one line fix: call `sl_zigbee_af_send_command_unicast_to_bindings`.

@kilian-f-za I believe Sonoff's firmware is somewhat robust in this regard for most use cases as it reverts to switching the relay (even when configured in detached relay mode) when the coordinator becomes unavailable.

I would in fact prefer to ignore coordinator status and rely on bindings for robustness instead.

> Maybe you could create a support ticket at Sonoff. They should know users need the feature.

They told me they had no plans to add this feature. To work around this issue I've started working on my own firmware - I got bindings to work just fine, but all additional fluff (ota, 'factory' reset on 5x toggle) is not yet implemented.

I'm a bit short on time lately, but if I manage to get ota to work I could look into publishing it. Two major drawbacks though: you need to flash it via swd and there's no way to revert to Sonoff firmware..

bobo-cher, 2025-01-02:
@ius Your project sounds very interesting. I would gladly use one or two of my spare zbminir2 for testing if you are willing to share your work and could provide some instructions for flashing.

My ultimate goal is to use one zbminir2 as sender (connected to wall switch/button) and another zbminir2 as receiver (connected to lamp). It's a shame Sonoff doesn't support that use case out of the box as that would make these small devices such versatile.

That would allow me to transform a simple ceiling light with one wall switch into two ceiling lamps with two independent switches without adding of a new wire. I would use two minis (as senders) in the wall box behind physical switches and two minis (as receivers) on the ceiling.

bobo-cher, 2025-01-07:
I've created a support ticket with Sonoff and got feedback that direct binding is a known feature request.

I encourage every user interested in the feature to contact Sonoff to vote for the feature. That increases chances we got this. Thanks.

ius, 2025-01-16:
I've published my ZBMINIR2 firmware to https://github.com/ius/zbminir2 (in progress, experimental!)

Only the essentials required for my binding use case are implemented (now does include OTA and 'reset'). Anyone interested in _ZBMINIR2-with-bindings_ do check it out.

Disclaimer: there _might_ be reliability issues (might just be my n=1 setup) - additional feedback is thus very welcome.

@bobo-cher Also happy to help out on adapting it to your use case if you're willing to give it a try - preferably catch me on Discord (see README).

bobo-cher, 2025-01-16:
@ius Thank you very much for sharing your project. Appreciate your efforts for publishing of the repository and writing of the docs.

I'll order the required hardware and give it a try. Will get back to you via project issue tracker or Discord. Thanks again!

Pferdebockwurst, 2025-03-12:
> [@Pferdebockwurst](https://github.com/Pferdebockwurst) where is your source for this?

[Here](https://www.reddit.com/r/sonoff/comments/1hjuz4h/comment/mded4i6/?utm_source=share&utm_medium=mweb3x&utm_name=mweb3xcss&utm_term=1&utm_content=share_button).

bjornarfjelldal, 2025-03-12:
> > [@Pferdebockwurst](https://github.com/Pferdebockwurst) where is your source for this?
> 
> [Here](https://www.reddit.com/r/sonoff/comments/1hjuz4h/comment/mded4i6/?utm_source=share&utm_medium=mweb3x&utm_name=mweb3xcss&utm_term=1&utm_content=share_button).

Thanks!

frankforpresident, 2025-09-21:
@ius,

You should update the README if the project is stable, because it still says it is experimental which was one of the reasons I decided to skip it

Hydro8, 2025-11-09:
When the coordinator is down, ZBMINIR2 fallback to relay mode even if he’s in detached mode right ?

andrazek, 2025-12-03:
This is the message I received from Sonoff support:

> The ZBMINIR2 V1.0.8 firmware has begun its phased rollout on our eWeLink platform, adding a binding function that only takes effect after the detach relay mode is enabled. Uploads from third-party platforms will be a little later. You can keep a close eye on the updates.

Can someone using eWeLink upgrade to the latest firmware and try pairing the device with Z2M to confirm whether direct binding works?

andrei-lazarov, 2025-12-04:
Thanks for sharing the good news @andrazek!
They already pushed the update to Z2M: https://github.com/Koenkk/zigbee-OTA/pull/940
@ius @diegosantamarta @frankforpresident @kilian-f-za 

***SONOFF ZBMINIR2 supports binding both ways on stock firmware v1.0.8!***

I updated a unit and did some tests:
- It sends the **toggle** command to bound lights as soon as you press the button (genOnOff cluster).
  - 'edge' and 'pulse' work as expected
  - 'follow on' and 'follow off' don't work for binds! - same behavior as 'edge'
  - individually toggles each bound device (or each member of the bound group) - unpleasant, but expected behavior (this is how Zigbee toggle works)
- Commands to bound lights are sent only while 'detached relay mode' is active
  - this would conflict with the **fallback mechanism** (disable detached mode when coordinator/Z2M offline), but it seems they **removed it** @Hydro8 
  - I found a workaround to 'toggle relay AND send command' on button press: **bind the sonoff to itself** (it uses endpoint 1 for both input and output)
- It can be bound to individual devices and/or groups
  - bind to device works normally
  - bind to group also toggles itself (sonoff relay) for some reason? 
I double-checked: it's not inside the group and I removed the self-bind
- bind commands are sent even if network is down
- they still did not implement on_with_timed_off as far as I can tell

Of course it's missing more advanced features like hold to dim (genLevelCtrl) and syncing relay state with bound lights.
But this is a great improvement! So far I'm happy with it.

Hydro8, 2025-12-04:
Thanks a lot for your feedback on this last firmware. I'm happy that Sonoff put binding option to ZBMINIR2.
This switch seems to be the best one around.

Do you know if it's possible to disable binding from HA ? In case the smart bulb is dead, I would like to easily go back to classic mode for my wife while waiting to change the bulb.

andrei-lazarov, 2025-12-04:
@Hydro8 yes, you just disable 'detached relay mode'.

uschindler, 2025-12-04:
> * bind to group also toggles itself (sonoff relay) for some reason?
>   I double-checked: it's not inside the group and I removed the self-bind

Hi, this is really annoying. It also happens if the ZBMINIR2 is part of a group (I have a group to turn off everything when I leave the house). When you go to detached relay mode and toggle the switch suddenly all other lights in this group also turn on.

This is a bug: When a device is part of a group it should only listen to events for that group and toggle itsself. If a device binds the ONOFF cluster to a group it should toggle that group (and not all groups it is part of).

andrei-lazarov, 2025-12-04:
@uschindler I couldn't replicate this part. It only toggles the group I bound it to.

Pferdebockwurst, 2025-12-04:
Can everyone here confirm that the direct binding doesn't support dimming (GenLevelCtrl cluster)?

uschindler, 2025-12-04:
> [@uschindler](https://github.com/uschindler) It's very much a fault! **It toggles itself even if it's not part of the bound group.**

OK, I did not test this, because my wanted behaviour was to let it switch its own relay in addition to other lamps (see my previous comment).

I have no idea how that happens. Could be some bug, but that's not relevant for me. My main purpose was to synchronize other lights. 

P.S.: You can also synchronize toggling of multiple ZBMINR2 when you put them all in same group and bind themselves to this group. Of course if they are async (one is on, other is off) then each button press toggles between them. 🤣 Nice effects, but absolutely how it should behave (although not always wanted).