Koenkk/zigbee2mqtt #25396: IKEA LED2201G8 can't be added

davzavada, 2024-12-31:
### What happened?

Same issue as in this [thread](https://github.com/Koenkk/zigbee2mqtt/issues/19707#issue-1992371357).

Resetting the bulb and putting into pairing mode is successful but I am unable to connect to it or even see it. There is no mention of the device in the logs.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20221226

### Adapter

SLZB-06

### Setup

Add-on on Home Assitant OS running on LarkBox 2023

### Debug log

[log.log](https://github.com/user-attachments/files/18281872/log.log)

davzavada, 2025-01-18:
Just to give an update, I was still not able to pair the lighbulbs. For anyone dealing with the same issue, I have since tried:

- updating the coordinator firmware
- updating Zigbee2MQTT to version 2.0.0-2 

I have bought an IKEA Dirigera gateway to check if maybe the hardware was faulty in some way - and it paired instantly.

I am still at a loss at what to do to get the lighbulbs to pair with Z2M.

FranciscoSGPa, 2025-01-18:
Same issu happens to me. I had old hass OS and core and Zigbee2MQTT 1.42.0 and at least I could see it but pairing was unsuccesfull. With latest OS, core and  Zigbee2MQTT to version 2.0.0-2 I can not even see it.

gvlekke, 2025-02-25:
maybe related? https://github.com/Koenkk/zigbee2mqtt/issues/25438

closed 2025-05-05 by github-actions[bot]