Koenkk/zigbee2mqtt #25399: TS011F_plug_3 - Incorrect calculation of energy consumption

tOOmki, 2024-12-31:
### What happened?

I have 3 sockets that have been reporting inflated energy consumption values ​​for about 3-4 months. Power, voltage and current values ​​are correct. All are version TS011F_plug_3 20B+TZSKT41BS101. 
[image]
[image]

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20240710

### Adapter

SLZB-06M

### Setup

Proxmox container

mrmaximas, 2025-01-02:
please show the messages from this device in the z2m log. it looks like:
2025-01-02 14:31:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Test Plug L', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":0.23,"indicator_mode":"off/on","last_seen":"2025-01-02T14:31:13+03:00","linkquality":83,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":80,"latest_version":80,"state":"idle"},"voltage":230}'

tOOmki, 2025-01-02:
Logs from the 3 sockets with which I have a problem:

`info 2025-01-02 12:34:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt_kanciapa/PP_LAP_Martyna', payload '{"child_lock":"UNLOCK","current":0.08,"energy":42949696.83,"indicator_mode":"off/on","linkquality":76,"power":4,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":225}'`

`info 2025-01-02 12:33:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt_kanciapa/PP_KMP_Agata', payload '{"child_lock":null,"current":0.14,"energy":42949840.49,"indicator_mode":"off/on","linkquality":21,"power":4,"power_outage_memory":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":240}'`

`info 2025-01-02 12:33:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt_kanciapa/PP_KMP_Marta', payload '{"child_lock":null,"current":0.05,"energy":42949854.9,"indicator_mode":"off/on","linkquality":10,"power":3,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":238}'`

tOOmki, 2025-01-02:
I managed to trace this jump in the influx database:
[image]

mrmaximas, 2025-01-02:
yes, seems like converter issue, not ha.
please post the database.db entry for this device

mrmaximas, 2025-01-02:
have you tried updating the firmware on these plugs?

tOOmki, 2025-01-02:
There are no updates available from zigbee2mqtt.

In total, I have 12 TS011F_plug_3 sockets. I only have problems with these.

[image]

mrmaximas, 2025-01-03:
Are all these plugs of yours _TZ3000_r6buo8ba or not?
I bought the cheapest tuya zigbee hub for such purposes, at the same time I'm intercepting newer firmwares to publish it here.

tOOmki, 2025-01-03:
I only have 3 pieces _TZ3000_r6buo8ba.

The rest are, and they work fine:
_TZ3000_5f43h46b: 5
_TZ3000_u5u4cakc: 4
_TZ3000_gvn91tmx: 4
_TZ3000_au1rjicn: 3
_TZ3000_okaz9tjs: 2
_TZ3000_g5xawfcq: 1
_TZ3000_fllyghyj: 1
_TZ3000_xr3htd96: 1

I ordered a Tuya gateway yesterday. Maybe I can fix something with the gateway.

mrmaximas, 2025-01-03:
> I ordered a Tuya gateway yesterday. Maybe I can fix something with the gateway.

thanks!
If you take the time to study it https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html , it will be useful to all of us. 
it's not trivial, but I did it ;)

tOOmki, 2025-01-03:
I am analyzing the database.db file and I am curious why other sockets (_TZ3000_u5u4cakc:) have so little code, for example:

`{"id":2,"type":"Router","ieeeAddr":"0x540f57fffefdd331","nwkAddr":11112,"manufId":4098,"manufName":"_TZ3000_u5u4cakc","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���.\u0016","65506":31,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_u5u4cakc","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":238,"rmsCurrent":50,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":27877}},"genOnOff":{"attributes":{"tuyaBacklightMode":1,"moesStartUpOnOff":1,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1735834311380}`

However, these problematic sockets have a lot of entries:
`"seMetering":{"attributes":{"37":"","293":"","549":"","805":"","1061".........`
`"seMetering":{"attributes":{"37":{"type":"Buffer","data":[]},"293":{"type":"Buffer","data":[]},"549":{"type":"Buffer","data":[]},"805":.........`
`"seMetering":{"attributes":{"37":0,"293":0,"549":0,"805":0,"1061":0,"1317":0,"1573"..............`

tOOmki, 2025-01-12:
I bought a Moes gateway. I paired the sockets, but it doesn't count the energy consumption. The current, power and voltage data are ok. When deleting a socket in the application, there is an option to clear data - but that didn't help either.
[image]
[image]

mrmaximas, 2025-01-12:
The total power consumption should be reported with the 0.01kWh increment.
Do you have account on iot.tuya.com??

tOOmki, 2025-01-12:
I have an account

tOOmki, 2025-01-24:
up

closed 2025-04-03 by github-actions[bot]

WybeBosch, 2025-04-23:
@tOOmki did you end up figuring this one out?

Im having similair issues with the 

vendorname: Blitzwolf SHP15
model:     TS011F_plug_3
version: _TZ3000_u5u4cakc

except its a different version than yours, mine with problems is _TZ3000_u5u4cakc, 
while your with problems is _TZ3000_r6buo8ba.

For me im not having issues with "energy" total reading, but the W "power" live reading is about 200% What it should be (?)

------------------------

I just recently bought these, and i did a test with a water kettle for which i know the specs (1850-2200W)
The smart elecricity module of my entire house (known good) reported 1952.4w
While the zigbee plug that the kettle was plugged into reported ~4048–4053 W

tOOmki, 2025-04-23:
I gave up. I bought new ones.
I only had 3 to replace.
Two methods of resetting the counter found in web didn't work.
I bought a Tuya gateway - it also couldn't reset the counter.

BugBuster1701, 2025-09-02:
@WybeBosch 
> For me im not having issues with "energy" total reading, but the W "power" live reading is about 200% What it should be (?)
>
> I just recently bought these, and i did a test with a water kettle for [...]

I have the same Problem, the "current" (A) displays the double value, so the "load_power" (W) is also the double value.

I use ioBroker with Zigbee Adapter 2.0.5.
Is there a solution?