Koenkk/zigbee2mqtt #25402: OTA updates with Zigbee2MQTT almost always fail

KatyPurry, 2025-01-01:
### What happened?

Hello,

I was able to pair my first devices without any problems. I tried to update the equipment (LIDL and Ikea sockets). The first OTA update went through without a hitch a few days ago (it took 10 or 15 minutes).

But since yesterday, no more OTA updates have been able to finish.I get errors after a few minutes of updating. I can restart (it resumes at the same percentage) but it re-crashes a few minutes later. 

- I’ve changed the USB extension cord
- I’ve changed the USB port
- I’ve restarted HomeAssistant/Zigbee2MQTT a lot of time / between errors, etc.
- I’ve unplugged/replugged the outlets between errors, etc.
- I’ve moved the dongle.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.5.0 build 0

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E (tested EZSP and Ember)

### Setup

Home Assistant VM on Unraid

### Debug log

[log(1).log](https://github.com/user-attachments/files/18284075/log.1.log)

bobo-cher, 2025-01-01:
I have the same issue, which I reported in a discussion thread [EFR32MG21 and CC2652P usage #14261](https://github.com/Koenkk/zigbee2mqtt/discussions/14261#discussioncomment-11665266). I repeat the important parts of posts from there.

I have Sonoff Dongle-E with EFR32MG21 and use Ember driver with Z2M. All devices work well (48 devices). I have however one specific problem: OTA updates are usually fail to install. Currently 18 devices want to update to a newer firmware. During update (which normally takes 15 minutes) errors occur and the adapter is disconnected/reconnected followed by a restart of Z2M. After many attempts I've managed to update only few devices. I believe updates did work better when I had less devices in my network.

I tested many 7.x firmwares and all have the same issue (connection to coordinator getting lost during OTA updates). As a last resort tried old 6.10.3 firmware (ncp-uart-sw_EZNet6.10.3_V1.0.1.gbl) with ezsp driver. With this firmware OTA updates work without issues. Updated 6 devices without a single glitch.

Sonoff Dongle-E initially came with 6.x firmware (I'm not sure what version exactly it was), which I updated to 7.4 before use.

I would rather use a recent 7.x firmware but not working OTA updates is an issue.

I tried all itead (Sonoff) firmwares and also 7.4.1-7.4.5 from https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee.

I also tried 8.0.2 firmware and got the same issue.

The only firmware where OTA updates work without issues is 6.10.3 from itead.

OTA updates did work with 7.4.x previously, when I had only few devices. But now, with 48 devices in the network, OTA updates almost always fail. They might work if I do a lot of attempts.

- - -

I did a test with a second dongle on a second computer. The first dongle on first PC was disconnected during this test. Only one device was paired to the second dongle.  OTA updates did work in this test, even when using 7.4.x firmware.

Then I did another test: on the second pc/second dongle I used network key and pan id/extended id from first dongle/pc. OTA updates fail, although only one device is registered in z2m.

I think, when network key and pan id are reused on the second dongle, that makes the coordinator to accept/process packets sent by all devices in the house, even though they are not listed/visible in z2m on the second dongle/pc.

Thoughts after this test: I think 7.x firmware has difficulties processing all traffic in my network during OTA updates. I don't think it's a problem with one particular device. Difficulties with OTA updates sequentially increased with the grow of my network until they reached a point, where OTA updates become impossible at all.

- - -
I'm now using firmware 6.10.3 for the last 5 days and haven't encountered any issues with my devices so far. I must say the network works even better than with 7.4.4 before. I had occasionally messages "device xxx become unavailable" followed by a proper device status 15 seconds later. I haven’t had a single "become unavailable" message after downgrading of firmware. Not mentioning OTA updates, which now work flawlessly.

- - -
I don't think z2m is responsible for the issue. That's why I haven’t created an issue myself but I thought I'll add my feedback after the issue was created by other user.

bobo-cher, 2025-01-01:
My devices:

Total 48

By device type
Router: 31
End devices: 17

By power source
Mains (single phase): 31
unknown: 9
Battery: 8

By vendor
SONOFF: 27
_TZ3000_5iixzdo7: 6
_TZ3000_mrpevh8p: 4
eWeLink: 3
_TZ3000_w0qqde0g: 2
NodOn: 1
_TZ3000_lf56vpxj: 1
_TZE204_nklqjk62: 1
_TZ3000_e3vhyirx: 1
LUMI: 1
_TZ3000_gjnozsaz: 1

By model
ZBMINIR2: 19
SNZB-05P: 7
TS130F: 7
TS0041: 4
TS011F: 3
SNZB-02P: 2
SIN-4-RS-20: 1
SNZB-02D: 1
TS0202: 1
TS0601: 1
SNZB-04P: 1
lumi.remote.b1acn01: 1

KatyPurry, 2025-01-01:
Thank you for your answer @bobo-cher 

Now that you've raised this point, I remember that the first OTA update I did, I'd just unpacked the key and it was version 6 (I can't remember exactly which one). 

And indeed, it was from the moment I switched to 7.4.4 that I had problems (7.4.5 gives the same results), I haven't tested versions 8.0.X but it seems to be the same thing from what you say.

It's still a mystery, but it's a step in the right direction!

bobo-cher, 2025-01-03:
@KatyPurry It would be interesting to see if 6.10.3 solves OTA updates issues for you too. I use [Web-flasher](https://darkxst.github.io/silabs-firmware-builder/) and select last option (custom firmware / select file). You can download Sonoff firmware 6.10.3 here: https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/NCP.

KatyPurry, 2025-01-09:
> @KatyPurry It would be interesting to see if 6.10.3 solves OTA updates issues for you too. I use [Web-flasher](https://darkxst.github.io/silabs-firmware-builder/) and select last option (custom [...]

Sorry for the delay, I'll try ASAP with the 6.10.3 firmware !

KatyPurry, 2025-01-09:
OK, I tested firmware 6.10.3 with an Ikea outlet : https://www.zigbee2mqtt.io/devices/E1603_E1702_E1708.html#ikea-e1603%252Fe1702%252Fe1708

The operation was a complete success! There were no mistakes !
[image]

bobo-cher, 2025-01-12:
This issue is discussed in details in #23861. There is even already a solution in testing.

I suggest to close current issue as duplicate.

Nerivec, 2025-01-12:
Closing, per message above. 👍

closed 2025-01-12 by Nerivec