Koenkk/zigbee2mqtt #25403: OTA UI does not report correct firmware version

rsyring, 2025-01-01:
### What happened?

There is a persistent issue in the UI where the firmware version is not updated after an OTA update succeeds.  Related issues:

* https://github.com/Koenkk/zigbee2mqtt/issues/5869
* https://github.com/Koenkk/zigbee2mqtt/issues/6579
* https://github.com/Koenkk/zigbee2mqtt/issues/8526
* https://github.com/Koenkk/zigbee2mqtt/issues/14891
* https://github.com/Koenkk/zigbee2mqtt/issues/16670
* https://github.com/Koenkk/zigbee2mqtt/issues/17853
* https://github.com/Koenkk/zigbee2mqtt/issues/23279

This obviously causes a lot of problems and troubleshooting is not straightforward b/c the bug leads one to believe the firmware update didn't succeed.  It can be a long road to figure out that it's actually a stale UI problem.

## Workaround to Force UI Update

For those finding this issue in the future, a workaround that causes Z2M to read the correct value is detailed at the bottom of:

https://help.inovelli.com/en/articles/8377465-blue-series-how-to-update-firmware-with-home-assistant-zigbee2mqtt 

# My Inovelli Switches

In my specific instance, after updating an Inovelli Blue switch to firmware 2.18, the UI continued to show the previous version (2.15):

[image]

Not from the screenshot:

* Entry 9 has been updated to 2.18 and the workaround described above has been performed.  This is the correct UI.
* Entry 10 is truly at version 2.15 and still needs to be updated.
* Entry 11 is actually at version 2.18.  A tell-tale sign that this switch is showing the wrong version in the UI is that the button to says "Check for new updates" instead of "Update device firmware."

### What did you expect to happen?

I expected the Z2M UI to show the firmware version 2.18 after successfully updating an Inovelli Blue switch to firmware 2.18.

### How to reproduce it (minimal and precise)

It's currently happening consistently for all these switches.  But I have no idea how to provide an example, sorry.

### Zigbee2MQTT version

1.42 (and many previous)

### Adapter firmware version

7.1.1.0 build 273

### Adapter

N/A

### Setup

home assistant add-on

mrmaximas, 2025-01-03:
Hi!
I seem to have a similar problem with a completely different device.
https://github.com/Koenkk/zigbee-OTA/issues/638

wduda, 2025-01-30:
I am experiencing the same with Aqara E1 Thermostats.

Nerivec, 2025-02-02:
Can you confirm if a re-interview (blue `i` button) updates the values to expected?
_The end of an OTA update is tricky, because every device/network is different, how long it takes to report back, how it reports back... for this reason, Z2M does not enforce a check of the new values (would fail too much, even though updates actually succeeded). The interview button allows to bypass the Z2M caching mechanism and retrieve the latest values. In some cases, for some devices, it is the only way to reliably get updated information. For battery devices, make sure to trigger them before (push the button), to wake them up, to avoid interview failure._

PS: same device model does not necessarily means same actual device, some have different firmware, so, sometimes versions can vary from one device to another.

rsyring, 2025-02-03:
> Can you confirm if a re-interview (blue i button) updates the values to expected?

In my experience, the re-interview button alone is not sufficient to cause the firmware version to update.  The process I used was:

* Click on Dev console -> Endpoint: 1 -> cluster: genBasic -> attribute (s): swBuildId, dateCode-> read
* Re-interview device

Nerivec, 2025-02-03:
The "interview" blue `i` button (_not the orange "re-configure"_) does just that, it re-retrieves all reported genBasic attributes
https://github.com/Koenkk/zigbee-herdsman/blob/e986201434cef3e5f4f5afdc800203c2d151a5d4/src/controller/model/device.ts#L271-L282
https://github.com/Koenkk/zigbee-herdsman/blob/e986201434cef3e5f4f5afdc800203c2d151a5d4/src/controller/model/device.ts#L933

rsyring, 2025-02-03:
> The "interview" blue i button (not the orange "re-configure") does just that, it re-retrieves all reported genBasic attributes

I see.  It's possible I was clicking the re-configure button instead.  It's been awhile since I did this.  Next time firmware updates are available I'll experiment with just using the interview button.

mrmaximas, 2025-02-03:
@wduda hi! I just updated Aqara E1 TRV from 0.0.0.0_0010 to 0.0.0.0_0030 with fw date Feb 20 2023 but I see you have 0.0.0.0_1030. what is the latest and correct firmware???

[image]

rsyring, 2025-02-03:
> [@wduda](https://github.com/wduda) hi! I just updated Aqara E1 TRV from 0.0.0.0_0010 to 0.0.0.0_0030 with fw date Feb 20 2023 but I see you have 0.0.0.0_1030. what is the latest and correct firmware???
> <img alt="Image" width="1145" src="https://private-user-images.githubusercontent.com/61165387/409240907-3fd225fa-a345-42b4-b432-d8333758585e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzg2MDk4MDksIm5iZiI6MTczODYwOTUwOSwicGF0aCI6Ii82MTE2NTM4Ny80MDkyNDA5MDctM2ZkMjI1ZmEtYTM0NS00MmI0LWI0MzItZDgzMzM3NTg1ODVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjAzVDE5MDUwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVhYzJkZmUzYWM0NDcwNzE5MTJiZTk5ZTBmZmJkZGU2NTE3NzhlOTNmYTkwM2I1MDhiNDRhOTBlMmZkZWQ4OTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.51ubtYCXyw8I7TMO64wxA2I7RMaQvgnjUwuzkZTDAQU">

@mrmaximas this question is a bit off-topic for this issue.  Would you mind moving this conversation to a separate [discussion](https://github.com/Koenkk/zigbee2mqtt/discussions)?  Just would like to keep this issue focused on the UI update issue.  :)  Thanks.

wduda, 2025-02-10:
> Can you confirm if a re-interview (blue `i` button) updates the values to expected? _The end of an OTA update is tricky, because every device/network is different, how long it takes to report back, [...]

Hey, when I tested I used the interview button. However on some of the devices, it would give me a message that the interview was successful, but they would still report an older version of firmware even after the OTA update was complete. It really felt as if there is some outdated caching of the data either on the side of the thermostats or Z2M.

I unfortunately do not recall if I correctly woke them up, but they are battery powered.

Not sure if I mentioned it before, but after a few days I started seeing this - all versions up to date, just not all devices reporting firmware build date:

[image]

I can try and check if interviewing after wake-up will refresh the build date on those where it is missing.

wduda, 2025-02-10:
I just gave it a try with one of them, the log says:

```
2025-02-10 12:41:04z2m: Interviewing 'Thermostat - Living Room'
2025-02-10 12:42:40z2m: Successfully interviewed 'Thermostat - Living Room'
```
But I still see now firmware build date and only the following clusters show up for that device:

[image]

wduda, 2025-02-10:
And now I am going crazy, after a second interview the build date shows up but firmware version goes down:

[image]

Trying to check if I can update again gives me:

`error 2025-02-10 12:46:21z2m: Failed to check if update available for 'Thermostat - Living Room' ([0x54ef441000a5a649 | lumi.airrtc.agl001] Failed to find an endpoint which supports the OTA cluster)`

mrmaximas, 2025-02-10:
@Nerivec hi! seems to be a similar problem https://github.com/Koenkk/zigbee2mqtt/issues/26292

vogtmh, 2025-02-10:
`{
    "battery": 65,
    "last_seen": "2025-02-10T19:41:14+01:00",
    "linkquality": 98,
    "update": {
        "installed_version": 604241926,
        "latest_version": 604241926,
        "state": "idle"
    }
}`

My Ikea dimmer has the same issue. Yes, there are many devices out there that behave differently at the end of the update. But actually, the state is correctly reported by the device each time and Z2M only needs to reflect this in the UI.

Koenkk, 2025-02-10:
Does restarting z2m fix this? If not, can somebody provide me a debug log of the whole update procedure where it does not update afterwards?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

vogtmh, 2025-02-10:
I don't think it's an issue of the update. It's an issue of displaying the new version and only happens for the Ikea dimmers E1743 in my setup. I restarted Z2M and even Homeassistant without success. Even removing and readding the device will not always show the correct version. However, some identical buttons do show the correct version, it seems to be random.

wduda, 2025-02-11:
It seems to not be connected to the update alone. For my Aqara SRTS-A01 the firmware data alternates between 0.0.0_0030 and 0.0.0_1030, with sometime the error message of no OTA cluster, without a pattern discernable to me when and what affects what version is reported.

[image]

rsyring, 2025-02-11:
Are you guys also trying the manual refresh method?

*  Click on Dev console -> Endpoint: 1 -> cluster: genBasic -> attribute (s): swBuildId, dateCode-> read
* Reconfigure device: Click on About (for that device) reconfigure button (not interview button)

That has never failed to get the updated firmware version for me from my Aqara (LUMI) temperature devices.  I understand from the comments above that we are trying to see if re-interviewing the device will work without the manual process above.  But if you are finding it does not, and the manual method does work, that is useful information.  

My devices haven't needed a firmware update again yet.  When they do I'll try the interview button first.

> Does restarting z2m fix this? 

It did not for me previously.  The above method is the only thing that resolved it for me.

wduda, 2025-02-12:
> Are you guys also trying the manual refresh method?
> 
> * Click on Dev console -> Endpoint: 1 -> cluster: genBasic -> attribute (s): swBuildId, dateCode-> read
> * Reconfigure device: Click on About (for that device) reconfigure button (not interview button)
> 

This does help with some of my devices, and the reconfigure aligns he version shown in OTA UI with what the dev console read reports.

But at the same time not with all devices, because some devices also mess up the list of available clusters and I see this, and the manual query of genBasic from dev console iss impossible:

[image]

vogtmh, 2025-02-12:
> Are you guys also trying the manual refresh method?
> 
> * Click on Dev console -> Endpoint: 1 -> cluster: genBasic -> attribute (s): swBuildId, dateCode-> read
> * Reconfigure device: Click on About (for that device) reconfigure button (not interview button)

I just tried the manual refresh. It's reporting the wrong version, too.

"installed_version": 604241926
Version reported: 25.4.5 (which equals 604241925)

However before I tried the manual way, some of my updated devices suddenly reported the old version again and I was offered an update. Not sure what is going on here. It only happens for my E1743 devices, all other Ikea devices seem to update reliable.

wduda, 2025-02-16:
A follow up to my last message:

- manual check reports and refreshes version to 0.0.0_0030
- after leaving the devices running for a few days, they all claim to be 0.0.0_1030 in OTA view
- in the screenshot you can see how manual dev console call and reconfigure set one device back to 0.0.0_0030
- no manual attempts at OTA updates were made 

[image]

mrmaximas, 2025-02-21:
I have same issue > manual check reports and refreshes version to 0.0.0_0030
```
 "update": {
        "installed_version": 2590,
        "latest_version": 2590,
        "state": "idle"
    },
<img 
```width="1452" alt="Image" src="https://github.com/user-attachments/assets/36b1292c-ab55-435a-a464-e207ef8f751c" />

Nerivec, 2025-02-21:
Aqara is known for messing up file version in firmware from time to time (decreasing instead of increasing), so, this is not really surprising... _They just don't follow Zigbee specification in general..._

> At the end of the update process of each TRVs, I saw an error

Looks like the converter may need an update to properly support this new firmware.
I don't use Aqara stuff, so, can't help much there.

mrmaximas, 2025-02-21:
> Looks like the converter may need an update to properly support this new firmware.

For some reason I'm not sure if the z2m repo has the latest fw. Maybe my mistake was to update TRVs via z2m. But it is strange that after this update via z2m I transferred TRV to my Aqara testing hub to check the version and it still shows 0.0.0_1030 there. May be @Koenkk can help us?
UPDATE: 
it's mystical, but one of the TRVs (which I didn't move to the Aqara hub to check the version) is showing again 0.0.0_1030 in z2m.
[image]

wduda, 2025-02-21:
> For some reason I'm not sure if the z2m repo has the latest fw. Maybe my mistake was to update TRVs via z2m. But it is strange that after this update via z2m I move TRV to my Aqara testing hub to check the version and it still shows 0.0.0_1030 there.

This is what I see without the Aqara hub even, just need to wait 1-2 days and Z2M will also show 0.0.0_1030 again.

mrmaximas, 2025-02-21:
> This is what I see without the Aqara hub even, just need to wait 1-2 days and Z2M will also show 0.0.0_1030 again.

Yes, but then it raises questions. If we request a version via z2m debug consile, we get 30, if we wait, we got 130. It is also unclear why the update does not start from 0%. Moreover, if i request now an update of the one with version 130, z2m will offer to update it.

wduda, 2025-02-21:
> Yes, but then it raises questions. If we request a version via z2m debug console, we get 0030, if we wait, we got 1030

It seems almost as if there is some caching in the communication with those thermostats, once the cache's TTL is over it shows the new version. But also querying gives the old version.

I am wondering if the implementation by Aqara/Lumi is just bad.

uffevedenbrant, 2025-04-07:
I had the same issue. 3 of 4 SNBZ-02D reports correct data in OTA tab. The last one reported the old version but when checking it was correct. Followed the hints in Issue #14891 on how to make the OTA tab show the correct version and it worked.
But.. In my case i upgraded all 4 devices on my test HA ( same setup with Z2M , same HW etc ). All devices there reported the correct OTA data. Moving to my prod HA 3 shows correct and the last the wrong version. Moving back the funny device to my test HA and it shows the correct version , moving back to prod HA and it shows the wrong version. tested both with nice remove and then force remove in prod HA and same result.

Is this data in some way cached somewhere even if a device is removed , either nice or force removal? 

I.e. that it reappear with old versions etc for a specific device despite that i could see that the reported versions was correct.
Like     "installed_version": 8960,         "latest_version": 8960

But in OTA UI it reported the old values..
As mentioned this worked out for me to get the correct data.
1. From the dev console for the device select genBasic / swBuildId and read the value (returns the correct firmware version below).
2. From the device 'main page' select the yellow 'reconfigure' button. The correct firmware version and build date will show.

Makes me suspect that there is a miss somewwre in the interview where data is dropped , not asked or missed or believed to be in a cache from an old interview..

Ricc68, 2025-05-03:
Fixed by mentioned PR https://github.com/nurikk/zigbee2mqtt-frontend/pull/2470.

KennethLavrsen, 2025-05-03:
Ricc68 fixed a different problem with the referenced PR. This issue relates to not showing the new revision after an OTA upgrade. And we are not sure if it is because the OTA failed or it just does not load the new revision from the device.

Ricc68, 2025-05-03:
There are two sets of versioning in a firmware:

- a "firmware version" and "firmware build date" that are parameters values reported by firmware, these parameters are not mandatory so a manufacturer can chose to not report them: this is the reasons some devices or also some firmware versions/build dates are not reported
- a current running "file version" that is a state which is reported by the firmware, this is mandatory if the devices want to run OTA updates. The same "file version" is present also in the new firmware file and the two are compared when checking for the presence of an OTA

With this PR the versions used are the file versions which must be present if the device supports firmware updates, therefore ensuring that the OTA page is always populated with firmware running version and new firmware available version.

So when you compare the "firmware versions" and "build dates" you don't know if there was a mistake in the code (static version numbers).
When you compare "file versions" you are sure these are the proper version values otherwise the firmware update won't happen.
If the same "file versions" are shown after an update means that either the page somehow did not update or the firmware update didn't happen.

The PR may not fix this, I agree, but surely will give better hints on what's going on with this issue.

KennethLavrsen, 2025-07-03:
Ping the bot

rsyring, 2025-09-03:
I've not had OTA updaes for awhile so I can't test this.  But, I wouldn't want it to closed for staleness yet.