Koenkk/zigbee2mqtt #25408: [New device support]: zigbee water leakage detector

liuk84, 2025-01-01:
### Link

https://it.aliexpress.com/item/1005006966713885.html?spm=a2g0o.order_list.order_list_main.17.179136964LpsDT&gatewayAdapt=glo2ita

### Database entry

0xa4c138c7d33a1553

### Zigbee2MQTT version

1.27.2-1

### Comments

the item is not supported. i'm able to associate but not work

### External definition

```shell
the item is not supported. i'm able to associate but not work
```

### What does/doesn't work with the external definition?

the item not work

mrmaximas, 2025-01-02:
how about upgrading to the latest version of z2m?

liuk84, 2025-01-02:
i'm not able sorry. is possibile to integrate this sensor with metod manual?

mrmaximas, 2025-01-02:
> i'm not able sorry. is possibile to integrate this sensor with metod manual?

Supposedly not. Your version of z2m is too old.

closed 2025-03-11 by github-actions[bot]