Koenkk/zigbee2mqtt #25411: [New device support]: Tuya Zigbee Signal Extender

jbarriosmuriel, 2025-01-02:
### Link

https://www.loratap.com/products/rp280

### Database entry

{     "last_seen": "2025-01-02T07:08:05.650Z",     "linkquality": 255 }

### Zigbee2MQTT version

 1.34.0 commit: aae7312

### Comments

Hi

Please help me to add this signal extender device. 

Anything you need to add support, ask me.

### External definition

```shell
I can't generate "Generate external definition"

The button to do so does not appear.
```

### What does/doesn't work with the external definition?

I can't generate "Generate external definition"

The button to do so does not appear.

jbarriosmuriel, 2025-01-02:
> ### Link
> https://www.loratap.com/products/rp280
> 
> ### Database entry
> { "last_seen": "2025-01-02T07:08:05.650Z", "linkquality": 255 }
> 
> ### Zigbee2MQTT version
> 1.34.0 commit: [aae7312](https://github.com/Koenkk/zigbee2mqtt/commit/aae7312d8a9f8b543f406c5c57140e4ae342fe30)
> 
> ### Comments
> Hi
> 
> Please help me to add this signal extender device.
> 
> Anything you need to add support, ask me.
> 
> ### External definition
> ```shell
> const definition = {
>     zigbeeModel: [''],
>     model: '',
>     vendor: '',
>     description: 'Automatically generated definition',
>     extend: [],
>     meta: {},
> };
> 
> module.exports = definition;
> ```
> 
> ### What does/doesn't work with the external definition?

jbarriosmuriel, 2025-01-02:
I update to 1-42 version andI did the device interviews again and they were already supported.

mrmaximas, 2025-01-02:
please close the issue

closed 2025-03-11 by github-actions[bot]