Koenkk/zigbee2mqtt #25412: BEOK TVR-705ZB Thermostatic Radiator Valve

John-Rookie, 2025-01-02:
### Link

https://www.beok-controls.com/thermostatic-radiator-valve/zigbee-trv-actuator/zigbee-smart-home-radiator-valves.html

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0xa4c1385130f554d1","nwkAddr":41220,"manufId":4417,"manufName":"_TZE284_ltwbm23f","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"�\u000b\t/\u0011","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":77}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1735804210365}

### Zigbee2MQTT version

1.42.0

### Comments

Please support this device.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_ltwbm23f',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

nothing

magister86, 2025-01-04:
Hello,
I was able to get working this device with the following .js external converter file (only window detection does not work). It's a Moes external converter with some modifications I made:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE284_ltwbm23f',
        },
    ],
    model: 'TRV705ZB',
    vendor: 'Tuya',
    description: 'Thermostatic radiator valve.',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            e.battery(),
            e.child_lock(),
            e.max_temperature(),
            e.min_temperature(),
            e.position(),
            e.window_detection(),
            e.binary('window', ea.STATE, 'OPEN', 'CLOSE').withDescription('Window status closed or open '),
            e
                .climate()
                .withLocalTemperature(ea.STATE)
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperatureCalibration(-10, 10, 0.1, ea.STATE_SET)
                .withPreset(['off', 'antifrost', 'eco', 'comfort', 'auto', 'on'])
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withSystemMode(['auto', 'heat', 'off'], ea.STATE),
            ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
            e.comfort_temperature().withValueMin(5).withValueMax(30).withDescription('Comfort mode temperature'),
            e.eco_temperature().withValueMin(5).withValueMax(30).withDescription('Eco mode temperature'),
            e.holiday_temperature().withValueMin(5).withValueMax(30).withDescription('Antifreeze mode temperature'),
            e.enum('display_brightness', ea.STATE_SET, ['high', 'medium', 'low']).withDescription('Display brightness'),
            e.enum('screen_orientation', ea.STATE_SET, ['up', 'down']).withDescription('Screen orientation'),
            e
                .enum('hysteresis', ea.STATE_SET, ['comfort', 'eco'])
                .withDescription(
                    'Hysteresis - comfort > switches off/on exactly at reached ' +
                        'temperature with valve smooth from 0 to 100%, eco > 0.5 degrees above or below, valve either 0 or 100%',
                ),
            e.enum('motor_thrust', ea.STATE_SET, ['strong', 'middle', 'weak']),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
                [
                    2,
                    'preset',
                    tuya.valueConverterBasic.lookup({
                        off: tuya.enum(0),
                        antifrost: tuya.enum(1),
[... 53 lines cut ...]
                    }),
                ],
                [113, 'screen_orientation', tuya.valueConverterBasic.lookup({up: tuya.enum(0), down: tuya.enum(1)})],
                [114, 'position', tuya.valueConverter.divideBy10],
                [119, 'comfort_temperature', tuya.valueConverter.divideBy10],
                [120, 'eco_temperature', tuya.valueConverter.divideBy10],
                [121, 'holiday_temperature', tuya.valueConverter.divideBy10],
                [
                    127,
                    'hysteresis',
                    tuya.valueConverterBasic.lookup({
                        comfort: tuya.enum(0),
                        eco: tuya.enum(1),
                    }),
                ],
            ],
    },
    extend: [],
};

module.exports = definition;

The problem I have from the last z2m update is it seems the way to include the external converters has changes and now you can add the .js file in the configuration.yaml directly , you have to publish a message but I wasn't able to do it.

I hope the code is useful.

John-Rookie, 2025-01-06:
Oh, thanks.

_"The problem I have from the last z2m update is it seems the way to include the external converters has changes and now you can't add the .js file in the configuration.yaml directly , you have to publish a topic message (zigbee2mqtt/bridge/request/converter/save) but I wasn't able to do it."_

I've been trying for a while and haven't succeeded yet. I'm still looking for a good description for beginners.

closed 2025-03-16 by github-actions[bot]