Koenkk/zigbee2mqtt #25414: [New device support]: TZE204_byzdayie Din rail type kWh meter（1 phase 2 wire ）

Gexwh, 2025-01-02:
### Link

https://fr.aliexpress.com/item/1005005864560815.html?spm=a2g0o.order_list.order_list_main.45.601d5e5bKEKlNY&gatewayAdapt=glo2fra

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0x70ac08fffeaa5e62","nwkAddr":7175,"manufId":4098,"manufName":"_TZE204_byzdayie","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000i","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_byzdayie","stackVersion":0,"dateCode":"","appVersion":70,"zclVersion":3,"powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1735749097332}

### Zigbee2MQTT version

1.42.0-2

### Comments

Hello,
I bought this device, thinking it was supported as dds238-2 was. However, no luck. I tried to generate a converter but i am a little bit stuck. Anybody has the datapoints?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_byzdayie',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

This is what i generated. 
Datapoint 1 ist true/false
14 & 20 seems to be voltage

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'DDS238-1-Z1',
    vendor: 'Hiking',
    description: 'Zigbee Single Phase 63A Energy Meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [
        {
            key: ['state'],
            convertSet: async (entity, key, value, meta) => {
                const dp = 16; // Relay Switch
                const state = value.toLowerCase() === 'on' ? 1 : 0;

                // Vérifier et utiliser le logger
                if (meta && meta.logger) {
                    meta.logger.info(`Setting state to ${value.toUpperCase()} (dp=${dp}, state=${state})`);
                }

                // Envoyer la commande au datapoint Tuya
                await tuya.sendDataPointBool(entity, dp, state);

                // Retourner l'état mis à jour
                return {state: {state: value.toUpperCase()}};
            },
        },
    ],
    exposes: [
        e.numeric('1', ea.STATE),
        e.numeric('11', ea.STATE),
        e.numeric('12', ea.STATE),
        e.numeric('13', ea.STATE),
        e.numeric('14', ea.STATE),
        e.numeric('17', ea.STATE),
        e.numeric('18', ea.STATE),
        e.numeric('19', ea.STATE),
        e.numeric('20', ea.STATE),
    ],
    meta: {
        tuyaDatapoints: [
            [1, '1', tuya.valueConverter.raw],
            [11, '11', tuya.valueConverter.raw],
            [12, '12', tuya.valueConverter.raw],
            [13, '13', tuya.valueConverter.raw],
            [14, '14', tuya.valueConverter.raw],
            [17, '17', tuya.valueConverter.raw],
            [18, '18', tuya.valueConverter.raw],
            [19, '19', tuya.valueConverter.raw],
            [20, '20', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

Gexwh, 2025-01-05:
Hello, i got it worked with:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_byzdayie',
        },
    ],
    model: 'TS0601_DDS238-1-Z1',
    vendor: 'Tuya',
    description: 'Zigbee Single Phase 63A Energy Meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
exposes: [
    tuya.exposes.switch(),
    e.current().withAccess(ea.STATE).withUnit('A').withDescription('Current'),
    e.power().withAccess(ea.STATE).withUnit('W').withDescription('Power'),
    e.voltage().withAccess(ea.STATE).withUnit('V').withDescription('Voltage'),
    e.produced_energy().withUnit('kwh').withDescription('Produced energy')
],

    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff], // DP 1: Switch
            [17, 'total_energy', tuya.valueConverter.divideBy1000], // DP 17 Add Electricity
            [18, 'current', tuya.valueConverter.divideBy1000], // DP 10: Current in mA -> A
            [19, 'power', tuya.valueConverter.divideBy10], // DP 19: Power in dW -> W
            [20, 'voltage', tuya.valueConverter.divideBy10], // DP 20: Voltage in dV -> V
        ],
    },
};

module.exports = definition;

closed 2025-03-16 by github-actions[bot]

daimonkor, 2025-03-16:
@Gexwh  hello, did you check the last converter?

Gexwh, 2025-03-16:
Not yet, it worked fine for me,
I will recheck with the new update