Koenkk/zigbee2mqtt #25426: TS0001 _TZ3000_jsfzkftc - won't return energy, power, current & voltage

chanwee77, 2025-01-03:
### What happened?

Buy a new 20A switch with power monitor but it do not show any energy info after paired.

Here's the info
Zigbee Model : TS0001
Zigbee Manufacturer : _TZ3000_jsfzkftc
Manufacturer : Tuya

External definition
[image]

### What did you expect to happen?

I expected it to report Power, Current, Voltage & Energy. I can turn the relay on and off, adjust the switch type and power outage memory. I can even adjust "Settings (specific)" options such as power, current and voltage calibration

### How to reproduce it (minimal and precise)

pair it

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ezsp

### Setup

HA Raspberry Pi 4B

Koenkk, 2025-01-03:
Fixed!

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow

closed 2025-01-03 by Koenkk

chanwee77, 2025-01-04:
updated Zigbee2MQTT to 2.0.0-1 but still having the same issue
[image]

Koenkk, 2025-01-05:
Sorry I did not merge https://github.com/Koenkk/zigbee-herdsman-converters/pull/8556 yet, will do it later today.