Koenkk/zigbee2mqtt #25427: Errors with Sinope SW2500ZB switches

raetha, 2025-01-03:
### What happened?

I recently replaced my coordinator with an HA Connect ZBT-1 and repaired all devices. Most things seem to work, however I'm seeing frequent errors logged for a number of things with my Sinope SW2500ZB switches. Example errors are in the debug logs section. I've tried reconfiguring the devices hoping that would clean things up, however one of the errors happens on every reconfigure, and an interview also does not seem to eliminate them.

Other than being unable to set Power-On Behavior, functions seem to work correctly. It just seems like something is missing in the config for this device model, or my instance is somehow corrupt.

Please let me know any other information I can provide, or things to keep trying.

### What did you expect to happen?

I expect supported devices to have all functions and controls work without an error message.

### How to reproduce it (minimal and precise)

In my case, the errors are generated when hitting an exposed control that isn't liked, or attempting a reconfiguration.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.4 [GA}

### Adapter

Home Assistant Connect ZBT-1

### Setup

Add-on on Home Assistant OS as an OVA VM on TrueNAS

### Debug log

2025-01-02 22:50:23z2m: Failed to configure 'Sinope Switch', attempt 3 (Error: ZCL command 0x500b914000057eb5/1 seMetering.configReport([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"instantaneousDemand"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":100,"attribute":"currentSummDelivered"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17) at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:1981:29) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:206:21) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))

2025-01-02 22:53:34z2m: Publish 'get' 'power_on_behavior' to 'Sinope Switch' failed: 'Error: ZCL command 0x500b9140000575cd/1 genOnOff.read(["startUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

raetha, 2025-01-05:
As a follow up, I’ve upgraded to version 2 and there was no change in the error messages.

raetha, 2025-03-08:
This issue is still occurring on the most recent version. Not stale.

vinnygambiny, 2025-03-08:
Same error with the 2.1.3

jstebenne, 2025-06-15:
I have the same issue with those other Sinopé devices:

- `DM2500ZB`
- `RM3250ZB`

raetha, 2025-08-15:
Still not stale. Even an explanation on why this is occurring would be helpful.