Koenkk/zigbee2mqtt #25429: [New device support]: Zigbee Motor Curtain _TZE204_a2jcoyuk

hsimm, 2025-01-03:
### Link

https://www.aliexpress.com/item/1005007956016252.html?spm=a2g0o.order_list.order_list_main.16.2f9b1802Q9fKoZ

### Database entry

{"data":{"friendly_name":"0xa4c1380d9bd7c54f","ieee_address":"0xa4c1380d9bd7c54f","status":"successful","supported":false},"type":"device_interview"}

### Zigbee2MQTT version

 1.42.0

### Comments

This is the data points investigation outcome through Tuya Gateway:
[Based on the new Tuya device instructions](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html):

{"1":"Control","2":"Curtain position setting","3":"Current curtain position","5":"Motor Direction","6":"Auto Power","7":"Work State","11":"Situation_set","12":"Fault"}

[image]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE204_a2jcoyuk'}
    ],
    model: 'TS0601_cover_dev',
    vendor: 'Tuya',
    description: 'Curtain motor',
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.enum('motor_direction', ea.STATE_SET, ['normal', 'reversed']).withDescription('Set the motor direction'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'control', tuya.valueConverterBasic.lookup({OPEN: tuya.enum(0), STOP: tuya.enum(1), CLOSE: tuya.enum(2), CONTINUE: tuya.enum(3)})], // <- mapped the datapoint
            [2, 'percent_control', tuya.valueConverter.raw],
            [3, 'percent_state', tuya.valueConverter.coverPosition],
            [5, 'motor_direction', , tuya.valueConverterBasic.lookup({forward: true, back: false})],
            [6, 'auto_power', tuya.valueConverter.trueFalse1],
            [7, 'work_state', tuya.valueConverterBasic.lookup({opening: tuya.enum(0), closing: tuya.enum(1)})],
            [11, 'situation_set', tuya.valueConverterBasic.lookup({opened: tuya.enum(0), closed: tuya.enum(1)})],
            [12, 'fault', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

created external definition as ext_converter.js
added this file via zigbee2MQTT UI
[image]

device still showing up as unsupported when pairing

hsimm, 2025-01-09:
@Koenkk are you able to assist here?

Koenkk, 2025-01-09:
Could you try to:
- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE204_a2jcoyuk` to `_TZE284_udank5zs`
- Start Z2M and see if the device now works

hsimm, 2025-01-10:
Thanks @Koenkk , it's working now!
[image]

Koenkk, 2025-01-10:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2025-01-10 by Koenkk