Koenkk/zigbee2mqtt #25430: SONOFF Zigbee 3.0 USB Dongle Plus-P shown as unknown

marcszy91, 2025-01-03:
### What happened?

My SONOFF Zigbee 3.0 USB Dongle Plus-P shown as unkown and i cant enable the pairing mode

[image]

### How to reproduce it (minimal and precise)

I'm on the dev branch with dev-latest tag:

```yaml
services:

  mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto
    restart: always
    ports:
      - "1883:1883"
      - "9001:9001"
    environment:
      TZ: Europe/Berlin
      PGID: 100
      PUID: 1026      
    volumes:
      - /volume1/docker/mosquitto/config:/mosquitto/config
      - /volume1/docker/mosquitto/data:/mosquitto/data
      - /volume1/docker/mosquitto/log:/mosquitto/log

  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev
    restart: always
    ports:
      - 8080:8080
    volumes:
      - /volume1/docker/zigbee2mqtt/data:/app/data
      - /run/udev:/run/udev:ro
    environment:
      TZ: Europe/Berlin
      PGID: 100
      PUID: 1026
    devices:
      - "/dev/ttyUSB0:/dev/ttyUSB0"
    depends_on:
      - mosquitto      
    privileged: true
```

```yaml
version: 4
homeassistant:
  enabled: false
frontend:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto
  user: ****
  password: ****
serial:
  port: /dev/ttyUSB0
  adapter: zstack
advanced:
  channel: 20
  network_key
```

### Zigbee2MQTT version

?.?.?.?

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P 

### Setup

Docker-Compose on Synology NAS

### Debug log

Using '/app/data' as data directory

Starting Zigbee2MQTT without watchdog.

[2025-01-03 10:39:30] info: 	z2m: Logging to console, file (filename: log.log)

[2025-01-03 10:39:30] info: 	z2m: Starting Zigbee2MQTT version 1.42.0-dev (commit #2d6b00bc69d34643d0d3614a5dbd5f2451b63923)

[2025-01-03 10:39:30] info: 	z2m: Starting zigbee-herdsman (3.2.1)

[2025-01-03 10:39:30] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}

[2025-01-03 10:39:30] info: 	zh:zstack:znp: Serialport opened

[2025-01-03 10:39:49] info: 	zh:controller: Wrote coordinator backup to '/app/data/coordinator_backup.json'

[2025-01-03 10:39:50] info: 	z2m: zigbee-herdsman started (reset)

[2025-01-03 10:39:50] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20240710,"transportrev":2},"type":"zStack3x0"}'

[2025-01-03 10:39:50] info: 	z2m: Currently 0 devices are joined.

[2025-01-03 10:39:50] info: 	z2m: Connecting to MQTT server at mqtt://mosquitto

[2025-01-03 10:39:50] info: 	z2m: Connected to MQTT server

[2025-01-03 10:39:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'

[2025-01-03 10:39:50] info: 	z2m: Started frontend on port 8080

[2025-01-03 10:39:50] info: 	z2m: Zigbee2MQTT started!

geni0602, 2025-01-03:
You can try to map your device by ID
Check here:
https://www.zigbee2mqtt.io/guide/installation/01_linux.html

marcszy91, 2025-01-03:
After 2 days of researching the stick and the configuration, I found the actual problem in the browser console.
I have set up Z2M via my Synology NAS and my synology domain. Unfortunately, only the http and https protocols work there. And the frontend tries to get the data via websocket based on the domain I used to access the Z2m web interface. And the forwarding does not exist.

If I call up the page via the internal IP of my Nas it works. Is there a way to keep the domain but run the socket via localhost or the internal IP of the NAS?

closed 2025-01-03 by marcszy91