Koenkk/zigbee2mqtt #25433: [New device support]: Shelly BLU TRV

c10l, 2025-01-03:
### Link

https://shellystore.co.uk/product/shelly-blu-trv/

### Database entry

{"id":3,"type":"EndDevice","ieeeAddr":"0x694f69634762684a","nwkAddr":6129,"manufId":4098,"manufName":"Shelly","powerSource":"Battery","modelId":"BLU TRV","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,4,513,4096],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"BLU TRV","manufacturerName":"Shelly"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f25c4","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"swBuildId":"1.0.0","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1735913536058}

### Zigbee2MQTT version

1.42.0-2

### Comments

I tried to follow the guide to support new devices. Since the external definition didn't provide much functionality, I tried browsing other devices for TRV implementations but unfortunately found the code a bit outside of my comfort zone.

I'm happy to run experiments, gather logs and even trying stuff out on my side but I'm not sure how to proceed.

### External definition

```shell
const {battery} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['BLU TRV'],
    model: 'BLU TRV',
    vendor: 'Shelly',
    description: 'Automatically generated definition',
    extend: [battery()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

It reports battery level and link quality, both seemingly correctly. Nothing else.

MaJerle, 2025-01-03:
When I look at their website, they don't mention Zigbee. Is it supported?
https://kb.shelly.cloud/knowledge-base/shelly-blu-trv

c10l, 2025-01-03:
Interesting. In the link I posted, which is where I bought pretty much all my Shelly devices, it says it's supported. It does pair with the Zigbee adapter as you can see in the original post.

In the [API Docs](https://shelly-api-docs.shelly.cloud/docs-ble/Devices/trv/) it shows how to pair with Zigbee (3 button presses), which is how I did it.

I have messaged their support for clarification as well.

MSL-DA, 2025-01-04:
[image]

c10l, 2025-01-04:
Thanks!

So at the end of November support was in an unpublished alpha firmware. I imagine the latest firmware which is from about a month after that already has it, since I can pair with Z2M. I have firmware version `20241223-130807/v1.1.3@59915705`.

I just tried it in ZHA since he says it works with HA, and it does pair, recognise and configure it. Basic thermostat functionality seems to be working.

I paired it with the Shelly BLU Gateway afterwards as I wanted to try to pair the TRV with an H&T sensor. At that point, the TRV still sends sensor data (and the temperature I set on it) over Zigbee, but it ceases to receive commands (i.e. I can't set desired temp on HA).

If I first pair with the gateway, then pair the H&T sensor, then pair with Zigbee, everything _seems_ to work.

I wonder if it would be possible to use the ZHA integration code as inspiration to add this device to Z2M?

c10l, 2025-01-04:
Welp, spoke too early about everything continuing working. Got this error when trying to set desired temperature in HA this time:

```
Failed to perform the action climate/set_temperature. Failed to send request: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>```

c10l, 2025-01-06:
I heard back from Shelly support. They informed me that Zigbee is in the beta firmware. My experiments tell me it's already in the stable firmware though it's probably not officially released yet.

I asked to get hold of the beta firmware but I'm waiting for a response.

I have also asked about Z2M support and pointed them to this ticket. Whether that will trigger any action on their side or not, I don't know but it's worth a shot.

If I get access to the beta firmware I'll try the pairing again and see if there's anything more that comes out of the external definition step.

c10l, 2025-01-08:
Shelly support have now confirmed that Zigbee is available from version 1.1.x. I'm on 1.1.3 which is the latest so it should be supported.

ivdimitro, 2025-01-13:
[Here](https://community.home-assistant.io/t/new-shelly-trv-blu-integration/777837/140) the guys are working on integration to HA mainly via the gateway and MQTT. 
Not exactly ZB2MQTT discussion but can give you inspiration ;)

MeisterQ, 2025-01-24:
Any news here? Im using the latest firmware and they pair on zigbee, but doesnt work.

c10l, 2025-01-24:
Looks like it will be fully supported in Home Assistant via the native Shelly integration via BLU Gateway Gen3.

Link to forum message, with PRs: https://community.home-assistant.io/t/new-shelly-trv-blu-integration/777837/151

Unfortunately that doesn't help those of us who would rather not require the gateways as we already have Zigbee coordinators and routers in place.

I have also received this from Shelly support:

> I have just hear back from the developers that ZHA has been tested and is working, but they still have not tested it with Zigbee2MQTT.  They all also stated that to enter Zigbee mode, you must press the button beneath the device 3 times, and the refresh rate is 30 seconds.

My understanding is that via BLE the refresh rate is 8s though I can't find where I got that from. 30s may or may not be good enough for you, so something to keep in mind.

MeisterQ, 2025-01-24:
Well, i think native integration via shelly integration in HASS is not the point here. 

The Blu TRVs FW already supports full zigbee mode. Only certification is missing. So is there any converter already for the blu trv?

c10l, 2025-01-28:
> i think native integration via shelly integration in HASS is not the point here.

I never meant to imply it is. I was just adding a data point.

> So is there any converter already for the blu trv?

In the same message I mentioned Home Assistant support, I have also quoted Shelly support around ZHA and Z2M support.

Furthermore, if you're following this thread you probably already have the answer to your question.

ivdimitro, 2025-01-30:
The device connects to the network and interrogates with no issues. All the clusters from the documentation seem to be discovered, but... 
I looked to create custom converter but it seems to be way more complicated than adding key/value pairs in a json/jaml file. 
Another thing is that Shelly is not present as vendor in the github repo at all. 

The dev console works ok and I can send commands and such but does anybody have some more detailed guide how to "generate" the converter with the dev console? The auto generate buttons only creates whatever is visible in the GUI and the "Support new devices" wiki is really lacking details.

MeisterQ, 2025-04-01:
Did someone paired the blu TRV with z2m and could export a definition file?

c10l, 2025-04-01:
@MeisterQ it's on the OP. I can try again later if there's reason to believe this time it will have more information on it but I fail to see why it would.

MeisterQ, 2025-04-01:
> [@MeisterQ](https://github.com/MeisterQ) it's on the OP. I can try again later if there's reason to believe this time it will have more information on it but I fail to see why it would.

I think zigbee exposes changed because they made some bigger updates since when you created this issue.
You should do a fw update over bluetooth and then try to get the definitions again.

closed 2025-06-09 by github-actions[bot]

KC-inDomus, 2025-10-30:
Now, with TRV's firmware 1.2.10, nothing had change Z2M side. The interview still present a limit set of informations - just battery.

The very ODD thing is that ZHA (ZigBee bridge included with Home Assistant Core) pair the device with no problems at all, all the thermostat functions work fine:

[image]

no lags, no problems, everithing works fine.