Koenkk/zigbee2mqtt #25438: Unable to add Philips Hue Bloom & Iris

Rido, 2025-01-03:
### What happened?

Unfortunately, I am experiencing issues when adding the Philips Hue Bloom and Iris. Both devices triggers the following error:
```Interview failed because can not get node descriptor ('0x00178801001cf795')```

I have tried resetting them in various ways and placed them next to the coordinator, but unfortunately, this did not help.

The strange thing is that I can add other devices, like a Silvercrest smart plug, without any problems.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

Sonoff ZigBee 3.0 USB Dongle Plus (EFR32MG21)

### Setup

Add-on on virtual Home Assistant OS on Mac Mini M4 in UTM

### Debug log

[log.log](https://github.com/user-attachments/files/18303439/log.log)

Dazzel, 2025-01-07:
Having the same issue with my iris lamp. Previously worked with the Hue bridge but not paring to z2m now. Getting the same `Interview failed for '0x00178801000cc574 with error 'Error: Interview failed because can not get node descriptor ('0x00178801000cc574')'` error.

I am quite sure this is [Philips 7199960PH](https://www.zigbee2mqtt.io/devices/7199960PH.html#philips-7199960ph) lamp but cannot confirm, since the sticker on the cord is lost. The lamp has the round "click wheel" (Generation 2):
[image]

Using the Sonoff Dongle-E v2 with the 7.4.4 firmware. Running z2m as a home assistant addon (2.0.0-2).

The lamp will pair with a conbee II stick using the deconz addon in home assistant.

Habile2019, 2025-01-18:
I had the same issue so tried flashing [7.4.5](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl) powered on the Iris, re-interviewed immediately after powering on and it got added!

Dazzel, 2025-01-18:
I've tried this, but unfortunately without any success. Getting the same interview error.

gvlekke, 2025-02-04:
Light model: 7199960PH (on the sticker on the power cable)
Zigbee2mqtt version: 2.1.0-1
Adapter: using HA Yellow with build in zigbee

Restarted zigbee2mqqt and then tried again but still got this error:
```
[15:52:00] INFO: Preparing to start...
[15:52:00] INFO: Socat not enabled
[15:52:01] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-04 15:52:06] error: 	zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=RESET_WATCHDOG.
[2025-02-04 15:59:11] error: 	zh:controller: Interview failed for '0x00178801012258a1 with error 'Error: Interview failed because can not get node descriptor ('0x00178801012258a1')'
[2025-02-04 15:59:11] error: 	z2m: Failed to interview '0x00178801012258a1', device has not successfully been paired
[2025-02-04 16:00:24] info: 	z2m: Successfully changed options
[2025-02-04 16:00:24] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"tz5ux-1"}'
```

Do we need to make or own profile for this or is there something wrong?

gvlekke, 2025-02-22:
Im a bit stuck to research this, if anyone has any method or some pointers that i can make a environment to test what the problem is that would be nice :)

gvlekke, 2025-02-25:
btw in the documentation it's also saying that if a device is not supported the interview succeeds but it should say in the log that its [not supported](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_1-pairing-the-device-with-zigbee2mqtt)... That doesn't happen in my case. The interview is aborted somehow

Searched a bit and I think this is related 

https://github.com/Koenkk/zigbee2mqtt/issues/19012
https://github.com/Koenkk/zigbee2mqtt/issues/26394
https://github.com/Koenkk/zigbee2mqtt/issues/23180
https://github.com/Koenkk/zigbee2mqtt/issues/25711
https://github.com/Koenkk/zigbee2mqtt/issues/25396
https://github.com/Koenkk/zigbee2mqtt/issues/24864

Habile2019, 2025-02-25:
@gvlekke 

> did you ever succeeded in the past with pairing it to zigbee2mqtt?

This was the first time I tried, after deciding to retire the Hue Bridge. Whilst it did get added and worked for a couple of weeks, it is now offline and won't come back 😢 - it still shows in the map though.

Dazzel, 2025-02-26:
@gvlekke Never got it do work and gave up trying.

gvlekke, 2025-02-26:
Thanks for your response, some topics with the same error also suggested to keep it close to the bridge when pairing. Think that's a bit weird but going to try that anyway. @Habile2019 so your interview was successfully a few weeks ago and then it stopped working all of a sudden... So maybe still an update that made it incompatible...

Habile2019, 2025-02-28:
I turned the Iris back on today, and it appeared back on the network immediately. I'm wondering if it's the way it's routing, but don't know enough to properly troubleshoot. When it came back it was connected directly with the coordinator.

[image]

grvy, 2025-03-02:
I experienced the same problem with my Iris bulb. It worked for a limited time after doing this:

I placed the Iris bulb on top of the USB receiver (Skyconnect in my case), and it was finally discovered. However, when it was farther away from the Skyconnect (one floor down), it would not get discovered.

Then, I moved it back to its original place. It would work for about 30 minutes, but then it would fail when I tried to turn it on or off. I would get these errors:

`2025-03-02 14:19:32z2m: Publish 'get' 'state' to 'Light-Livingroom-Iris' failed: 'Error: ZCL command 0x0017880103329585/11 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '14692'.)'`

When I restarted Z2M, it would start working again for a limited time...

PanDoreS, 2025-03-17:
I've had the same issue.
Re-pairing it to my Hue hub and un-pair it to be discovered by Z2M multiple times did worked (after the 7th or 8th attempt).
However, I face a similar problem to you @grvy, which is apparently known:

- [https://github.com/Koenkk/zigbee2mqtt/issues/24710](https://github.com/Koenkk/zigbee2mqtt/issues/24710)
- [https://github.com/Koenkk/zigbee2mqtt/discussions/20835](https://github.com/Koenkk/zigbee2mqtt/discussions/20835)
- [https://forum.phoscon.de/t/1st-gen-hue-bulbs-with-67-91-1-firmware-not-working/2132](https://forum.phoscon.de/t/1st-gen-hue-bulbs-with-67-91-1-firmware-not-working/2132)

Seems like doing a firmware downgrade may do the trick, but couldn't find how to do it as of today.