Koenkk/zigbee2mqtt #25442: Tuya TS0044 Entities not reported properly after recent update

ThemeIT, 2025-01-03:
### What happened?

After the recent update my Tuya TS0044 switch has no "action" entity anymore, therefore breaking any automation.

In the z2m fronted under the details tab the action is still visible and reported correctly

### What did you expect to happen?

I expected that the Switch still has all 3 entities (link quality, action, battery)

### How to reproduce it (minimal and precise)

Just as you couple the switch it is reproduced. I retried it many times, always only generating the "battery" entity.

(Did i miss something in the breaking changelog?)

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20210708

### Adapter

sonoff zigbee 3.0 usb dongle plus

### Setup

Add-On on HAOS running in Proxmox VM

ThemeIT, 2025-01-03:
Edit:

I oversaw something in the changelog.

Legacy Action entity can be reactivated using
legacy_action_sensor: true

in configuration.yaml

closed 2025-01-03 by ThemeIT

lukasjakobi, 2025-01-03:
I don't think this change is ideal, because legacy installations and blueprint rely on the action sensor. I see that the upgrade guide clearly states this change, but its not quite obvious as proven by you and me overseeing it.

Kibukx, 2025-01-04:
@ThemeIT Is `legacy_action_sensor: true` supposed to go under each device that supports the action feature? I tried this but don't see my action sensor still Spook is complaining. Please provide an example.

lukasjakobi, 2025-01-04:
It's not a per-device setting. You will have to set it in the `homeassistant` node in your `configuration.yaml` like this:

```
homeassistant:
  legacy_action_sensor: true
```

For reference see: https://github.com/Koenkk/zigbee2mqtt/discussions/24198

ThemeIT, 2025-01-04:
> I don't think this change is ideal, because legacy installations and blueprint rely on the action sensor. I see that the upgrade guide clearly states this change, but its not quite obvious as proven by you and me overseeing it.

i think so too. I use a prebuild blueprint for the tuya switch that accepts an entity for the "action" field, so to use the MQTT trigger i would need to rebuild the blueprint.. hmm..