Koenkk/zigbee2mqtt #25443: After upgrade to 2.0.0.1  herdsman wil not start

grimelke001, 2025-01-03:
### What happened?

After upgrade herdsman wil not start, if i rollback to previous version all wil normal

### What did you expect to happen?

Normal start 

### How to reproduce it (minimal and precise)

simpel after update zigbee2mqtt wil not start

### Zigbee2MQTT version

Zigbee2MQTT Huidige versie: 2.0.0-1

### Adapter firmware version

Firmware version:	20240710	v2.6.8.dev25

### Adapter

SLZB-06

### Setup

proxmox x86-64 i5

### Debug log

[00:02:32] INFO: Preparing to start...
[00:02:32] INFO: Socat not enabled
[00:02:32] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-04 00:02:34] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 00:02:34] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 00:02:34] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 00:02:34] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 00:02:34] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 00:02:34] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 00:02:34] error: 	z2m: Exiting...
[2025-01-04 00:02:34] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

wheelyjoe, 2025-01-03:
In the notes it says to add an entry:
`
adapter: zstack # <- it's probably this`

to serial:

This solved that error for me, however I now get 

`[2025-01-03 23:14:25] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-03 23:14:25] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-03 23:14:25] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-03 23:14:25] error: 	z2m: Exiting...
[2025-01-03 23:14:25] error: 	z2m: Error: Error: Not supported, cannot set
/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/.pnpm/winston@3.17.0/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:88:16)
    at Object.onceWrapper (node:events:633:26)
    at SerialPort.emit (node:events:530:35)
    at port.close.then.closing (/app/node_modules/.pnpm/@serialport+stream@12.0.0/node_modules/@serialport/stream/dist/index.js:242:18)`

wheelyjoe, 2025-01-03:
To solve the above - put the right adaptor type in, for me on a sonoff-E - ezsp

EvgveneSV, 2025-01-03:
Similar problem. I updated to HA 2025.1 and next spet to Zigbee2MQTT 2.0.0-1  - and after that addon Zigbee2MQTT can't start. In logs have message:

Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:135
integration: Home Assistant Supervisor (documentation, issues)
First occurred: 1:19:12 AM (1 occurrences)
Last logged: 1:19:12 AM

Failed to to call /addons/45df7312_zigbee2mqtt/stats - Container addon_45df7312_zigbee2mqtt is not running

grimelke001, 2025-01-03:
Youre compleetly right Problem Solved, Thanks a lot Wheelyjoe

GuidoKJ, 2025-01-03:
Hi there!

Im having these problems after adding the adapter change as suggested:

Sonoff P dongle

serial:
  port: /dev/ttyUSB1
  adapter: zstack

Log:
[20:31:30] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[03-01-2025 20:31:31] info: 	z2m: Logging to console, file (filename: log.log)
[03-01-2025 20:31:31] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[03-01-2025 20:31:31] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[03-01-2025 20:31:31] error: 	z2m: Error while starting zigbee-herdsman
[03-01-2025 20:31:31] error: 	z2m: Failed to start zigbee-herdsman
[03-01-2025 20:31:31] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[03-01-2025 20:31:31] error: 	z2m: Exiting...
[03-01-2025 20:31:31] error: 	z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:523:15)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:33)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[03-01-2025 20:31:31] error: 	z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364

It getting late and I have no lights =P

Let me know if there any extra log needed @Koenkk 

Thanks!

EugenKomi, 2025-01-03:
The same problem, i use tcp XZG-01 adapter. 
After update it stop working.

2025-01-04 00:33:42] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

EvgveneSV, 2025-01-03:
Can you explain how to solve? I have USB-Zigbee dongle CC2652P in one of ports in Pi 4B. Here screens from hardware and form Zigbee2MQTT settings. What changes I need ?
[image]
[image]

itjf-de, 2025-01-03:
> In the notes it says to add an entry: ` adapter: zstack # <- it's probably this`
> 
> to serial:

@wheelyjoe thanks for your hint - i hadn't read that anywhere and now everything works again!

but why does zstack have to be specified at all if this is the default and also matches the coordinator?

EvgveneSV, 2025-01-03:
> Can you explain how to solve? I have USB-Zigbee dongle CC2652P in one of ports in Pi 4B. Here screens from hardware and form Zigbee2MQTT settings. What changes I need ? [...]

___
Solved:   add "adapter: zstack"  after "port: /dev/ttyUSB0"

MRobi1, 2025-01-04:
Just adding another data point that adding adapter: zstack fixed it for me as well.

wheelyjoe, 2025-01-04:
For anyone seeing this - it depends on what device you use, I needed to use ezsp for my unflashed sonoff dongle E

Disgruntled-Duck, 2025-01-04:
> Solved: add "adapter: zstack" after "port: /dev/ttyUSB0"

Worked for me as well, thanks!

DJBenson, 2025-01-04:
I'm using an SLZB06m and on upgrade mine wouldn't start (herdsman crashed then would not connect to MQTT and got into a boot loop). I installed and started the Edge build, stopped it and started the production version and it works now so it looks like some migration failed.

jpborjas, 2025-01-04:
For SLZB06, I used adapter: zstack as well, and it works.

Kevinvant, 2025-01-04:
for my Sonoff_Zigbee_3.0 zstack also did the trick

tomlut, 2025-01-04:
Dammit. I misspelled `adapter: zstack ` as `adaptor: zstack` once I fixed that the (king's English) adaptor connected. 

However even though I could see all my devices in the z2m web interface they were all missing in Home Assistant until I restarted Home Assistant.

ZHonry, 2025-01-04:
我也是相同问题，在serial中添加adapter: zstack后恢复正常（xzg-01）
[image]

Tomer-Isr, 2025-01-04:
Hello everyone, what if I had this setting but now I have the same problem?

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  mqtt:
    server: mqtt://core-mosquitto:1883
    base_topic: zigbee2mqt
serial:
  port: tcp://192.168.1.11:6638

What needs to be changed?

Jeppedy, 2025-01-04:
> To solve the above - put the right adaptor type in, for me on a sonoff-E - ezsp

I'm using a UZG-01 network adapter.  What's the right "adapter" entry for that controller?

danysahne96, 2025-01-04:
> > To solve the above - put the right adaptor type in, for me on a sonoff-E - ezsp
> 
> I'm using a UZG-01 network adapter. What's the right "adapter" entry for that controller?

For me zstack ist working in my UZG-01

https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#basic-configuration

Jeppedy, 2025-01-04:
See anything wrong here?
[image: image.png]

danysahne96, 2025-01-04:
> See anything wrong here? 

Not really, i had to remove the baudrate as well. I suggest that you start with the minimal and test If that works for you. Afterwards you can add the needed Parameters as well e.g. LED, etc.

Just put in the serial window:

port: tcp://x.x.x.x:6638
adapter: zstack

ltoolio, 2025-01-04:
> In the notes it says to add an entry: ` adapter: zstack # <- it's probably this`
>
> to serial:
>
> This solved that error for me, however I now get
>
> `[2025-01-03 23:14:25] error: z2m: Error while [...]

Same here on my Sonoff_Zigbee_3.0 with adapter: zstack.

As my FIL used to say - RTFM.

Jeppedy, 2025-01-04:
> writeAfterEnd ***@***.***/node_modules/readable-stream/lib/_stream_writable.js:264:12)
> at DerivedLogger.log ***@***.***/node_modules/winston/lib/winston/logger.js:231:12)

Just put that in the serial window?  Not under a "serial:" section header?
Take a second look please at my screenshot.  Is it supposed to have the
"serial:" heading before the other rows?

GarryG, 2025-01-04:
Thanks for the tips, ezsp worked for my Sonoff stick ...

tobjer, 2025-01-04:
> For anyone seeing this - it depends on what device you use, I needed to use ezsp for my unflashed sonoff dongle E

Same for me, adapter: ezsp worked! Thank you for this! <3

I kept getting the "z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)" after setting it to adapter: zstack. 

My device identifies as: 
QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2

Luke001IT, 2025-01-04:
> Solved: add "adapter: zstack" after "port: /dev/ttyUSB0"

Worked for me!
Thansk a lot

RyeGuy-IN, 2025-01-04:
SLZB-06, same error.  In the addon my configuration looked like shown below.  This is due to me using the config.yaml file for settings.  I had to add the adapter line to my config.yaml file using studio code server.  

**Addon Config Below:**
data_path: /config/z2m_house
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8481,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial: {} 

**Configuration.yaml Partial Code below**
serial:
  adapter: zstack
  port: tcp://10.100.100.23:6638
  baudrate: 115200

kzajac83, 2025-01-04:
FYI, have ZigStar with PoE on ESP32 Olimex and adding "adapter: zstack" resolve my problem with below error after migration to v2.0

```
[2025-01-04 23:17:34] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-04 23:17:34] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-04 23:17:34] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-04 23:17:34] error: 	z2m: Exiting...
[2025-01-04 23:17:34] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
```

ExCuDo-HA, 2025-01-04:
Hello, I also updated to version 2.0.0-1.

Zigbee2MQTT does not start. I followed your advice and restarted several times, but it still doesn't work.

See below configuration and logs.

Laptop with proxmox and HA, ZBT-1 Nabu Casa USB Zigbee Dongle:

[image]

/homeassistant/zigbee2mqtt/configuration.yaml

[image]
[image]

But it’s still not working. Can anyone help me?

digitui, 2025-01-05:
adapter: zstack does not work for zzh adapters

ExCuDo-HA, 2025-01-05:
I try also
adapter: ember 
but don't work (see logs below)

[09:13:06] INFO: Preparing to start...
[09:13:06] INFO: Socat not enabled
[09:13:06] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 09:13:08] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 09:13:08] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 09:13:08] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 09:13:08] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-05 09:13:08] info: 	zh:zstack:znp: Serialport opened
[2025-01-05 09:13:08] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-05 09:13:10] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-05 09:13:28] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-05 09:13:28] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-05 09:13:28] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-05 09:13:28] error: 	z2m: Exiting...
[2025-01-05 09:13:28] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

fc1608, 2025-01-05:
I'm using SonOff dongle E, and tried ember, and zstack, but always an error ! How can I do? 
Logs (1st with ember, 2nd with ztack) : 

[10:03:30] INFO: Preparing to start...
[10:03:30] INFO: Socat not enabled
[10:03:30] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 10:03:31] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 10:03:31] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 10:03:31] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 10:03:31] info: 	zh:ember: Using default stack config.
[2025-01-05 10:03:31] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-05 10:03:31] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-05 10:03:31] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-05 10:03:31] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-01-05 10:03:31] info: 	zh:ember:uart:ash: Serial port opened
[2025-01-05 10:03:31] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-05 10:03:33] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-01-05 10:03:33] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-01-05 10:03:33] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-01-05 10:03:33] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-05 10:03:33] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-05 10:03:33] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-05 10:03:33] error: 	z2m: Exiting...
[2025-01-05 10:03:33] error: 	z2m: Error: Adapter EZSP protocol version (8) is not supported by Host [13-16].
    at EmberAdapter.emberVersion (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1403:19)
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:674:9)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

-----
[10:05:42] INFO: Preparing to start...
[10:05:42] INFO: Socat not enabled
[10:05:42] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 10:05:44] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 10:05:44] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 10:05:44] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 10:05:44] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231218213358-if00","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-05 10:05:44] info: 	zh:zstack:znp: Serialport opened
[2025-01-05 10:05:44] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-05 10:05:45] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-05 10:05:45] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-05 10:05:45] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-05 10:05:45] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-05 10:05:45] error: 	z2m: Exiting...
[2025-01-05 10:05:45] error: 	z2m: Error: Error: Not supported, cannot set
/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/.pnpm/winston@3.17.0/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:88:16)
    at Object.onceWrapper (node:events:633:26)
    at SerialPort.emit (node:events:530:35)
    at port.close.then.closing (/app/node_modules/.pnpm/@serialport+stream@12.0.0/node_modules/@serialport/stream/dist/index.js:242:18)

Erikwout, 2025-01-05:
Fot the TubeSZ
adapter: zstack
fixed the problem also

REHolt, 2025-01-05:
I'm having the same problem. 

I have a Home Assistant Blue, with a Skyconnect Dongle configured for Zigbee. Prior to the update it was working great. Now Zigbee2MQTT will not start. I tried adding the recommended lines:

port: /dev/ttyUSB0
adapter: zstack

but I get the following error:
Starting Zigbee2MQTT without watchdog.
[2025-01-05 13:53:41] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 13:53:41] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 13:53:41] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 13:53:42] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-05 13:53:42] info: 	zh:zstack:znp: Serialport opened
[2025-01-05 13:53:42] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-05 13:53:43] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-05 13:54:02] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-05 13:54:02] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-05 13:54:02] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-05 13:54:02] error: 	z2m: Exiting...
[2025-01-05 13:54:02] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

Anybody have an idea of how I can solve this?

REHolt, 2025-01-05:
OK, I managed to get it working by changing the adapter to:

adapter: ezsp

I do get a warning though telling me ezsp has been deprecated and I should use ember instead, however; when i try that it doesn't work.

also, I now have a problem that the WEBUI won't start, either from the side panel or from within the Zigbee2MQTT addon.

Erikwout, 2025-01-05:
Hi REHolt
I think you might want to look at https://github.com/Koenkk/zigbee2mqtt/discussions/21462

REHolt, 2025-01-05:
Thanks Erikwout, I appreciate the heads up. This led me down another rabbit trail and another  headache. At one point I thought I would have been better off to just quite while I was ahead. In the end though I go my Skyconnect updated to the latest software and got Zigbee2MQTT working again, even if it did take 2 hours :)

closed 2025-01-06 by grimelke001

c4sti3l, 2025-01-09:
I have the SLZB-06P7 Coordinator, only adding 

`adapter: zstack` 

to serial was not enough, i had to add also the baudrate:

`baudrate: 115200`

after adding both values to serial, my Zigbee2MQTT starts again.

digitui, 2025-01-09:
> adapter: zstack does not work for zzh adapters

Working since upgrade to 2.0.0-2

alexstrzelecki, 2025-01-10:
@fc1608 You need to upgrade your adapter such that it works with Ember. You might need to flash the adapter firmware first. I had to do that to get v2.0 to work. [guide](https://github.com/Koenkk/zigbee2mqtt/discussions/21462#)

narayanvs, 2025-01-11:
Hi, I managed to start the container, i had also add the device in the container with the same name as in the config file

[image]

at least the container starts now, however the WebUI is not working, i will debug and see if some additional settings need to be added

djurny, 2025-01-11:
Hi @narayanvs,
Even though you say it worked perfectly fine before, it is still good to confirm some things.
How are you trying to connect to your web UI? Using the unraid docker containers interface or ..? 
How about the port for the unraid docker webUI might be connecting to :80 instead of :8080 (z2m's default frontend port)?
Grt,

narayanvs, 2025-01-13:
sorry for not responding earlier as i was busy, i think the issue is that unraid container has 9442 as webui port and i haven't overridden that default port in config file, i will try that tomorrow and see if it helps and report back.

chukaonline, 2025-02-01:
I have CC2652P adapter and on Zigbee2MQTT 2.0.0-2. after adding below it worked for one day and error resumed. not sure what is wrong now.

`port: /dev/ttyUSB0
adapter: zstack`

Log
`[18:43:00] INFO: Preparing to start...
[18:43:00] INFO: Socat not enabled
[18:43:00] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-01 18:43:05] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-01 18:43:05] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-02-01 18:43:05] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-02-01 18:43:05] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-01 18:43:05] info: 	zh:zstack:znp: Serialport opened
[2025-02-01 18:43:05] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-01 18:43:06] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-02-01 18:43:25] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-01 18:43:25] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-01 18:43:25] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-01 18:43:25] error: 	z2m: Exiting...
[2025-02-01 18:43:25] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[18:44:37] INFO: Preparing to start...
[18:44:37] INFO: Socat not enabled
[18:44:38] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-01 18:44:42] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-01 18:44:42] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-02-01 18:44:42] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-02-01 18:44:42] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-01 18:44:42] info: 	zh:zstack:znp: Serialport opened
[2025-02-01 18:44:43] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-01 18:44:44] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-02-01 18:45:02] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-01 18:45:02] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-01 18:45:02] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-01 18:45:02] error: 	z2m: Exiting...
[2025-02-01 18:45:02] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)`

chukaonline, 2025-02-01:
I restored Zigbee2MQTT version 1.41.0 but still getting the error. Any help is appreciated.

[19:03:57] INFO: Preparing to start...
[19:03:57] INFO: Socat not enabled
[19:03:58] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-02-01 19:04:03] info: 	z2m: Logging to console, file (filename: log.log)
[2025-02-01 19:04:03] info: 	z2m: Starting Zigbee2MQTT version 1.41.0 (commit #unknown)
[2025-02-01 19:04:03] info: 	z2m: Starting zigbee-herdsman (2.1.7)
[2025-02-01 19:04:04] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-01 19:04:04] info: 	zh:zstack:znp: Serialport opened
[2025-02-01 19:04:04] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-01 19:04:05] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-02-01 19:04:24] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-01 19:04:24] error: 	z2m: Failed to start zigbee
[2025-02-01 19:04:24] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2025-02-01 19:04:24] error: 	z2m: Exiting...
[2025-02-01 19:04:24] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:161:27)
    at start (/app/index.js:154:5)

manoloxxl, 2025-02-04:
> I have CC2652P adapter and on Zigbee2MQTT 2.0.0-2. after adding below it worked for one day and error resumed. not sure what is wrong now.
> 
> port: /dev/ttyUSB0 adapter: zstack
> ....

Excact same behaviour here. But GUI and system is working after a while. Don't know why.

djurny, 2025-02-04:
Hi @manoloxxl @chukaonline,
Did you try to power cycle the dongle? (Unplug/plug.)
Also try to share some debug logging so there is more to look at.
Thx,
Grt,

dothebart, 2025-02-13:
for some reason re/plugging my TI/zstack stick was the cure for me.

chukaonline, 2025-04-07:
@djurny and all, I got mine working after i switched back to the Sonoff Zigbee dongle P. i had changed coordinator just after the upgrade and got the error. Now back to the coordinator to Sonoff and no issues so far.

Jucy74, 2025-04-24:
Hello everyone,

I am also struggling to find a solution, even if I tried my best to follow your previous advices...
Can you have a look at my logs/config and let me know if you find any solution please ?

_______________________________________________________________________________________

[22:36:29] INFO: Preparing to start...
[22:36:29] INFO: Socat not enabled
[22:36:30] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-04-24 22:36:34] info: 	z2m: Logging to console, file (filename: log.log)
[2025-04-24 22:36:35] info: 	z2m: Starting Zigbee2MQTT version 2.2.1 (commit #unknown)
[2025-04-24 22:36:35] info: 	z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-24 22:36:35] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-04-24 22:36:35] info: 	zh:zstack:znp: Serialport opened
[2025-04-24 22:36:36] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-04-24 22:36:37] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-04-24 22:36:55] error: 	z2m: Error while starting zigbee-herdsman
[2025-04-24 22:36:55] error: 	z2m: Failed to start zigbee-herdsman
[2025-04-24 22:36:55] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-04-24 22:36:55] error: 	z2m: Exiting...
[2025-04-24 22:36:56] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:104:13)
    at start (/app/index.js:149:5)

___________________________________________________

[image]

Dongle : Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2

mschwadorf, 2025-05-03:
Hello,

after Update to version 2.3.0.1 z2m doesn´t start.
Ubdate happened automatically.

[07:46:07] INFO: Preparing to start...
[07:46:07] INFO: Socat not enabled
[07:46:08] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-05-03 07:46:21] error: 	z2m: Error while starting zigbee-herdsman
[2025-05-03 07:46:21] error: 	z2m: Failed to start zigbee-herdsman
[2025-05-03 07:46:21] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-05-03 07:46:21] error: 	z2m: Exiting...
[2025-05-03 07:46:21] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:666:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1548:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:70:27)
    at Controller.start (/app/lib/controller.ts:104:13)
    at start (/app/index.js:149:5)

Restored Backup doesn´t help.
HA always recognizes the new Update an makes the update mandatory.

Help needed

JDMunnerz, 2025-05-12:
I had exactly this with back up to previous version still leaving me not working.

Multiple software restarts, changing adapter settings etc did not work.

What worked for me was a physical power cycle of the RPI5 that HA is running on. All worked on the latest versions once it came back on.