Koenkk/zigbee2mqtt #25444: Failing to bind more than one cluster of a hue bulb (TABLE_FULL error)

joncage, 2025-01-04:
### What happened?

I've been trying to add a bulb (https://www.zigbee2mqtt.io/devices/8718696695203.html) to my mainly hue Zigbee mesh (tried resetting it and re-adding it many times now trying to resolve this) but each time I try and bind it I get errors when attempting to bind it's clusters to the coordinator.

Here's what I'm trying:

[image]

The result:

[image]

```
info 2025-01-03 23:44:29z2m: Successfully bound cluster 'genBasic' from 'Lounge Main Light 1' to 'Coordinator'
error 2025-01-03 23:44:29z2m: Failed to bind cluster 'genOta' from 'Lounge Main Light 1' to 'Coordinator' (Error: Bind 0x0017880104a8c884/11 genOta from '0x00124b002e20cb36/1' failed (Status 'TABLE_FULL'))
info 2025-01-03 23:44:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genBasic"],"failed":["genOta"],"from":"Lounge Main Light 1","from_endpoint":"11","to":"Coordinator","to_endpoint":null},"status":"ok","transaction":"cptuu-9"}'
```

```
info 2025-01-03 23:44:43z2m: Successfully bound cluster 'genBasic' from 'Lounge Main Light 1' to 'Coordinator'
error 2025-01-03 23:44:43z2m: Failed to bind cluster 'genOnOff' from 'Lounge Main Light 1' to 'Coordinator' (Error: Bind 0x0017880104a8c884/11 genOnOff from '0x00124b002e20cb36/1' failed (Status 'TABLE_FULL'))
info 2025-01-03 23:44:43z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genBasic"],"failed":["genOnOff"],"from":"Lounge Main Light 1","from_endpoint":"11","to":"Coordinator","to_endpoint":null},"status":"ok","transaction":"cptuu-10"}'
```

If I unbind then rebind just `genOnOff` for example it works:

```
info 2025-01-03 23:48:10z2m: Successfully bound cluster 'genOnOff' from 'Lounge Main Light 1' to 'Coordinator'
info 2025-01-03 23:48:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genOnOff"],"failed":[],"from":"Lounge Main Light 1","from_endpoint":"11","to":"Coordinator","to_endpoint":null},"status":"ok","transaction":"cptuu-12"}'
```

Looking at the attached debug logs we can see some examples of failures in a bit more detail stuch as:

```
[2025-01-03 23:00:38] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/bind' with data '{"clusters":["lightingColorCtrl","manuSpecificPhilips2","genOnOff","genBasic","genOta","genGroups","genScenes","genLevelCtrl"],"from":"Lounge Main Light 1/11","to":"Coordinator","transaction":"nktyp-16"}'
[2025-01-03 23:00:38] debug: 	z2m: binding cluster 'lightingColorCtrl' from 'Lounge Main Light 1' to 'Coordinator'
[2025-01-03 23:00:38] debug: 	zh:controller:endpoint: Bind 0x0017880104a8c884/11 lightingColorCtrl from '0x00124b002e20cb36/1'
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:writer: --> frame [254,23,37,33,64,191,132,200,168,4,1,136,23,0,11,0,3,3,54,203,32,46,0,75,18,0,1,50]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,33,0,69]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,33,0,69]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 33 - [0] - 69
[2025-01-03 23:00:38] debug: 	zh:zstack:znp: <-- SRSP: ZDO - bindReq
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,161,64,191,0,24]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,161,64,191,0,24]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 161 - [64,191,0] - 24
[2025-01-03 23:00:38] debug: 	zh:zstack:znp: <-- AREQ: ZDO - bindRsp
[2025-01-03 23:00:38] debug: 	zh:controller: Received ZDO response: clusterId=BIND_RESPONSE, status=SUCCESS, payload=undefined
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-03 23:00:38] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-01-03 23:00:38] info: 	z2m: Successfully bound cluster 'lightingColorCtrl' from 'Lounge Main Light 1' to 'Coordinator'
[2025-01-03 23:00:38] debug: 	z2m: binding cluster 'manuSpecificPhilips2' from 'Lounge Main Light 1' to 'Coordinator'
[2025-01-03 23:00:38] debug: 	zh:controller:endpoint: Bind 0x0017880104a8c884/11 manuSpecificPhilips2 from '0x00124b002e20cb36/1'
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:writer: --> frame [254,23,37,33,64,191,132,200,168,4,1,136,23,0,11,3,252,3,54,203,32,46,0,75,18,0,1,206]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,33,0,69]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,33,0,69]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 33 - [0] - 69
[2025-01-03 23:00:38] debug: 	zh:zstack:znp: <-- SRSP: ZDO - bindReq
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,161,64,191,140,148]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,161,64,191,140,148]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 161 - [64,191,140] - 148
[2025-01-03 23:00:38] debug: 	zh:zstack:znp: <-- AREQ: ZDO - bindRsp
[2025-01-03 23:00:38] debug: 	zh:controller: Received ZDO response: clusterId=BIND_RESPONSE, status=TABLE_FULL, payload=undefined
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-03 23:00:38] debug: 	zh:controller:endpoint: Error: Bind 0x0017880104a8c884/11 manuSpecificPhilips2 from '0x00124b002e20cb36/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Bind.onMQTTMessage (/app/lib/extension/bind.ts:306:29)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17)
[2025-01-03 23:00:38] error: 	z2m: Failed to bind cluster 'manuSpecificPhilips2' from 'Lounge Main Light 1' to 'Coordinator' (Error: Bind 0x0017880104a8c884/11 manuSpecificPhilips2 from '0x00124b002e20cb36/1' failed (Status 'TABLE_FULL'))
[2025-01-03 23:00:38] debug: 	z2m: binding cluster 'genOnOff' from 'Lounge Main Light 1' to 'Coordinator'
[2025-01-03 23:00:38] debug: 	zh:controller:endpoint: Bind 0x0017880104a8c884/11 genOnOff from '0x00124b002e20cb36/1'
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:writer: --> frame [254,23,37,33,64,191,132,200,168,4,1,136,23,0,11,6,0,3,54,203,32,46,0,75,18,0,1,55]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,33,0,69]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,33,0,69]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 33 - [0] - 69
[2025-01-03 23:00:38] debug: 	zh:zstack:znp: <-- SRSP: ZDO - bindReq
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,161,64,191,140,148]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,161,64,191,140,148]
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 161 - [64,191,140] - 148
[2025-01-03 23:00:38] debug: 	zh:zstack:znp: <-- AREQ: ZDO - bindRsp
[2025-01-03 23:00:38] debug: 	zh:controller: Received ZDO response: clusterId=BIND_RESPONSE, status=TABLE_FULL, payload=undefined
[2025-01-03 23:00:38] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-03 23:00:38] debug: 	zh:controller:endpoint: Error: Bind 0x0017880104a8c884/11 genOnOff from '0x00124b002e20cb36/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Bind.onMQTTMessage (/app/lib/extension/bind.ts:306:29)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17)
```

Other bulbs seem to work as expected but this one seems to be problematic.

### What did you expect to happen?

I expected that I could bind more than one cluster to the coordinator.

### How to reproduce it (minimal and precise)

Try and add multiple clusters from this Hue white ambience E14 bulb. Serial number 2FF400 8L9

### Zigbee2MQTT version

 2.0.0 commit: unknown

### Adapter firmware version

 20210708

### Adapter

SONOFF Universal Zigbee 3.0 USB Dongle Plus

### Setup

Add-on on home assustant on a RPI 5

### Debug log

[Logs.zip](https://github.com/user-attachments/files/18304717/Logs.zip)

These are all the logs from a lot of messing around today trying to resolve this for this and another bulb. I've tried updating to the 2.0.0 release today and that doesn't seem to have improved matters (same behaviour from both).

groenmarsmannetje, 2025-01-05:
I am experiencing something similar while binding a new IKEA remote with 3 new IKEA bulbs. I am also on latest Z2M version. I was able to workaround it by removing some older/unused devices and bindings.

joncage, 2025-01-08:
Digging into the stack trace / saource for another failure from another attempt today:

```
[2025-01-08 23:31:29] debug: 	z2m: binding cluster 'genOta' from 'Lounge Main Light 1' to 'Coordinator'
[2025-01-08 23:31:29] debug: 	zh:controller:endpoint: Bind 0x0017880104a8c884/11 genOta from '0x00124b002e20cb36/1'
[2025-01-08 23:31:29] debug: 	zh:zstack:unpi:writer: --> frame [254,23,37,33,64,191,132,200,168,4,1,136,23,0,11,25,0,3,54,203,32,46,0,75,18,0,1,40]
[2025-01-08 23:31:29] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,33,0,69]
[2025-01-08 23:31:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,33,0,69]
[2025-01-08 23:31:29] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 33 - [0] - 69
[2025-01-08 23:31:29] debug: 	zh:zstack:znp: <-- SRSP: ZDO - bindReq
[2025-01-08 23:31:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-08 23:31:29] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,161,64,191,140,148]
[2025-01-08 23:31:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,161,64,191,140,148]
[2025-01-08 23:31:29] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 161 - [64,191,140] - 148
[2025-01-08 23:31:29] debug: 	zh:zstack:znp: <-- AREQ: ZDO - bindRsp
[2025-01-08 23:31:29] debug: 	zh:controller: Received ZDO response: clusterId=BIND_RESPONSE, status=TABLE_FULL, payload=undefined
[2025-01-08 23:31:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-08 23:31:29] debug: 	zh:controller:endpoint: Error: Bind 0x0017880104a8c884/11 genOta from '0x00124b002e20cb36/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:565:23)
    at Bind.onMQTTMessage (/app/lib/extension/bind.ts:357:25)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17)
[2025-01-08 23:31:29] error: 	z2m: Failed to bind cluster 'genOta' from 'Lounge Main Light 1' to 'Coordinator' (Error: Bind 0x0017880104a8c884/11 genOta from '0x00124b002e20cb36/1' failed (Status 'TABLE_FULL'))
[2025-01-08 23:31:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genOnOff"],"failed":["genOta"],"from":"Lounge Main Light 1","from_endpoint":"11","to":"Coordinator","to_endpoint":null},"status":"ok","transaction":"z42c0-1"}'
```

This is in endpoint.ts: https://github.com/Koenkk/zigbee-herdsman/blob/eb5b89cc3a657f20fd0405029bfb928b33a57c5b/src/controller/model/endpoint.ts#L565

..which doesn't look all the helpful. Searching that repo for TABLE_FULL does yield a bit more background: https://github.com/search?q=repo%3AKoenkk%2Fzigbee-herdsman%20TABLE_FULL&type=code

..and this seems to suggest it's a problem with the device giving the ZDO response i.e the light bulb in my case:

```
/**
 * ZDO response status.
 *
 * Most responses to ZDO commands contain a status byte.
 * The meaning of this byte is defined by the ZigBee Device Profile.
 *
 * Zigbee Document – 05-3474-23 - Table 2-129. ZDP Enumerations Description
 *
 * uint8_t
 */
export enum Status {
    /** The device does not have table space to support the operation. */
    TABLE_FULL = 0x8c,
```

joncage, 2025-01-08:
Page 98 of the spec talks about binding tables within the device: https://csa-iot.org/wp-content/uploads/2023/04/05-3474-23-csg-zigbee-specification-compressed.pdf

..now in my case, the bulb was an old one. So maybe I just need a means to reset the binding table in this bulb.

I wonder if there's a way to reset / clear this table? Seems like resetting the bulb isn't doing this (assuming it's the bulb and not the controller that's the problem).

@Koenkk is there a way to get more feedback on which device's table is full and how to clear it out of any unused devices?

Koenkk, 2025-01-09:
> is there a way to get more feedback on which device's table is full and how to clear it out of any unused devices?

You can see the existing bindings in the z2m frontend -> device -> bindings tab. Unbinding should remove entries from the table.

joncage, 2025-01-09:
Have I got the wrong end of the stick with this altogether? I assumed I needed to bind the lights to the coordinator to get feedback on their current state(?)

Koenkk, 2025-01-10:
>  I assumed I needed to bind the lights to the coordinator to get feedback on their current state(?)

True but Hue bulbs don't support this AFAIK.

closed 2025-03-19 by github-actions[bot]