Koenkk/zigbee2mqtt #25446: Deprecated flag `color_mode` used in MQTT JSON light config (Will break in HA soon)

SHxKM, 2025-01-04:
### What happened?

When HA starts up, it complains:

> Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/kitchen_lights/set', 'device': {'identifiers': ['zigbee2mqtt_[REDACTED]_4'], 'manufacturer': 'Zigbee2MQTT', 'model': 'Group', 'sw_version': 'Zigbee2MQTT 1.42.0', 'via_device': 'zigbee2mqtt_bridge_[REDACTED], 'name': 'kitchen_lights'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'candle', 'fireplace', 'colorloop', 'finish_effect', 'stop_effect', 'stop_hue_effect'], 'json_attributes_topic': 'zigbee2mqtt/kitchen_lights', 'max_mireds': 454, 'min_mireds': 153, 'object_id': 'kitchen_lights', 'origin': {'support_url': 'https://www.zigbee2mqtt.io', 'name': 'Zigbee2MQTT', 'sw_version': '1.42.0'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/kitchen_lights', 'supported_color_modes': ['color_temp'], 'unique_id': '4_light_zigbee2mqtt', 'name': None}. **This will stop working in Home Assistant Core 2025.3**

From previous threads, I see this should have been fixed quite a long while ago. The approaching breakage in `2025.3` has me a bit worried as I'm not sure what is causing these warnings.

### What did you expect to happen?

No warnings are emitted.

### How to reproduce it (minimal and precise)

Unusure

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

0x26580700

### Adapter

Conbee II

### Setup

Add-on HassOS Intel Nuc

SHxKM, 2025-01-04:
Ah, it might be due to this in my `groups.yaml`:

```
homeassistant:
  light:
    color_mode: true
    supported_color_modes:
      - color_temp
```

But how do I tell Z2M then that these lights support color temperature and not color?

SHxKM, 2025-01-04:
I removed `color_mode: true` from all config entries. Everything is still working fine but the warning is still being emitted.

SHxKM, 2025-01-04:
In my `state.json`, I see many `"color_mode": "color_temp"` and `"color_mode": "xy"` - is this is what's causing the issue? should I stop z2m, remove those props, and restart z2m and HA?

SHxKM, 2025-01-04:
OK, found something interesting. 

This warning is only emitted for two Z2M groups that are comprised of Philips Hue lights with _only_ color temperature capabilities. These were the subject of [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/10120) that I opened, after which @Koenkk added support for overriding a **group's discovery payload** (see [this](https://github.com/Koenkk/zigbee2mqtt/issues/10120#issuecomment-997017838)). This change essentially allows us to group 2 lights: 1 with color capabilities and one without, have the group entity itself be exposed as an entity _without_ color capabilities, but still keep the light with color capabilities exposed as is.

@Koenkk - could it be that when the deprecation warnings were handled for individual lights, groups were missed?

SHxKM, 2025-01-04:
I think I found it: [lib/extension/groups.ts](https://github.com/Koenkk/zigbee2mqtt/blob/060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a/lib/extension/groups.ts#L25) and [test/extensions/groups.test.ts](https://github.com/Koenkk/zigbee2mqtt/blob/060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a/test/extensions/groups.test.ts#L406) still reference `color_mode` for groups.

Koenkk, 2025-01-05:
@SHxKM I don't see any mention of `color_mode` in `homeassistant.ts`, `color_mode` for HA has already been removed a long time ago (https://github.com/Koenkk/zigbee2mqtt/pull/22142). Do you have any manually discovered groups?

SHxKM, 2025-01-05:
> @SHxKM I don't see any mention of `color_mode` in `homeassistant.ts`, `color_mode` for HA has already been removed a long time ago (#22142). Do you have any manually discovered groups?

Hi. Thanks for taking a look. I'm not sure what you mean by "manually discovered groups". Also, I upgraded to `2.0.0` today and the warning is still emitted.

Let me share what I think is the relevant config from `groups.yaml`:

```yml
homeassistant:
  light:
    color_mode: true
    supported_color_modes:
      - color_temp
```

But even if I remove `color_mode: true`, the error is still emitted:

> Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/kitchen_lights/set', 'device': {'identifiers': ['zigbee2mqtt_1221051039810110150109113116116_4'], 'manufacturer': 'Zigbee2MQTT', 'model': 'Group', 'sw_version': 'Zigbee2MQTT 2.0.0', 'via_device': 'zigbee2mqtt_bridge_0x00212effff0616cb', 'name': 'kitchen_lights'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'candle', 'fireplace', 'colorloop', 'finish_effect', 'stop_effect', 'stop_hue_effect'], 'max_mireds': 454, 'min_mireds': 153, 'object_id': 'kitchen_lights', 'origin': {'support_url': 'https://www.zigbee2mqtt.io', 'sw_version': '2.0.0', 'name': 'Zigbee2MQTT'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/kitchen_lights', 'supported_color_modes': ['color_temp'], 'unique_id': '4_light_zigbee2mqtt', 'name': None}. This will stop working in Home Assistant Core 2025.3

If you remember, this basically allows me to say this Z2M group only supports color temperatures, while keeping (at least) one of its lights with its original color capabilities.

SHxKM, 2025-01-05:
@Koenkk I'm trying to debug this through looking at source code. Are the payloads defined in `homeassistant.ts` merged with any other values, perhaps from `states.json`, before being sent as the discovery payload to HA? because `color_mode` is being included:

[image]

SHxKM, 2025-01-05:
There's gotta be some kind of post-processing or caching of `discoveryEntries` here:

https://github.com/Koenkk/zigbee2mqtt/blob/33cc16c3abf226b3668451d36ed9c0fd4b15f455/lib/extension/homeassistant.ts#L191-L204

Otherwise `color_mode: true` really has nowhere to come from. (I already removed it from `groups.yaml`)

SHxKM, 2025-01-05:
This happens later and is now my prime suspect, but I don't know where from "user settings" it's getting `color_mode`:

https://github.com/Koenkk/zigbee2mqtt/blob/03ba647dc6b5f299f8f3ab441712999fcb3a253e/lib/extension/homeassistant.ts#L1269-L1290

SHxKM, 2025-01-05:
OK, you'll call me crazy but I just restarted HA and the warning just didn't appear. I wonder if some kind of cache/MQTT message retaining was at play here.

closed 2025-01-05 by SHxKM