Koenkk/zigbee2mqtt #25447: 2.0 cant find external converters

my5123, 2025-01-04:
### What happened?

devices with external converters at 1.4 lodt

### Zigbee2MQTT version

2.0

### Adapter firmware version

20230507

### Adapter

cc2652

### Setup

x86

rp2019433, 2025-01-04:
@my5123  Just to confirm: After upgrading to Z2M 2.0, you have moved your external converters to data/external_converters [(New Procedure)](https://www.zigbee2mqtt.io/advanced/more/external_converters.html) and Z2M is not able to find the converters?

If this is the case, I'm experiencing the same problem. Z2M does not load converters on folder data/external_converters.

Mrsash2020, 2025-01-04:
The instructions are not really clear for me. Can anyone clarify why it was removed from the GUI?

scaytrase, 2025-01-04:
Check permissions and check logs. For me I had 

```
[2025-01-04 06:15:06] error: 	z2m: Failed to call 'ExternalConverters' 'start' (/app/data/extension/zFlora_ProMax_R3.js:131
        exposes: [e.soil_moisture(), e.battery(), e.battery_low(), e.battery_voltage(), e.temperature(), e.humidity(), e.illuminance(), e.illuminance_lux(),
                                                                                                                                          ^

TypeError: e.illuminance_lux is not a function
    at /app/data/extension/zFlora_ProMax_R3.js:131:139
    at Script.runInContext (node:vm:149:12)
    at Script.runInNewContext (node:vm:154:17)
    at runInNewContext (node:vm:310:38)
    at ExternalConverters.loadModuleFromText (/app/lib/extension/externalJS.ts:187:24)
    at ExternalConverters.loadFiles (/app/lib/extension/externalJS.ts:156:52)
    at ExternalConverters.start (/app/lib/extension/externalJS.ts:45:20)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9))
```

in logs, fix was elemantary as this was noted in BC changes

https://github.com/Koenkk/zigbee2mqtt/discussions/24198

Mrsash2020, 2025-01-04:
The instructions say external converters are loaded from data/external_converters. I tried creating a folder in zigbee2mqtt folder along the lines of zigbee2mqtt\data\external_converters but this did not work on restarting zigbee2mqtt.

Lotambena, 2025-01-04:
> The instructions say external converters are loaded from data/external_converters.
> 
> I tried creating a folder in zigbee2mqtt folder along the lines of zigbee2mqtt\data\external_converters and moved external converters to it, but did not load the external converters on restarting zigbee2mqtt.

i am keeping track on the solution, i am in similar situation

my5123, 2025-01-04:
> @my5123  Just to confirm: After upgrading to Z2M 2.0, you have moved your external converters to data/external_converters [(New [...]

yes, i have moved all the js file to new folder. but it is not functional after rebooting.

rp2019433, 2025-01-04:
> The instructions say external converters are loaded from data/external_converters.
> 
> I tried creating a folder in zigbee2mqtt folder along the lines of zigbee2mqtt\data\external_converters and moved external converters to it, but did not load the external converters on restarting zigbee2mqtt.

Same here. Devices are not working properly since then.

magister86, 2025-01-04:
I'm in the same situation from the last update. I've created the folder manually, inside zigbee2mqtt folder (and outside just in case) and it doesn't work. 
I also tried to use the API method to save the coverter code (zigbee2mqtt/bridge/request/converter/save) following the new documentation, and it gets all the time an error (Payload invalid). Even I publish all the js code to the topic in one line and I get a different error.
So right now, my thermostat is not supported.

m0wlheld, 2025-01-04:
> ### External converters and extensions
> *  The external_converters setting is no longer used. Instead all external converters inside data/external_converters directory are now automatically loaded. Make sure to move all your external converters to this directory.

Open a terminal, navigate to the location of Zigbee2MQTT's configuration.yaml and create a folder named "external_converters" there. Move all you external converts to this directory. Remove the external_converters statement from configuration.yaml. Restart Zigbee2MQTT afterwards.

Pretty straight forward.

magister86, 2025-01-04:
> Open a terminal, navigate to the location of Zigbee2MQTT's configuration.yaml and create a folder named "external_converters" there. Move all you external converts to this directory. Remove the [...]

You nailed it!! Now it works. The problem is the path of the documentation, it's not really clear. As you say is:  **zigbee2mqtt\external_converters** not zigbee2mqtt\data\external_converters, and it's very important to delete the line of the external converter file in the configuration.yaml (already did after the update automatically).

Thanks a lot!

m0wlheld, 2025-01-04:
Thanks. Can you please close this issue?

rp2019433, 2025-01-04:
> Open a terminal, navigate to the location of Zigbee2MQTT's configuration.yaml and create a folder named "external_converters" there. Move all you external converts to this directory. Remove the [...]

Perfect. Now it works. Thank you!

How can we update the docs?

magister86, 2025-01-04:
> Perfect. Now it works. Thank you!
>
> How can we update the docs?

If I'm not wrong, in the following link: https://github.com/Koenkk/zigbee2mqtt.io/edit/master/docs/advanced/more/external_converters.md

magister86, 2025-01-04:
> Thanks. Can you please close this issue?

Sorry for my ignorance, but I'm not able to see a "Close" button".

m0wlheld, 2025-01-04:
> Ok so did not work for me. How do I confirm it worked if the unsupported devices are still unsupported?
> 
> ![image](https://private-user-images.githubusercontent.com/62192773/400138711-27139c85-c254-431c-8b2c-db0ad0a99def.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU5OTY3MzUsIm5iZiI6MTczNTk5NjQzNSwicGF0aCI6Ii82MjE5Mjc3My80MDAxMzg3MTEtMjcxMzljODUtYzI1NC00MzFjLThiMmMtZGIwYWQwYTk5ZGVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA0VDEzMTM1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI5MGE4OTE4Yzc2ZTM3OGE5ZTI1ZjMyYzg0M2ZiMTExNDVmZjJhMDk4OTU3OTJhMzVkYzY1Yjk2ZWM1MTJiZTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.5pyVhfUVmdYcwwVnF-Ri9kacYDRYOB6CJyQfEB66C4s)

I don't see a "data" sub-directory in your path, the release notes mention "data/external_converters".

m0wlheld, 2025-01-04:
> > Thanks. Can you please close this issue?
> 
> Sorry for my ignorance, but I'm not able to see a "Close" button".

My bad. You are not the creator of this issue, only @my5123 can close it.

Mrsash2020, 2025-01-04:
> Open a terminal, navigate to the location of Zigbee2MQTT's configuration.yaml and create a folder named "external_converters" there. Move all you external converts to this directory. Remove the [...]

I thought you said make the external_converters folder in the folder Zigbee2MQTT folder with the Zigbee2MQTT's configuration.yaml

[image]

m0wlheld, 2025-01-04:
I don't have external converters, but the docs say "data/external_converters" and I assume that "data" must be a directory next to configuration.yaml . Just give it a try, create directory "data" and move "external_converters" into it. Then restart Z2M.

Mrsash2020, 2025-01-04:
> I don't have external converters, but the docs say "data/external_converters" and I assume that "data" must be a directory next to configuration.yaml . Just give it a try, create directory "data" and move "external_converters" into it. Then restart Z2M.

Nope. That's the first thing I tried.

[image]

m0wlheld, 2025-01-04:
I am sorry, I am out of ideas (not using external converters) by myself.

Giro-S, 2025-01-04:
> Nope. That's the first thing I tried. [...]

Same here, z2m not load files.js in external_converters directory placed in or out "data" directory

Mrsash2020, 2025-01-04:
> If I'm not wrong, in the following link: https://github.com/Koenkk/zigbee2mqtt.io/edit/master/docs/advanced/more/external_converters.md

Are you able to post your config please? It may have something to do with it for me. I have 3 other config files in the directory so unsure what the issue is.

my5123, 2025-01-04:
> Open a terminal, navigate to the location of Zigbee2MQTT's configuration.yaml and create a folder named "external_converters" there. Move all you external converts to this directory. Remove the [...]

this solved it. thx bro.

my5123, 2025-01-04:
> Ok so did not work for me. How do I confirm it worked if the unsupported devices are still unsupported?
> 
> 
> 
> 
> 
> [image]
> 
> 

I tried this and it worked.
Just for your question, if thoes devices can not worked, meens converters is not running right.

closed 2025-01-04 by my5123

Mrsash2020, 2025-01-04:
ok so one of the converters still had e.illuminance_lux or something which had an error come up but it then ignores all external converters. I removed it from the folder and other converters were picked up from 
[image]

SuperWangKai, 2025-01-04:
There is no **data** directory needed as the parent folder of "external_converters". Such an critical difference. Release notes should be corrected!

Timieh, 2025-01-11:
Indeed I had same issue. and a lot of time debugging. 
When you restart z2m you should be able to see your <name>.js converter loaded! 

In the end the 'data' folder shouldn't be there. 
Gonna look if I can change the docs.. becuase this is confusing!

Lotambena, 2025-01-11:
Any changes on the config file on the addon.
There is place external_converters then we mention them. Do we delete that line or what. I have tries all way my converters havenot loaded

Timieh, 2025-01-11:
You have to delete it, if the integration didn't did it automatically (also mentioned in the Breakning Changes).