Koenkk/zigbee2mqtt #25449: Tuya ZigBee Light Sensor Luminance Sensor _TZ3000_do6txrcw

KirkKirk, 2025-01-04:
### Link

https://www.aliexpress.com/item/1005007877339487.html?spm=a2g0o.order_list.order_list_main.35.5feb1802eiWVW1

### Database entry

{"id":33,"type":"EndDevice","ieeeAddr":"0xa4c1389c0e974eb7","nwkAddr":19339,"manufId":4417,"manufName":"_TZ3000_do6txrcw","powerSource":"Battery","modelId":"TS0222","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[1,3,1026,1029,1024,1280,0],"outClusterList":[3,25,10],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":35106}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c0dd47","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1735953833076}

### Zigbee2MQTT version

1.42.0 commit: unknown

### Comments

I tried to make it to work but I stuck. Please Help. 

### External definition

```shell
const {battery, temperature, humidity, illuminance, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0222'],
    model: 'TS0222',
    vendor: '_TZ3000_do6txrcw',
    description: 'Automatically generated definition',
    extend: [battery(), temperature(), humidity(), illuminance(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Except for Availability, nothing else works

jaymemaurice, 2025-01-11:
I have just encountered the same sensors :D

Let me know if there is any way I can help.

jaymemaurice, 2025-01-11:
Okay I think I got it working.

I am using 2.X on home assistant addon and the following worked for me...

create the external converter
```
cd /usr/share/hassio/homeassistant/zigbee2mqtt
mkdir -p external_converters
mkdir -p data
ln -s ../external_converters data/external_converters
vi TZ3000_do6txrcw.js
```

with the contents: 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const legacy = require('zigbee-herdsman-converters/lib/legacy')
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    fingerprint: [
        {modelID: 'TS0222', manufacturerName: '_TYZB01_4mdqxxnn'},
        {modelID: 'TS0222', manufacturerName: '_TYZB01_m6ec2pgj'},
        {modelID: 'TS0222', manufacturerName: '_TZ3000_do6txrcw'},
    ],
    model: 'TS0222',
    vendor: 'Tuya',
    description: 'Light intensity sensor',
    fromZigbee: [fz.battery, fz.illuminance, legacy.fromZigbee.TS0222],
    toZigbee: [],
    exposes: [e.battery(), e.illuminance()],
    configure: tuya.configureMagicPacket,
};

module.exports = definition;
```

restart.

I'll try to submit a pull request.

Cheers

jaymemaurice, 2025-01-11:
Actually I am 20 hours too late. The requisite upstream change is here:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/8584

closed 2025-01-13 by Koenkk