Koenkk/zigbee2mqtt #25450: Aqara button WXKG11LM action no longer exposed

patfelst, 2025-01-04:
### What happened?

After update to v2.0.0.0 the Aqara button WXKG11LM no longer exposes to MQTT. The documentation says that it is available.

Not sure if this is the same issue just created https://github.com/Koenkk/zigbee2mqtt/issues/25448

### How to reproduce it (minimal and precise)

Looking at the device in home assistant add-on Mosquitto broker, I can not see `Action` any more and my button automations are not working.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20220219

### Adapter

CC2652P2 Based Zigbee to PoE

### Setup

Home Assistant add-on

### Debug log

NA

patfelst, 2025-01-04:
OK I just found https://github.com/Koenkk/zigbee2mqtt/pull/24494

I'll have to look into it to see what I need to do to get my Aqara buttons working again.

K4RLMORG4N, 2025-01-04:
This is the same for most of my switches, I now have a whole bunch of Hue switches that do not have any actions

patfelst, 2025-01-04:
I use a blueprint for the Aqara buttons, which is broken now. But it's actually pretty easy to create the automations using the UI editor.
1. Create new automation
2. Trigger on `device`
3. Select your MQTT button from drop down list
4. Another drop down called `trigger` appears Select `single`, `double` or `hold` etc
5. Add a trigger ID
6. Set your action as normal.

Here's my example:
```yaml
alias: Button - workshop
description: ""
triggers:
  - domain: mqtt
    device_id: 656b55c99717ed704e669e0d521be10f
    type: action
    subtype: single
    trigger: device
    id: trigger single
  - domain: mqtt
    device_id: 656b55c99717ed704e669e0d521be10f
    type: action
    subtype: double
    trigger: device
    id: trigger double
  - domain: mqtt
    device_id: 656b55c99717ed704e669e0d521be10f
    type: action
    subtype: hold
    trigger: device
    id: trigger hold
conditions: []
actions:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - trigger single
        sequence:
          - action: switch.toggle
            metadata: {}
            data: {}
            target:
              entity_id: switch.workshop_power_strip
      - conditions:
          - condition: trigger
            id:
              - trigger double
        sequence:
          - action: light.toggle
            metadata: {}
            data: {}
            target:
              entity_id: light.led_strip_workshop_shelf_dimmer
      - conditions:
          - condition: trigger
            id:
              - trigger hold
        sequence:
          - action: light.turn_off
            metadata: {}
            data: {}
            target:
              area_id: electronics_lab
mode: single
```
Graphically in the UI editor
[image]

[image]

 I'll close this now.

closed 2025-01-04 by patfelst

icanoncn, 2025-01-16:
Same issue, I found this
https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/707#issuecomment-2585209493
It works for me.