Koenkk/zigbee2mqtt #25452: illuminance_lux sensors remain in Home Assistant after updating to 2.0.0

jackwilsdon, 2025-01-04:
### What happened?

After updating to 2.0.0, the old `illuminance_lux` sensors still exist in HA (but are now "Unavailable") and the new `illuminance` ones are missing.

I can see on MQTT that the `illuminance_lux` discovery topic is still retained.

I am seeing this on the following devices:

* [IKEA E2134](https://www.zigbee2mqtt.io/devices/E2134.html#ikea-e2134)
* [Philips 9290030675](https://www.zigbee2mqtt.io/devices/9290030675.html#philips-9290030675)

### What did you expect to happen?

`illuminance_lux` should be replaced with `illuminance`.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652RB stick

### Setup

Add-on on Home Assistant OS on Raspberry Pi 4

meiser79, 2025-01-04:
@jackwilsdon I had to activate the illuminance sensor in the device settings.
@jdcynical that's according to the release notes, illuminance_lux was replaced by illuminance.

adminotaurus44, 2025-01-04:
> @jackwilsdon I had to activate the illuminance sensor in the device settings. @jdcynical that's according to the release notes, illuminance_lux was replaced by illuminance.

How do I activate it?

ukmgranger, 2025-01-04:
I'm also having issues with this. If I enable illuminance, it doesn't seem to show any data in HA

svengemen, 2025-01-04:
The same problem with brightness detection. This no longer works after the update. What do I need to activate?

adminotaurus44, 2025-01-04:
You need to activate in the MQTT integration
[image]

KennethLavrsen, 2025-01-04:
I was bitten by this. The new sensor - if added - all mine were added - need to be enabled in HA. When you enable an entity it takes 30 seconds to see it enabled and maybe up to a minute to get a valid value. And while it does that HA makes all MQTT entities unavailable for some seconds. Really badly made in HA. I do not understand the Zigbee2MQTT decision to make the light sensors disabled by default. Even an Aqara light sensor that only reports light had the sensor disabled.

m0wlheld, 2025-01-04:
The illuminance_lux attribute has been altered to illiuminance. But for HA, the illuminance_lux entity has dissappeared, while a new illuminance entity has appeared.  If you disabled auto-activation in the Home Assistant MQTT Integration, you need to manually enable every illuminance entity there. You also may want to remove the old illuminance_lux entities in HA, too. 

This is a consequence of the announced breaking change in Zigbee2MQTT but actually a Home Assistant issue.

jackwilsdon, 2025-01-04:
I only have the `_lux` sensor - the new `illuminance` one hasn't been created at all in HA (I've restarted everything multiple times):

[image]

[image]

m0wlheld, 2025-01-04:
It has not been reported a value so far, but is is available. The Valhorn is a battery powered device and it will not constantly report values, but rather periodically or on significant changes. Try trigger one by moving into it's range.

jackwilsdon, 2025-01-04:
@m0wlheld that's the old `_lux` sensor. The VALHORN (and the Hue sensors I've got) update every few minutes and I haven't had a reading since updating to 2.0.0 (and I never had to swap out the `_lux` version for the non `_lux` version in my automations).

I also see this in my HA log repeatedly:

```
2025-01-04 11:42:47.124 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'illuminance_lux' when rendering '{{ value_json.illuminance_lux }}'
```

Which seems to be related - it's not coming from my automations.

I can prove it's the `_lux` version based on the discovery data in MQTT - on the topic `homeassistant/sensor/0x70c59cfffed95f97/illuminance_lux/config` I see this:

```json
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    },
    {
      "topic": "zigbee2mqtt/Front Motion Sensor/availability",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "availability_mode": "all",
  "device": {
    "identifiers": [
      "zigbee2mqtt_0x70c59cfffed95f97"
    ],
    "manufacturer": "IKEA",
    "model": "VALLHORN wireless motion sensor (E2134)",
    "name": "Front Motion Sensor",
    "sw_version": "1.0.64",
    "via_device": "zigbee2mqtt_bridge_0x00124b00214f230c"
  },
  "device_class": "illuminance",
  "enabled_by_default": true,
  "object_id": "front_motion_sensor_illuminance_lux",
  "origin": {
    "name": "Zigbee2MQTT",
    "sw": "1.42.0",
    "url": "https://www.zigbee2mqtt.io"
  },
  "state_class": "measurement",
  "state_topic": "zigbee2mqtt/Front Motion Sensor",
  "unique_id": "0x70c59cfffed95f97_illuminance_lux_zigbee2mqtt",
  "unit_of_measurement": "lx",
  "value_template": "{{ value_json.illuminance_lux }}"
}
```

But on the actual state topic I see this:

```
{
  "battery": 77,
  "illuminance": 984,
  "linkquality": 43,
  "occupancy": false,
  "update": {
    "installed_version": 16777316,
    "latest_version": 16777316,
    "state": "idle"
  }
}
```

So the discovery information (published by Z2M) is somehow outdated.

m0wlheld, 2025-01-04:
The new attribute name is "illuminance", without _lux, so the MQTT discovery is expected. But one of your automations is using the old illuminance_lux entity. Search it and replace it by illuminance.

jackwilsdon, 2025-01-04:
@m0wlheld that discovery data still tries to used `illuminance_lux`:

> ```json
>   "value_template": "{{ value_json.illuminance_lux }}"
> ```

It's also published to `0x70c59cfffed95f97/illuminance_lux/config`, which should be `0x70c59cfffed95f97/illuminance/config`. None of my automations use the old one, the error I am seeing matches _exactly_ the `value_template` in the discovery data.

jackwilsdon, 2025-01-04:
So it looks like this is just user error on my part :grin:

1. `homeassistant.enabled` had been removed from my configuration at some point (possibly when I was updating it with the 2.0 changes?)
2. I was previously disabling the `illuminance` sensor (in favor of `illuminance_lux`) using `device_options.homeassistant.illuminance`.

All working after fixing the above.

closed 2025-01-04 by jackwilsdon

crisc2000, 2025-02-27:
Go to:
Home assistant > setting > Devices & services > MQTT > devices > Aqara Motion sensor (use your light sensor name) > Illuminance > Setting Icon (:gear:) > Enable

Then replace anywhere in dashboard you use entity id :
sensor.xxxx_sensor_illuminance_lux
With:
sensor.xxxx_sensor_illuminance

Solution via https://community.home-assistant.io/t/2025-1-disabled-my-illuminance-sensor-entities/821694/8